Ysurac/openmptcprouter #1054: vlan on LAN

ctrlx83, 2020-06-03:
Bonjour à tous,

Je ne parviens pas à configurer une interface LAN avec un vlan.
Est ce qu'il y a une manipulation particulière à faire ou quelque chose à savoir ?

Merci à vous

Ysurac, 2020-06-03:
Il ne faut pas que MacVLAN soit utilisé, car c'est déjà des VLANs.
Ensuite tout dépend de l'arch, si il y a un switch intégré comme les WRTx/BPI-R2/... ou pas.

ctrlx83, 2020-06-03:
@Ysurac y a t il une mathode simple pour désactiver MacVLAN ?

merci

Ysurac, 2020-06-03:
Dans réseau->Macvlan en supprimant les interfaces.

ctrlx83, 2020-06-03:
@Ysurac 

ok merci
et ensuite pour assigner un vlan à une interface lan ?

Ysurac, 2020-06-03:
en utilisant comme interface physique eth1.10 pour le vlan 10 par exemple.

geekdomotique, 2020-06-11:
Bonjour, je souhaite aussi mettre en place un vlan pour du wifiguest, je vais dans réseau \ interface et la je ne sais plus quoi faire.... Faut-il ajouter une nouvelle interface ou éditer Lan-eth0 existante?
Merci d'avance pour ta réponse Ysurac

Ysurac, 2020-06-11:
@geekdomotique wifi interne ? Ca dépend comment il a été configuré. Le plus simple dans ce cas est qu'il soit sur sa propre interface avec vlan.

geekdomotique, 2020-06-12:
Pour être plus précis je souhaite que mon vlan50 (dédié wifiguest) soit aussi déclaré coté openmptprouteur, est ce par le macvlan qu'il faut passer ou alors créer une régle vlan, je suis perdu et ne sais pas comment faire...

Ysurac, 2020-06-12:
il faut créer une interface basée sur, par exemple, eth0.50

geekdomotique, 2020-06-18:
Je créé une nouvelle interface avec en nom : VLAN50 et protocole IPstatique, je bridge interface eth0 et après je ne vois pas ou je peux tague un vlan.... Help

Ysurac, 2020-06-18:
Lors de l'édition d'une interface, dans l'onglet paramètres physique en créant une interface personnalisée nommée eth0.50, sans bridge.

closed 2020-08-12 by Ysurac