Ysurac/openmptcprouter #1057: How to configure multiple Modems for Link Aggregation

vmmz, 2020-06-04:
## Expected Behavior
Use of multiple 3G Modems (UMTS/GPRS/EV-DO) and DSL (wifi client@home router) for link aggregation.
## Current Behavior
* 3G Modem intefaces only showing "Gateway DOWN" message
* No connection between default WAN1 and WAN2 interfaces to Modem devices

## Context (Environment)
Not able to set static IP addresses for modem devices. "UMTS" Protokol needs DHCP to get an IP from provider.

Please share example network/interfaces configuration for multiple modems and wifi interfaces.

## Specifications

  - OpenMPTCProuter version: v0.54.7 r0+12509-04a21c26a0
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter platform: x86_64

Ysurac, 2020-06-04:
For USB dongles, you need to create new interfaces in Network->interfaces, when they are working you can enable multipath in Network->MPTCP (and disable on default not used interfaces).

vmmz, 2020-06-04:
This can work for wifi connections. But how can modem ip be set to static one? Especially 3G modems are using DHCP to get ip from provider. As soon modem proto is changed to static -> no connection can be established anymore.

Ysurac, 2020-06-04:
You don't need to set interfaces as static for USB modem.

vmmz, 2020-06-04:
Ok, e.g. I am going to use a usb modem as a further Link. Interface configuration is done and I get an IP address from provider. Status Page shows both interfaces. Wifi (use static ip) is green but the modem interface still have a warning "Geteway DOWN". Why is it so?
[image]

Ysurac, 2020-06-04:
Ping to gateway is blocked by some dongle/ISP. It's only a warning but not a problem, all should work as normal. The warning can be disabled in System->OpenMPTCProuter and "advanced parameters" tab in "disable gateway ping" checkbox.

vmmz, 2020-06-04:
It seems to work now. Thank you very much for great support!

[image]