Ysurac/openmptcprouter #1058: Using multiple MPTCPROUTERs with a single VPS

blainvillem, 2020-06-04:
## Expected Behavior

Have several MPTCPROUTERs using a single VPS machine. But each MPTCPROUTER would have different keys, so you would have to generate /root/openmptcprouter_config.txt files for each MPTCPROUTER.
The VPS machine would have as many public IP and network cards as there are existing MPTCPROUTERS.

## Current Behavior

I have one MPTCPROUTER for one VPS.

## Possible Solution

Generate several /root/openmptcprouter_config.txt files and use one file per MPTCPROUTER.
And mention a unique public ip address in each mptcprouter.

## Specifications

  - OpenMPTCProuter version:  v0.54.7
  - OpenMPTCProuter VPS version: Ubuntu 18.04
  - OpenMPTCProuter platform: VMWARE ESXi

Ysurac, 2020-06-04:
Support of multi OpenMPTCProuter is available via the API but not documented yet.
It's in test/dev with partners.

blainvillem, 2020-06-04:
All right, thanks for the very quick response.
What do you mean by "API".
Isn't it possible to generate a new configuration with a different name without breaking the first one?

Ysurac, 2020-06-04:
There is an API (Application programming interface), using FastAPI in Python, on the server side that is used to retrieve keys, set firewall config, and all needed configuration.
The file "openmptcprouter_config.txt" is only a reminder of keys, it's not used anywhere.

Florent4223, 2020-06-09:
Hello

I am very interested in this discussion. How can we use this API?

thanks

Ysurac, 2020-06-12:
As I said, for now this API is only tested with partners. I don't want to have to answer to many issues that can be related to that and I don't have yet redacted a doc.

closed 2020-08-12 by Ysurac

sadahsan, 2021-11-22:
Hi Yannick, would appreciate if you could please that is there any solution for it?
Thanks

abpprkonsalting, 2024-09-21:
Hello, I've seen in multiple answers to this question that some work had been done in this issue, but it is not documented. At the same time it looks as if @Ysurac does not consider this a priority. Anyway for me it could be, and I would be happy to help in any task that could bring this to production stage, or published, or whatever.

rdmitry0911, 2025-03-13:
Still not working. Would like it to work too and also can offer a help in testing this feature when it is available for testing

Ysurac, 2025-03-13:
It's working correctly. Only not documented as I still don't want (and no time for that) to make support on this. There is no real reasons to use this as individual.

It's still quite easy to use via VPS API.

No need to comment on multiples closed issues.