Ysurac/openmptcprouter #1059: Slow ethernet speed with 'ondemand' governor

robertoglxda, 2020-06-04:
## Expected Behavior
Using iperf between a raspberry pi4 and a connected pc, the full gigabit speed of the ethernet port of the pi4 is expected.

## Current Behavior
After some hours or days after the boot of the pi4, the speed is halved or worst. Using iperf, I get ~480mbps from the pc to the pi4, and ~300mpbs from the pi4 to the pc.

## Possible Solution
With the default OMR settings, the 'ondemand' governor is used. Switching to 'performance' appears to fix the issue.
Rebooting the pi4 while the speed is halved temporarily solves the issue of course.

## Steps to Reproduce the Problem

  1. Leave OMR running some days
  2. Monitor the iperf performance 

## Context (Environment)
I'm testing everything with a raspberry pi4 and a Windows 10 pc connected directly to it. The iperf tool is already available in the OMR release. 
If the iperf test is performed after some minutes or hours (after the pi4 has booted), the test returns ~940mbps both from the pc to the pi4 and from the pi4 to the pc.

If the iperf test is performed after a couple of days, the speed is halved or worst. During my tests this appears to be the case only with the 'ondemand' governor: after switching to the 'performance' governor, I did not see the issue anymore.

Please note that I'm not sure if this is an issue with the pi4 itself or if this is related to OMR. I'm using the pi4 for my home connection so I cannot leave it with the default OS to test this for some days, sorry.

## Specifications

  - OpenMPTCProuter version: 0.54.7
  - OpenMPTCProuter VPS version: 0.1015 4.19.104-mptcp
  - OpenMPTCProuter platform: RPI4 with ext4 image 'openmptcprouter-v0.54.7-r012509-04a21c26a0-bcm27xx-bcm2711-rpi-4-ext4-factory.img.gz'

Ysurac, 2020-06-04:
This will be fixed in next release.
Changing speed on RPI (it's what ondemand try to do) seems to make many problems, an issue is already open in raspberry kernel github.
Thanks for the tests, that problem impact more speed than I would expect.

robertoglxda, 2020-06-04:
Thank you for your quick response, and for the fix you already have.

As a note, I'd like to point out that in my situation (with the 'ondemand' governor), when iperf reports the 'slow ethernet speed', some devices on the network have their speed halved for some reason. Example (pc1 and pc2 are connected to an access point, that is connected via ethernet to the pi4):
a) iperf reporting ~940mbps from pc1/pc2 and the pi4, speedtest from pc1 reports 100mbit, speedtest from pc2 reports 100mbit;
b) iperf reporting ~450mbps from pc1/pc2 and the pi4, speedtest from pc1 reports 100mbit, speedtest from pc reports 40mbps. iperf between pc1 and pc2 reports a full ~940mbps (because it is using the access point, and not the pi4 for this).

I'm not sure of the reason, but I've see this. It is also solved changing governor.
This is why I noticed the issue in first place, as my main pc was slower that a laptop I was using for some tests with OMR.

However everything is good with the performance governor.

closed 2020-10-08 by github-actions[bot]