Ysurac/openmptcprouter #1060: MPTCP only uses wan1 for download and wan2 for upload.

aFrosty0, 2020-06-05:
## Expected Behavior
Expect both wans available bandwidth to be combined.

## Current Behavior
When running a speed test and watching the realtime traffic graph in the MPTCP settings of OpenMPTCProuter, can see it only uses wan1 when downloading and wan2 when uploading.

While experimenting with settings (ShadowSocks on / off, various default VPN) on occasion I have seen it use both wans simultaneously, but I can't reliably reproduce this, and the total speed is always far worse than either of the connections would be alone.

## Possible Solution
Unknown

## Steps to Reproduce the Problem

  1. Setup 2 wans as normal, using default settings.
  2. Watch realtime traffic graphs for both wans under MPTCP settings.
  3. Run speedtest.net

[image]

## Context (Environment)
2 LTE connections on different networks, speeds vary throughout the day but on average:
wan1: 40 down 5 up
wan2: 10 down 10 up

My goal is to combine multiple LTE connections for reliable live video streaming from areas with limited signal strength.
I am currently just testing at home, but the plan is for 4 connections from different networks.

[image]

## Specifications

  - OpenMPTCProuter version: 0.54.7
  - OpenMPTCProuter VPS version: 0.1015 4.19.104-mptcp
  - OpenMPTCProuter platform: RPI4

Ysurac, 2020-06-05:
You can try in Network->MPTCP to set scheduler to blest and tcp congestion to cubic.
Check in the "MPTCP fullmesh" tab that you have IP of all your connections.
Check also in State->System log if you don't have errors.

aFrosty0, 2020-06-05:
**Additional note:**
Sometimes it swaps and uses wan2 for download, wan1 for upload!

**Status > System log:**
I'm not seeting any MPTCP errors. It does seem to temporarily switch off either wan quite often due to a ping error, which I also see on the status page quite often. But this is usually immediately after changing settings, sometimes while running a speed test.

**Setting blest and cubic:**
Still only uses one wan for up and other for down.
Speed also seriously reduced to less than 4 mbits!

**MPTCP Fullmesh:**
Index, Address-ID, Backup, IP-address, if-idx
IPv4, next v4-index: 6
3, 4, 0, 192.168.8.20 10
4, 5, 0, 192.168.91.20 11
IPv6, next v6-index: 1

**MPTCP support check:**

I'm not sure what most of this means, but it is quite different for each wan, may be useful to you.
(I have removed VPS IP and "senders key")

wan1:

1: 192.168.8.1 
2: * 
3: * 
4: * 
5: * 
6: XX.XX.XXX.XXX TCP::SrcPort TCP::DstPort TCP::SeqNumber TCP::AckNumber TCP::DataOffset TCP::Flags TCP::WindowsSize TCP::CheckSum IP::TotalLength IP::Identification IP::TTL IP::CheckSum IP::SourceIP IP::DestinationIP +TCPOptionMaxSegSize -TCPOptionMPTCPCapable 

Result:
< Ethernet (14 bytes) :: DestinationMAC = 02:01:e9:d3:f5:da , SourceMAC = e8:3f:67:6a:22:ad , Type = 0x800 , >
< IP (20 bytes) :: Version = 4 , HeaderLength = 5 , DiffServicesCP = 0 , ExpCongestionNot = 0 , TotalLength = 44 , Identification = 0xdb13 , Flags = 2 , FragmentOffset = 0 , TTL = 250 , Protocol = 0x6 , CheckSum = 0xdcd7 , SourceIP = XX.XX.XXX.XXX , DestinationIP = 192.168.8.20 , >
< TCP (20 bytes) :: SrcPort = 65101 , DstPort = 59879 , SeqNumber = 3394453443 , AckNumber = 184490020 , DataOffset = 6 , Reserved = 0 , Flags = ( SYN ACK ) , WindowsSize = 8190 , CheckSum = 0x9d64 , UrgPointer = 0 , >
< TCPOptionMaxSegSize (4 bytes) :: Kind = 2 , Length = 4 , MaxSegSize = 536 , >

wan2:

1: 192.168.91.1 
2: * 
3: 192.168.213.21  [PARTIAL] +PartialTCP IP::DiffServicesCP IP::TotalLength IP::TTL IP::CheckSum 
4: 192.168.213.22  [PARTIAL] +PartialTCP IP::DiffServicesCP IP::TotalLength IP::TTL IP::CheckSum 
5: * 
6: * 
7: 63.130.105.130 TCP::DataOffset TCP::CheckSum IP::DiffServicesCP IP::TotalLength IP::TTL IP::CheckSum +TCPOptionMaxSegSize 
8: * 
9: 91.121.215.119 TCP::DataOffset TCP::CheckSum IP::TotalLength IP::TTL IP::CheckSum +TCPOptionMaxSegSize 
10: * 
11: * 
12: * 
13: * 
14: XX.XX.XXX.XXX TCP::SrcPort TCP::DstPort TCP::SeqNumber TCP::AckNumber TCP::DataOffset TCP::Flags TCP::WindowsSize TCP::CheckSum IP::DiffServicesCP IP::TotalLength IP::Identification IP::TTL IP::CheckSum IP::SourceIP IP::DestinationIP +TCPOptionMaxSegSize TCPOptionMPTCPCapable::Sender's Key 

Result:
< Ethernet (14 bytes) :: DestinationMAC = 46:a5:0a:99:fa:52 , SourceMAC = 50:d4:f7:a6:b4:5e , Type = 0x800 , >
< IP (20 bytes) :: Version = 4 , HeaderLength = 5 , DiffServicesCP = 10 , ExpCongestionNot = 0 , TotalLength = 56 , Identification = 0x0 , Flags = 2 , FragmentOffset = 0 , TTL = 49 , Protocol = 0x6 , CheckSum = 0x2db8 , SourceIP = XX.XX.XXX.XXX , DestinationIP = 192.168.91.20 , >
< TCP (20 bytes) :: SrcPort = 65101 , DstPort = 61209 , SeqNumber = 2251655596 , AckNumber = 1065049156 , DataOffset = 9 , Reserved = 0 , Flags = ( SYN ACK ) , WindowsSize = 65535 , CheckSum = 0x7aa2 , UrgPointer = 0 , >
< TCPOptionMaxSegSize (4 bytes) :: Kind = 2 , Length = 4 , MaxSegSize = 1320 , >
< TCPOptionMPTCPCapable (12 bytes) :: Kind = 30 , Length = 12 , Subtype = 0 , Version = 0 , Checksum = 1 (Checksum Enabled) , Flags = 0 , Crypto = 1 (HMAC-SHA1) , Sender's Key = Sender's Key = XXXXXXXXXXXXXXXXXXXX , >

Ysurac, 2020-06-05:
"-TCPOptionMPTCPCapable" on wan1 is not good. This happen when MPTCP flag is filtered by ISP.
You can try MPTCP over VPN on this interface.

aFrosty0, 2020-06-05:
Ahh, so ISP for wan1 doesn't like MPTCP.
I have SIMs for 2 other networks so I'll use this to find which networks allow MPTCP.

I have tried using MPTCP over VPN during my testing, but regardless of the ISP I always get a "Gateway DOWN" error on the ovpnwan:

[image]

I can solve this by changing the OpenVPN instance "wan1" Protocol from udp to tcp-client.

Now MPTCP seems to be working, but it's very very slow:
(I have tested with both the default settings and BLEST / cubic from your earlier suggestion.)
[image]

I will now try with Glorytun / ADSVPN / MLVPN too.

aFrosty0, 2020-06-05:
Whichever "Default VPN" I choose in the advanced section of the Wizard, selecting MPTCP over VPN for a wan interface will always use OpenVPN, is this as intended?

Selecting MLVPN as the default, the ovpnwan1 interface now works on UDP!
(Only works with protocol tcp-client when the default VPN is set to OpenVPN). I can't understand this?!

Whatever VPN I use, the results are still very slow, or I simply get "VPN tunnel DOWN".

I think these VPN troubles are a problem for another day. I will change wan1 to a different network and resume my testing.

**Discovered an unrelated bug:**
With MPTCP over VPN enabled for a wan interface. If you change default VPN in the Wizard, it breaks the ovpnwan interface.
You have to manually delete the wan interface, the oVPN instance and the ovpnwan interface. Then re enter the wan details in the Wizard.

aFrosty0, 2020-06-05:
With an alternative ISP I'm getting much better performance :)

Also, with the original ISP that doesn't support MPTCP, if I put the LTE router / modem in bridge mode so OMR gets the wan IP address (it's not actually an internet IP as the ISP uses CGNAT), VPN performance is significantly improved. (MPTCP still doesn't work)

Is the VPN behavior I described in my earlier post as expected? I'm going to continue testing various configurations to find the most reliable setup, so it would be good to know if I have some problems with the VPN services.

Thank you so much for your help.

aFrosty0, 2020-06-05:
Experiencing a further VPN issue.
The MPTCP over VPN wan - ovpnwav2 - is showing "Gateway DOWN"... But it still works! 

[image]

Ysurac, 2020-06-05:
MPTCP over VPN use OpenVPN in UDP mode.
I fixed some issues about that for next release.

closed 2020-10-08 by github-actions[bot]