Ysurac/openmptcprouter #1061: dedicated server with 3 ip

ascomputer, 2020-06-05:
## Expected Behavior
hello everyone, having a dedicated server with 3 static public ip and three client computers can I make routing rules to make each client use a different server ip? example pc client 192.168.100.2 exits on the first ip of the server XXX.XXX.XX2 then pc client 192.168.100.3 exits on the second ip of the server xxx.xxx.xx3
Thanks in advance

## Specifications

  - OpenMPTCProuter version:  v0.54.7 r0+12509-04a21c26a0
  - OpenMPTCProuter VPS version: (Last version) 
  - OpenMPTCProuter platform: x86_64)

Adorfer, 2020-06-05:
This is possibly a duplicate of your ticket (where you had 8 public IPs on the VPS)
https://github.com/Ysurac/openmptcprouter/issues/1056
and which was a duplicate of:
https://github.com/Ysurac/openmptcprouter/issues/1006

ascomputer, 2020-06-05:
I apologize I did not remember having already asked the question ... forgive me

closed 2020-06-05 by ascomputer