Ysurac/openmptcprouter #1063: configuration with the Wrt3200acm speed

reteste, 2020-06-07:
I made the configuration with the Wrt3200acm, so I have 2 4g routers and I also configured the usb to connect the smatrphone. So far so good no problem apparently but the fact is that the speed test does not seem to give any improvement .... I explain myself, by running the test by connecting a single 4g router I get the same result that connecting them both or even adding the smartphone (Samsung s20 ultra), if tested individually, the speeds are approximately comparable but then in bonding it does not increase by anything and stands at the same value as one.
Looking at the graph in real time (from wrt) during Speedtest Ookla all the bandwidth is provided by a single modem, the others go to kb! What could it be?
  

  - OpenMPTCProuter version: <openmptcprouter v0.54.7 r0+12509-04a21c26a0 / LuCI Master git-20.069.54281-3fb8e3d>
  
[image]

ascomputer, 2020-06-07:
some isp operators may block or filter a multi path tcp connection ... so try flagging MPTCP over VPN in system -openmptcprouter - setting wizard flag for each link isp wan usb wifi etc.

reteste, 2020-06-07:
does not work

ascomputer, 2020-06-07:
show me the status screen in system -openmptcprouter-status

reteste, 2020-06-07:
made screen

ascomputer, 2020-06-07:
I do not see anything

reteste, 2020-06-07:
what exactly should i send you?

reteste, 2020-06-07:
Are these settings correct?

ascomputer, 2020-06-07:
Screenshoot

closed 2020-10-10 by github-actions[bot]