Ysurac/openmptcprouter #1064: Channge DNS Server

LeprovostNoam, 2020-06-07:
## Expected Behavior
By default, openmptcprouter use the VPS as DNS server but i want to use 1.1.1.1 & 1.0.0.1

## Current Behavior
I've changed "transmissions DNS" input, but not working DNS_PROBE_FINISHED_BAD_CONFIG
[image]

## Specifications

  - OpenMPTCProuter version:  openmptcprouter-v0.54.7
  - OpenMPTCProuter platform: Linksys WRT3200ACM

LeprovostNoam, 2020-06-07:
adding 1.1.1.1#53 & 1.0.0.1#53 is not possible, only on "Domaine local"

Ysurac, 2020-06-07:
You can set it as local server. It's an OpenWrt issue.
But why would anyone use Cloudfare DNS ? It's really strange...
Cloudfare DNS are used for OMR-Tracker, this can give some strange behaviour.

Adorfer, 2020-06-07:
if you want do something fancy, use DoH and combine it with Pi-Hole.
https://openwrt.org/docs/guide-user/services/dns/doh_dnsmasq_https-dns-proxy

LeprovostNoam, 2020-06-09:
> Vous pouvez le définir comme serveur local. C'est un problème OpenWrt.
> Mais pourquoi quelqu'un utiliserait-il Cloudfare DNS? C'est vraiment étrange ...
> Cloudfare DNS est utilisé pour OMR-Tracker, cela peut donner un comportement étrange.

c'est à dire un comportement étrange?

Ysurac, 2020-06-09:
Dans certains cas une route directe via un WAN est créée en direction des IPs utilisées par OMR-Tracker. Donc lors de l'interrogation par un client vers ces IPs la connexion peut ne pas aboutir de temps en temps.

LeprovostNoam, 2020-06-09:
D'accord, et bien je n'ai pas eue de problème encore. Le truc c'est que netflix me bloque si j'ai le dns de mon serveur google cloud et pas cloudflare.. J'avais le même problème avec speedify

Ysurac, 2020-06-09:
Etrange ça, normalement netflix bloque par IP source.
Par défaut les serveurs DNS racines sont utilisés (donc pas ceux du VPS ou d'un FAI).

closed 2020-10-14 by github-actions[bot]