Ysurac/openmptcprouter #1065: IPv6 is bypassing OMR

kevin39, 2020-06-08:
Hi,
## Expected Behavior
OMR used as ipv4 and ipv6 gw.

## Current Behavior
ipv4 is working
My ipv6 gateway is one of my ISP.

## Possible Solution
Don't know

## Steps to Reproduce the Problem

  1. I've 3 ISP
  2. DHCP disabled on all ISP routers
  3. One of my IPS is forcing DHCP ipv6 to ON. I can't disable it.
So in my local, my computers are having a local ipv4 from OMR (and all is working fine) and the local ipv6 is emitted from on of my box because ipv6 is enabled and I can't disable it.

is there a way to bypass my ISP's router dhcp ipv6 to force everything via OMR ? 
Because some device like iPhone, I can't disable ipv6 so I lost any benefit from agregation because ipv6 is priorized :(

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.54.7 r0+12509-04a21c26a0
  - OpenMPTCProuter VPS version: 0.1015 4.19.104-mptcp
  - OpenMPTCProuter platform: openmptcprouter v0.54.7 r0+12509-04a21c26a0

kevin39, 2020-06-09:
Fixed.
I blocked ipv6-forwarding through an EdgeRouter X, I'm using it as a switch.

FYI, to block ipv6 : 
interfaces ethernet eth2 ipv6 disable-forwarding

closed 2020-10-12 by github-actions[bot]

bdaylik, 2024-04-21:
Although it does not fix the actual issue, if you do not have a managed switch in between your ISP modem and your devices to prevent DHCPv6 traffic, you may at least prevent IPv6 AAAA records from showing up in unbound DNS responses.

Adding the following statement in OMR panel under Services -> Recursive DNS -> Files -> Edit: Server and saving and rebooting will remove any AAAA responses in DNS, thus would mostly block any ipv6 traffic:

`private-address: ::/0`

Note that this is a hacky solution and may cause some sites/services to be unreachable.