Ysurac/openmptcprouter #1066: Flooding issue on one interface

olivierb2, 2020-06-08:
## Expected Behavior

Using my second ADSL connection (Freebox Crystal) in normal condition as it's working perfectly working with my first + 4G connection.

## Current Behavior

As soon I'm enable the third interface (192.168.0.3/24 > 192.168.0.254), the number of network visible from status page become abnormal (>10000) while have usually few hundred. After few dozen of second, the box (192.168.0.254) become unresponsive (ping not answering anymore) and interface is finally disabled by the tracker.

## Possible Solution

No idea yet, ping come back as soon I disable the interface from MPTCP page.

## Steps to Reproduce the Problem

  1. Open wizard, and add interface (eth1) with IP adress.
  2. Save, interface appears into status page.
  3. Wait few second, the box goes green, then after a while become unresponsive.

## Context (Environment)

I'm using a VM into a Qemu/KVM environment. All network are in different VLAN (separated by a Switch, and host server).

## Specifications

  - OpenMPTCProuter version: 0.55beta7
  - OpenMPTCProuter VPS version: 0.1015
  - OpenMPTCProuter platform: x86_64

Ysurac, 2020-06-08:
I need full configuration, from System->OpenMPTCProuter and show all settings tab, without keys and/or server ip.

olivierb2, 2020-06-08:
Thanks for this fast answer, here the requested config dump.

[config-dump.txt](https://github.com/Ysurac/openmptcprouter/files/4744617/config-dump.txt)

Ysurac, 2020-06-08:
Please paste it on a message, I don't like to download file (it's easier to read when directly inside the issue for me).

closed 2020-10-11 by github-actions[bot]