Ysurac/openmptcprouter #1067: Question : Qos Voip

Florent4223, 2020-06-09:
Hello

I have to pass on the VPN both data and voice (SIP protocol)
Can we prioritize the UDP protocol (IP voice support) over TCP?

thanks

version : 0.54.7 on RPI3

Adorfer, 2020-06-09:
i did that manually by a) priorty for the AS of the SIP provider and b) by (the same) for the priority of the local SIP-phones IPs. 

An automatic rule for SIP-RTP would be nice (but since it uses lot's of ports, especially if you have dozends of SIP devices in a Network) 
[image]
[image]

Florent4223, 2020-06-09:
it's great that's exactly what i need ... but where is the menu or do you set that?

Florent4223, 2020-06-09:
sorry i just found
I will try it out thank you very much

closed 2020-06-11 by Florent4223