Ysurac/openmptcprouter #1068: How to Read MPTCP Check

ccmks, 2020-06-12:
I have following log to check MPTCP, but I don't know how to read it. How do I know what parameter is indicated that the ISP block MPTCP?

1: 192.168.5.1 
2: 139.XXX.220.XXX TCP::SrcPort TCP::DstPort TCP::SeqNumber TCP::AckNumber TCP::DataOffset TCP::Flags TCP::WindowsSize TCP::CheckSum IP::TotalLength IP::Identification IP::Flags IP::TTL IP::CheckSum IP::SourceIP IP::DestinationIP +TCPOptionMaxSegSize +TCPOptionPad -TCPOptionMPTCPCapable 

Result:
< Ethernet (14 bytes) :: DestinationMAC = 00:15:5d:37:bb:06 , SourceMAC = 50:d4:f7:f4:60:aa , Type = 0x800 , >
< IP (20 bytes) :: Version = 4 , HeaderLength = 5 , DiffServicesCP = 0 , ExpCongestionNot = 0 , TotalLength = 48 , Identification = 0xa40a , Flags = 0 , FragmentOffset = 0 , TTL = 127 , Protocol = 0x6 , CheckSum = 0x696a , SourceIP = 139.XXX.220.XXX , DestinationIP = 192.168.5.100 , >
< TCP (20 bytes) :: SrcPort = 65101 , DstPort = 33510 , SeqNumber = 3780553514 , AckNumber = 1691398281 , DataOffset = 7 , Reserved = 0 , Flags = ( SYN ACK ) , WindowsSize = 64512 , CheckSum = 0x3903 , UrgPointer = 0 , >
< TCPOptionMaxSegSize (4 bytes) :: Kind = 2 , Length = 4 , MaxSegSize = 1374 , >
< TCPOptionPad (1 bytes) :: Kind = 1 , >
< TCPOptionPad (1 bytes) :: Kind = 1 , >
< TCPOptionPad (1 bytes) :: Kind = 1 , >
< TCPOptionPad (1 bytes) :: Kind = 0 , >

Ysurac, 2020-06-12:
When there is "-TCPOptionMPTCPCapable", then MPTCP seems to be blocked.

ccmks, 2020-06-12:
So is my log indicated MPTCP being blocked? Just to make sure

Ysurac, 2020-06-12:
Difficult to say if it's blocked or if the server doesn't support MPTCP. Strange that there is only one hop.

talalmash, 2020-06-14:
I get the same result on my double NAT ADSL ISP, as one hop.

closed 2020-10-17 by github-actions[bot]