Ysurac/openmptcprouter #1070: Reboot all the day...

pepitoooooo, 2020-06-15:
Hello everybody,
i need you for a bug (i think).

I use Openmptcp on a Raspberry Pi 4 with 2 Wan VDSL and 1 VPS OVH.
Agregation work very good !
But... every day it's necessary to reboot openmptcp on my raspberry because agregation don't work. Openmptcp use only first Wan... When i reboot openmptcp it's good and agregation it's ok using 2 Wan.
The next day, openmptcp use only first wan, i reboot, it's ok ...

Have you an idea to resolve this bug ?

Thank you a lot !

Ysurac, 2020-06-15:
Use template when you open issues. Else I need to ask versions, arch,...

closed 2020-06-15 by Ysurac

pepitoooooo, 2020-06-15:
Oh sorry... do you want i create a new ticket ?

Ysurac, 2020-06-15:
yes thanks :)

pepitoooooo, 2020-06-15:
the specification don’t work on ticket.

## Expected Behavior

Hello everybody,
i need you for a bug (i think).

## Current Behavior

I use Openmptcp on a Raspberry Pi 4 with 2 Wan VDSL and 1 VPS OVH.
Agregation work very good !
But... every day it's necessary to reboot openmptcp on my raspberry because agregation don't work. Openmptcp use only first Wan... When i reboot openmptcp it's good and agregation it's ok using 2 Wan.
The next day, openmptcp use only first wan, i reboot, it's ok ...

Have you an idea to resolve this bug ?

Thank you a lot !

## Possible Solution

## Steps to Reproduce the Problem

  1. boot openmptcp on raspberry Pi 4. openmptcp it's ok
  2. Wait around 24 hours
  3. openmptcp it's not ok. Agregation use only first wan but not the secondary wan.

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter platform:

Ysurac, 2020-06-15:
This work, but  hide the text.