Ysurac/openmptcprouter #1072: Reboot all the day

pepitoooooo, 2020-06-15:
## Expected Behavior

Hello everybody,
i need you for a bug (i think).

## Current Behavior

I use Openmptcp on a Raspberry Pi 4 with 2 Wan VDSL and 1 VPS OVH.
Agregation work very good !
But... every day it's necessary to reboot openmptcp on my raspberry because agregation don't work. Openmptcp use only first Wan... When i reboot openmptcp it's good and agregation it's ok using 2 Wan.
The next day, openmptcp use only first wan, i reboot, it's ok ...

Have you an idea to resolve this bug ?

Thank you a lot !

## Possible Solution

## Steps to Reproduce the Problem

  1. boot openmptcp on raspberry Pi 4. openmptcp it's ok
  2. Wait around 24 hours
  3. openmptcp it's not ok. Agregation use only first wan but not the secondary wan.

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter platform:

pepitoooooo, 2020-06-15:
- OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter platform:

pepitoooooo, 2020-06-15:
## - OpenMPTCProuter version: 
 ## - OpenMPTCProuter VPS version: 
 ## - OpenMPTCProuter platform:

Ysurac, 2020-06-15:
There the template, but I still need OpenMPTCProuter version, platform (I think it's RPI4) and VPS script version.
Also, how do you see that you lost agregation ? In download or upload side ?

pepitoooooo, 2020-06-15:
OpenMPTCProuter version: v0.54.7 r0+12509-04a21c26a0
OpenMPTCProuter VPS version: Version 0.1015 4.19.104-mptcp
OpenMPTCProuter platform: Raspberry Pi 4 Model B Rev 1.2

I lost agregation in download.

Ysurac, 2020-06-15:
What is the result of ```omr-test-speed```, ```omr-test-speed wan1``` and ```omr-test-speed wan2``` via SSH on the RPI (ctrl+c to stop after 2 minutes) ?

pepitoooooo, 2020-06-15:
omr-test-speed
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  8 10.0G    8  830M    0     0  16.2M      0  0:10:28  0:00:51  0:09:37 15.8M
--------------------------------------------------------------------------------------
omr-test-speed wan1
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  5 10.0G    5  589M    0     0  10.0M      0  0:16:54  0:00:58  0:15:56 10.1M
-----------------------------------------------------------------------------------
omr-test-speed wan2
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  4 10.0G    4  470M    0     0  8435k      0  0:20:43  0:00:57  0:19:46 8447k

pepitoooooo, 2020-06-15:
Actually it's ok.
i run the same command tomorrow and i give you the results here ;)

pepitoooooo, 2020-06-16:
omr-test-speed
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  5 10.0G    5  539M    0     0  14.7M      0  0:11:34  0:00:36  0:10:58 14.3M
----------------------------------------------------------------------------------------------
omr-test-speed wan1
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  4 10.0G    4  424M    0     0  5222k      0  0:33:27  0:01:23  0:32:04 4933k
----------------------------------------------------------------------------------------------
omr-test-speed wan2
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  2 10.0G    2  305M    0     0  5211k      0  0:33:31  0:01:00  0:32:31 5083k

pepitoooooo, 2020-06-16:
if i reboot the RPi4 it's ok

Ysurac, 2020-06-16:
aggregated speed is not so bad, the problem seems to be with wan1 and wan2. Strange.
VDSL are ethernet modem/router/box ?

pepitoooooo, 2020-06-16:
VDSL box wan1 and wan2. The same box. I think it's not the box who create the bug because when i reboot RPi4 only it's ok

pepitoooooo, 2020-06-16:
when i test speed with fast.com i have only wan1 use

closed 2020-10-19 by github-actions[bot]