Ysurac/openmptcprouter #1073: RAM consumption problem

couili, 2020-06-17:
Hello,
I just got the fiber optic at home, so I ran the VPS script on a freshly installed Debian 10 on an ESXI. Everything works fine, but after a while, the RAM consumption becomes excessive, on my VPS at Scaleway, I didn't have this problem. Here, I have a constant consumption of 4GB of RAM, whereas when starting the machine the RAM consumption is less than 1GB of RAM.
I feel like the "omr-admin.py" script is running multiple times.

[image]

## Possible Solution
Reboot VPS

## Specifications

  - OpenMPTCProuter version: 0.54.7
  - OpenMPTCProuter VPS version: v0.1015
  - OpenMPTCProuter platform: x86_64

Can you please help me?

Thanks

Ysurac, 2020-06-17:
You can check using ```ps aux | grep omr``` that there is only one script running. htop list all threads of a process.
There is many running threads in your case.

couili, 2020-06-17:
Here is the result of the command :
[image]

couili, 2020-06-18:
Hello,
After a day of operation, almost all of my RAM is in use
[image]

Ysurac, 2020-06-18:
What is the result of ```ps aux --sort=-%mem | head``` ?

Ysurac, 2020-06-18:
Do you do some p2p ? or anything that is not browsing internet ?

couili, 2020-06-18:
I do openvpn for telecommuting, as well as openvpn when I'm away from home to access my LAN, and I have cameras that I think work in P2P when we want to watch the images from outside, but we rarely watch.

Ysurac, 2020-06-18:
I think it's related to tcp/udp buffer. This may be related to a MTU too high.
This should be better in next release.

JorgeMoort89, 2020-06-18:
Hi Ysurac, where I can edit the size of those buffers?

Ysurac, 2020-06-19:
This will be in next release, and it's in current snapshot VPS script and beta router release.

couili, 2020-06-19:
Thanks for your help

closed 2020-06-19 by couili