Ysurac/openmptcprouter #1074: VPN Tunnel is down when switching to a new VPS host.

codinandhaulin, 2020-06-17:
## Expected Behavior
Able to connect to VPS on ServerCheap.net

## Current Behavior
I have a perfectly working installation on Vultr.com (debian 10) release version of openmptcprouter.  I would like to instead use a VPS from ServerCheap.net (mostly because of the bandwidth limitations of Vultr)
When I enter the IP address and all of the keys from the new ServerCheap vps, the status says "VPN tunnel down" and it is reporting the following for the VPS:
Can't get public IP address from ShadowSocks
Can't contact Server Admin Script (no answer)
**Note, I noticed that I had to manually add the VPN keys in the advanced section of the wizard, as only configuring the server ip, name and key did not allow it to find the other keys on the server.**

[image]

## Possible Solution
none

## Steps to Reproduce the Problem

  1. Fresh instance of debian 10 on ServerCheap.net
  2. run the vps script:  `wget -O - https://www.openmptcprouter.com/server/debian10-x86_64.sh | sh`
  3. reboot VPS
  4. On working installation of Openmptcprouter, setup wizard, configure the new VPS (ip address, name, server key.  Also had to manually enter the ShadowSock settings and VPN settings in the advanced section of the server configuration (they were not automatically discovered by the router using the server ip, name and key).

## Context (Environment)
I would like to switch to a more powerful VPS with unmetered bandwidth.

## Specifications

  - OpenMPTCProuter version: v0.54.7 r0+12509-04a21c26a0
  - OpenMPTCProuter VPS version: < OpenMPTCProuter VPS 0.1015 >
  - OpenMPTCProuter platform:  x86_64

Ysurac, 2020-06-17:
Key are retrieved from the API, so when the API can't be reached, no settings/key are retrieved.
Can you give me the result of ```ps aux | grep omr``` and check in ```/var/log/daemon.log``` if there is any error about omr-admin ?

codinandhaulin, 2020-06-17:
```
root@405d4b8ca7:~# ps aux | grep omr
root       381  0.0  0.0   6728  3128 ?        Ss   08:34   0:01 /bin/bash /usr/local/bin/omr-service
root       531  0.2  1.1 169676 45588 ?        Ssl  08:34   0:11 python3 /usr/local/bin/omr-admin.py
root     14209  0.0  0.0   6076   824 pts/1    S+   09:47   0:00 grep om
```r

codinandhaulin, 2020-06-17:
Thank you so very much for the quick reply!

Here is the output of /var/log/daemon.log (from the first mention of omr-admin to the last mention.  **Please let me know if you want the entire log.**
```Jun 17 08:34:41 405d4b8ca7 systemd[1]: Started OMR-Admin.
Jun 17 08:34:41 405d4b8ca7 chronyd[390]: chronyd version 3.4 starting (+CMDMON +NTP +REFCLOCK +RTC +PRIVDROP +SCFILTER +SIGND +ASYNCDNS +SECHASH +IPV6 -DEBUG)
Jun 17 08:34:41 405d4b8ca7 omr-6in4-run[383]: usage: omr-6in4-run FILE
Jun 17 08:34:41 405d4b8ca7 chronyd[390]: Frequency 16.253 +/- 1.418 ppm read from /var/lib/chrony/chrony.drift
Jun 17 08:34:41 405d4b8ca7 chronyd[390]: Loaded seccomp filter
Jun 17 08:34:41 405d4b8ca7 systemd[1]: Starting Glorytun UDP on tun0...
Jun 17 08:34:41 405d4b8ca7 systemd[1]: Starting Glorytun TCP on tun0...
Jun 17 08:34:41 405d4b8ca7 systemd[1]: Started Shadowsocks-Libev Custom Manager Service for manager.
Jun 17 08:34:41 405d4b8ca7 systemd-udevd[238]: link_config: autonegotiation is unset or enabled, the speed and duplex are not writable.
Jun 17 08:34:41 405d4b8ca7 ss-manager[400]:  2020-06-17 08:34:41 INFO: using the default manager address: 127.0.0.1:8839
Jun 17 08:34:41 405d4b8ca7 ss-manager[400]:  2020-06-17 08:34:41 INFO: using tcp fast open
Jun 17 08:34:41 405d4b8ca7 ss-manager[400]:  2020-06-17 08:34:41 INFO: using tcp no-delay
Jun 17 08:34:41 405d4b8ca7 ss-manager[400]:  2020-06-17 08:34:41 INFO: using MPTCP
Jun 17 08:34:41 405d4b8ca7 ss-manager[400]:  2020-06-17 08:34:41 INFO: running from root user
Jun 17 08:34:41 405d4b8ca7 ss-manager[400]:  2020-06-17 08:34:41 INFO: working directory points to /root/.shadowsocks
Jun 17 08:34:41 405d4b8ca7 ss-manager[400]:  2020-06-17 08:34:41 INFO: try to bind interface: ::0, port: 65101
Jun 17 08:34:41 405d4b8ca7 ss-manager[400]:  2020-06-17 08:34:41 INFO: try to bind interface: 0.0.0.0, port: 65101
Jun 17 08:34:41 405d4b8ca7 ss-manager[400]:  2020-06-17 08:34:41 INFO: initializing acl...
Jun 17 08:34:41 405d4b8ca7 ss-manager[400]:  2020-06-17 08:34:41 INFO: enable TCP no-delay
Jun 17 08:34:41 405d4b8ca7 ss-manager[400]:  2020-06-17 08:34:41 INFO: enable multipath TCP
Jun 17 08:34:41 405d4b8ca7 systemd-udevd[249]: link_config: autonegotiation is unset or enabled, the speed and duplex are not writable.
Jun 17 08:34:41 405d4b8ca7 glorytun-udp-run[393]: running on device gt-udp-tun0 as pid 393
Jun 17 08:34:41 405d4b8ca7 post.sh[395]: Cannot find device "gt-udp-tun0"
Jun 17 08:34:41 405d4b8ca7 ss-manager[400]:  2020-06-17 08:34:41 INFO: try to bind interface: ::0, port: 65101
Jun 17 08:34:41 405d4b8ca7 ss-manager[400]:  2020-06-17 08:34:41 ERROR: bind: Address already in use
Jun 17 08:34:41 405d4b8ca7 ss-manager[400]:  2020-06-17 08:34:41 ERROR: Could not bind
Jun 17 08:34:41 405d4b8ca7 ss-manager[400]:  2020-06-17 08:34:41 ERROR: port is not available, please check.
Jun 17 08:34:41 405d4b8ca7 systemd-networkd[234]: gt-tun0: Gained carrier
Jun 17 08:34:41 405d4b8ca7 ss-server[424]: resolving hostname to IPv6 address first
Jun 17 08:34:41 405d4b8ca7 systemd-networkd[234]: gt-tun0: Gained IPv6LL
Jun 17 08:34:41 405d4b8ca7 ss-server[424]: using tcp fast open
Jun 17 08:34:41 405d4b8ca7 systemd[1]: Starting OpenVPN connection to tun0...
Jun 17 08:34:41 405d4b8ca7 ss-server[424]: UDP relay enabled
Jun 17 08:34:41 405d4b8ca7 systemd[1]: Started OpenVPN connection to tun1.
Jun 17 08:34:41 405d4b8ca7 ss-server[424]: enable TCP no-delay
Jun 17 08:34:41 405d4b8ca7 systemd[1]: Started OpenVPN service.
Jun 17 08:34:41 405d4b8ca7 ss-server[424]: initializing ciphers... chacha20-ietf-poly1305
Jun 17 08:34:41 405d4b8ca7 systemd[1]: omr6in4@user0.service: Main process exited, code=exited, status=1/FAILURE
Jun 17 08:34:41 405d4b8ca7 ss-server[424]: tcp server listening at [::0]:65101
Jun 17 08:34:41 405d4b8ca7 systemd[1]: omr6in4@user0.service: Failed with result 'exit-code'.
Jun 17 08:34:41 405d4b8ca7 ss-server[424]: tcp port reuse enabled
Jun 17 08:34:41 405d4b8ca7 systemd[1]: Failed to start OMR6IN4 on user0.
Jun 17 08:34:41 405d4b8ca7 ss-server[424]: tcp server listening at 0.0.0.0:65101
Jun 17 08:34:41 405d4b8ca7 systemd[1]: Started Glorytun UDP on tun0.
Jun 17 08:34:41 405d4b8ca7 ss-server[424]: tcp port reuse enabled
Jun 17 08:34:41 405d4b8ca7 ss-server[424]: udp server listening at [::0]:65101
Jun 17 08:34:41 405d4b8ca7 ss-server[424]: udp port reuse enabled
Jun 17 08:34:41 405d4b8ca7 systemd[1]: Started Glorytun TCP on tun0.
Jun 17 08:34:41 405d4b8ca7 ss-server[424]: udp server listening at 0.0.0.0:65101
Jun 17 08:34:41 405d4b8ca7 ss-server[424]: udp port reuse enabled
Jun 17 08:34:41 405d4b8ca7 ss-server[424]: running from root user
Jun 17 08:34:41 405d4b8ca7 ovpn-tun1[384]: OpenVPN 2.4.7 x86_64-pc-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [PKCS11] [MH/PKTINFO] [AEAD] built on Feb 20 2019
Jun 17 08:34:41 405d4b8ca7 ovpn-tun1[384]: library versions: OpenSSL 1.1.1d  10 Sep 2019, LZO 2.10
Jun 17 08:34:41 405d4b8ca7 sshd[377]: Could not load host key: /etc/ssh/ssh_host_ecdsa_key
Jun 17 08:34:41 405d4b8ca7 ovpn-tun1[384]: Diffie-Hellman initialized with 2048 bit key
Jun 17 08:34:41 405d4b8ca7 sshd[377]: Could not load host key: /etc/ssh/ssh_host_ed25519_key
Jun 17 08:34:41 405d4b8ca7 ovpn-tun0[414]: OpenVPN 2.4.7 x86_64-pc-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [PKCS11] [MH/PKTINFO] [AEAD] built on Feb 20 2019
Jun 17 08:34:41 405d4b8ca7 ovpn-tun0[414]: library versions: OpenSSL 1.1.1d  10 Sep 2019, LZO 2.10
[... cut ...]
Jun 17 08:34:41 405d4b8ca7 systemd-networkd[234]: gt-udp-tun0: Gained carrier
Jun 17 08:34:41 405d4b8ca7 systemd-networkd[234]: gt-udp-tun0: Gained IPv6LL
Jun 17 08:34:41 405d4b8ca7 systemd[1]: Stopping OMR-Admin...
Jun 17 08:34:41 405d4b8ca7 systemd[1]: omr-admin.service: Main process exited, code=killed, status=15/TERM
Jun 17 08:34:41 405d4b8ca7 systemd[1]: omr-admin.service: Succeeded.
Jun 17 08:34:41 405d4b8ca7 systemd[1]: Stopped OMR-Admin.
Jun 17 08:34:41 405d4b8ca7 systemd[1]: Started OMR-Admin.```

Ysurac, 2020-06-17:
There is no error in the log.
What it the result, from the VPS, of ```curl -k https://127.0.0.1:65500/``` ? If you get a result then the API is working and you should check the key on the router.
You can also try same check from the router by replacing 127.0.0.1 with VPS IP. If you get no result, I will need ```ip r``` result from the router.

codinandhaulin, 2020-06-17:
On VPS:
```
root@405d4b8ca7:~# curl -k https://127.0.0.1:65500/
"Welcome to OpenMPTCProuter Server part"root@405d4b8ca7:~#
```

On router:
```
root@OpenMPTCProuter:~# curl -k https://107.152.38.243:65500/
curl: (28) Failed to connect to 107.152.38.243 port 65500: Operation timed out
root@OpenMPTCProuter:~#
```

```
root@OpenMPTCProuter:~# ip r
default via 192.168.14.1 dev eth2
default via 192.168.10.1 dev eth1 metric 3
default via 192.168.14.1 dev eth2 metric 4
10.255.255.1 dev tun0 proto kernel scope link src 10.255.255.2
107.152.38.243 via 192.168.10.1 dev eth1 metric 3
107.152.38.243 via 192.168.14.1 dev eth2 metric 4
127.0.0.0/8 dev lo proto static scope link metric 1
192.168.10.0/24 dev eth1 scope link metric 3
192.168.14.0/24 dev eth2 scope link metric 4
207.148.9.140 via 192.168.14.1 dev eth2 metric 1
207.148.9.140 via 192.168.10.1 dev eth1 metric 3
207.148.9.140 via 192.168.14.1 dev eth2 metric 4
root@OpenMPTCProuter:~#
```

codinandhaulin, 2020-06-17:
I also noticed this in the daemon.log:
```
Jun 17 08:34:41 405d4b8ca7 post.sh[395]: Cannot find device "gt-udp-tun0"
Jun 17 08:34:41 405d4b8ca7 ss-manager[400]:  2020-06-17 08:34:41 INFO: try to bind interface: ::0, port: 65101
Jun 17 08:34:41 405d4b8ca7 ss-manager[400]:  2020-06-17 08:34:41 ERROR: bind: Address already in use
Jun 17 08:34:41 405d4b8ca7 ss-manager[400]:  2020-06-17 08:34:41 ERROR: Could not bind
Jun 17 08:34:41 405d4b8ca7 ss-manager[400]:  2020-06-17 08:34:41 ERROR: port is not available, please check
```

kevin39, 2020-06-17:
Hide all your IPs.

codinandhaulin, 2020-06-17:
VPS ports:
```
root@405d4b8ca7:~# lsof -i -P -n | grep LISTEN
iperf3      303            root    3u  IPv6   2878      0t0  TCP *:65400 (LISTEN)
glorytun-   396            root    4u  IPv4   5628      0t0  TCP *:65001 (LISTEN)
openvpn     414          nobody    6u  IPv4   7837      0t0  TCP *:65301 (LISTEN)
ss-server   424            root    6u  IPv6   5715      0t0  TCP *:65101 (LISTEN)
ss-server   424            root    7u  IPv4   5717      0t0  TCP *:65101 (LISTEN)
sshd        452            root    3u  IPv4   7840      0t0  TCP *:65222 (LISTEN)
sshd        452            root    4u  IPv6   7842      0t0  TCP *:65222 (LISTEN)
python3     531            root   12u  IPv4   9289      0t0  TCP *:65500 (LISTEN)
dsvpn      1064            root    4u  IPv4   8869      0t0  TCP *:65401 (LISTEN)
```

Ysurac, 2020-06-17:
It's working from the VPS. Do you have any result with ```curl http://www.google.com/``` ?
If yes check that there is not additional firewall on VPS host.

codinandhaulin, 2020-06-17:
Yes, I get results from  `curl http://www.google.com/` on both VPS and router.

Ysurac, 2020-06-17:
check that there is no additional firewall on VPS host, and check that interface in ```/etc/shorewall/params.net``` is the correct public interface.

Ysurac, 2020-06-17:
I checked with the curl to the VPS from my IP and it's working. Strange...
Can you ping the VPS from the router part ?

codinandhaulin, 2020-06-17:
```
root@405d4b8ca7:~# cat /etc/shorewall/params.net
NET_IFACE=eth0root@405d4b8ca7:~#
```

codinandhaulin, 2020-06-17:
Yes, i can ping VPS from the router.

codinandhaulin, 2020-06-17:
would it help if I PMd you the api key?  I'll give you root access if you care to as well.

codinandhaulin, 2020-06-17:
VPS hosting support says not ports blocked, no firewall either.

Ysurac, 2020-06-17:
I don't need the API key or root access, I have a result with ```curl -k https://107.152.x.x:65500/```.
What is the result from the router of ```curl --interface eth1 -k https://107.152.x.x:65500/``` and ``````curl --interface eth2 -k https://107.152.x.x:65500/``` ?

codinandhaulin, 2020-06-17:
> 
> 
> I have a result with `curl -k https://107.152.x.x:65500/`.
> What is the result from the router of `curl --interface eth1 -k https://107.152.x.x:65500/` and `curl --interface eth2 -k https://107.152.x.x:65500/` ?

Both interfaces:
`curl: (28) Failed to connect to 107.152.x.x port 65500: Operation timed out`

Ysurac, 2020-06-17:
Strange that it's with both wan, I think replacing with https://www.google.com/ is working ?

codinandhaulin, 2020-06-17:
> 
> Strange that it's with both wan, I think replacing with https://www.google.com/ is working ?

correct, www.google.com works on both interfaces

codinandhaulin, 2020-06-17:
I also tethered an iphone (eth3) and curl fails.  Of course, the iphone is AT&T service (same as hotspot device on eth2).  I will try to tether another phone with Verizon service.

Ysurac, 2020-06-17:
Fail2ban is not installed on the VPS ? If it's the case it may have banned your IPs.

codinandhaulin, 2020-06-17:
Fail2ban not installed - they claim they have not banned any IPs

codinandhaulin, 2020-06-17:
Verizon iphone (eth3) cannot curl to VPS on port 65500

codinandhaulin, 2020-06-17:
could something be blocked by my mobile ISPs?

Ysurac, 2020-06-17:
This seems strange that all ISPs block access to this server. And ping seems to work...
Did you try to reboot router ? And valid again wizard ? (even if I this all this will change nothing...)
What is the result if you only keep one connection ?

codinandhaulin, 2020-06-17:
Rebooted, disabled all but one WAN interface.  same result - cannot curl.  Seems like the entire world can access that port except me.  VPS host is going to assign another IP address.

codinandhaulin, 2020-06-17:
With new IP address on VPS, same results

Ysurac, 2020-06-17:
And if you try from a fresh router install ?
Or what is the result if you try to connect to https://107.152.x.x:65500/ from a computer connected directly to a modem ?

codinandhaulin, 2020-06-17:
Yes, I can curl and https://:65500 through a browser when directly using the modem

codinandhaulin, 2020-06-17:
and success with both by a machine connected to the router???

Ysurac, 2020-06-17:
ok. I think the problem is related to a rule that redirect all TCP traffic to shadowsocks, but as shadowsocks is not running, nothing work. I solved it for next release I think.
You can try on the router: ```/etc/init.d/shadowsocks-libev rules_down```

codinandhaulin, 2020-06-17:
> You can try on the router: `/etc/init.d/shadowsocks-libev rules_down`
no change

Ysurac, 2020-06-17:
ok so no idea.
You can try to do a fresh router install.

codinandhaulin, 2020-06-17:
Just did a fresh router install, no improvement.  rebooted VPS.  same result.

Ysurac, 2020-06-17:
I will try to rent a VPS from servercheap and test tomorrow.

codinandhaulin, 2020-06-17:
Thank you very much for your help today.  My offer stands for root access.

Ysurac, 2020-06-18:
I rented a VPS from servercheap and get same result as you.
With MPTCP enabled I can't reach the VPS over TCP, with MPTCP disabled this is working.
For now I have no success with MPTCP enabled... I'm still testing.

Ysurac, 2020-06-18:
This seems to be working with snapshot VPS script and latest OpenMPTCProuter 0.55beta11.

codinandhaulin, 2020-06-18:
I am rebuilding the VPS.  Is this how to run the "snapshot VPS script"?
```
wget -O - https://www.openmptcprouter.com/server-test/debian10-x86_64.sh | sh
```

Ysurac, 2020-06-18:
All the info are in https://github.com/Ysurac/openmptcprouter/wiki/Snapshots and the beta in https://github.com/Ysurac/openmptcprouter/issues/959

But as it's snapshot/beta, don't expect any support on this. It's not stable.

codinandhaulin, 2020-06-30:
I have tried two VPS other than Vultr and cannot get any combination of router/vps script to work on either.  I am using fresh router imaages and fresh Debian server images on the VPS.  Is there any known working VPS in the US other than Vultr?  I really don't understand what I could be doing wrong.

Ysurac, 2020-06-30:
Vultr is a tested VPS provider. If VPS & images of Debian 10 are the same in US and Europe (I think it's the case), then this should work (or at least it's not related to VPS and router but can be related to connections).

Ysurac, 2020-06-30:
Google Cloud Platform (GCP) and DigitalOcean are also tested and work well, I think they are also available in US.

closed 2020-11-03 by github-actions[bot]