Ysurac/openmptcprouter #1075: Problem forwarding port 9100

Corrado56, 2020-06-18:
Hi, I have a problem forwarding port 9100 (remote printer)

## Expected Behavior
The printer 192.168.1.17 should print

## Current Behavior
The printer 192.168.1.17 doesn't print

## Steps to Reproduce the Problem
see below the context

  1. Setup a port forward rule to an IP Printer
  2. Define the gateway of the IP printer = MPTCP Router
  3. Define an IP printer on local computer (W10)
  4. Send a test page to the IP printer

## Context (Environment)
Open MPTCP router is 192.168.1.20
the printer is 192.168.1.17 mask 255.255.255.0 gw 192.168.1.20
"Redirect all ports from server to this router" is checked
Port forwarding from "VPN" to "this device" port 9100 to lan IP 192.168.1.17 port 9100 enabled (tcp and udp)
on local computer defined a IP printer with VPS ip and port 9100 raw.
When I sent a test page, the printer queue opens and I could see the document in the queue, but the printer remains silent and after some seconds the document disappears without errors. (where is going the stream ? )

192.168.1.4 is an Asus Router (wan1) where a have the same vpn forwarding rule
If I configure the printer with gw 192.168.1.4 and send the test page to the Asus router IP, the printer is printing.

## Specifications
  - OpenMPTCProuter version: 0.54.7
  - OpenMPTCProuter VPS version: Debian 10 x64
  - OpenMPTCProuter platform: Raspberry PI4B

Ysurac, 2020-06-18:
You don't have to use "Redirect all ports from server to this router", to redirect only one port. Only using VPN as source zone open the specified port.
For 9100, you only need to it as TCP.
Did you try a SNAT ? (for info: https://github.com/Ysurac/openmptcprouter/wiki/Port-forwarding )

Ysurac, 2020-06-18:
No idea. Some data seems to be sent, but not all ?

Corrado56, 2020-06-20:
I will try to use an old PC instead of a Raspberry. Need a few days for test.

>> Some data seems to be sent, but not all ?
From a local PC I see the printer queue to receive the file and expect the printer to print or generate an error. Instead after few second the queue empties without any error. I doens't understand what is happening

Corrado56, 2020-07-02:
I was able to solve the problem on PI4B with the following steps:
0) Port forwarding port 9100 from public VPS IP to the local printer (see above)
1) I buy two adapters USB3 to RJ45 to have on eth0 only the bonded connection (only one IP linked to the physical port), and on eth1 and eth2 the two wan connections not bonded. After that the test page sent to the printer queue was no more vanishing after few second, but stay indefinitely, unfortunately the printer stay silent.
2) checked again "Redirect all ports from server to this router", and the printer starts printing (slowly)
I am not enough technically skilled to understand those result. This is the configuration is working for me and I found experimentally.

closed 2020-07-02 by Corrado56