Ysurac/openmptcprouter #1076: Fix for long-lived VPN connections to be fully aggregated at all times

uryupinsk, 2020-06-19:
## Expected Behavior

Long-lived connections such as VPN site-to-site or SSH sessions must be aggregated above all links at all times.

## Current Behavior

With the default OMR and Linux network stack settings, long-lived connections are sporadically not fully aggregated.

## Possible Solution

Some posts on MPTCP Github issues and the [Red Hat Knowledge Base](https://access.redhat.com/solutions/726753) suggest that the `/proc/sys/net/ipv4/tcp_retries2` default value is too high. This setting controls how many retries TCP makes on data segments.

Red Hat suggests that in a High Availability (HA) situation this settings is best decreased to a value of 3.

## Steps 

  1. SSH to OMR
  2. `echo "net.ipv4.tcp_retries2=3 > /etc/systctl.conf`
  3. Go to `Network > MPTCP` and increase `Multipath TCP SYN retries` to 3
  4. Reboot OMR

## Context (Environment)

After I changed this settings for a few days, on a Wireguard site-to-site VPN, all links are fully used at all times. 

I did not encounter any issues yet with my daily usage.

## Specifications

  - OpenMPTCProuter version: 0.54.7
  - OpenMPTCProuter VPS version: 0.1015
  - OpenMPTCProuter platform: x86_64

Ysurac, 2020-06-19:
Thanks I will test that and, if there is no issues, this will be in next release.

hewenhao2008, 2020-06-28:
@uryupinsk "After I changed these settings for a few days, on a Wireguard site-to-site VPN, all links are fully used at all times"
Are you sure the Wireguard VPN works for all links? I think wireguard works on udp only, and will not bonding all links. How do you make wireguard bonding all links?

uryupinsk, 2020-06-30:
@hewenhao2008 UDP aggregation is done with glorytun-tcp by default on OMR. There is also the newer DSVPN available but it is buggy on 0.54.7. I will try it again on the next OMR version.

uryupinsk, 2020-08-11:
@Ysurac Have you included this fix in OMR 0.55?

Ysurac, 2020-08-11:
Yes it's available in 0.55

uryupinsk, 2020-08-11:
Excellent, thanks a lot.

closed 2020-08-11 by uryupinsk