Ysurac/openmptcprouter #1077: I have a problem installing the package on OVH

NoobLinux, 2020-06-22:
Bonjour,

j'ai un soucis pour installer le package  sur OVH  👍 

Ign:1 http://debian.mirrors.ovh.net/debian stretch InRelease
Get:2 http://debian.mirrors.ovh.net/debian stretch-updates InRelease [93.6 kB]
Hit:3 http://security.debian.org stretch/updates InRelease
Get:4 http://debian.mirrors.ovh.net/debian stretch Release [118 kB]
Reading package lists... Done
E: Release file for http://debian.mirrors.ovh.net/debian/dists/stretch-updates/InRelease is expired (invalid since 7d 2h 27min 30s). Updates for this repository will not be applied.

pouvez-vous m'aider ? merci

Hello ! 

I have a problem installing the package on OVH 

Ign:1 http://debian.mirrors.ovh.net/debian stretch InRelease
Get:2 http://debian.mirrors.ovh.net/debian stretch-updates InRelease [93.6 kB]
Hit:3 http://security.debian.org stretch/updates InRelease
Get:4 http://debian.mirrors.ovh.net/debian stretch Release [118 kB]
Reading package lists... Done
E: Release file for http://debian.mirrors.ovh.net/debian/dists/stretch-updates/InRelease is expired (invalid since 7d 2h 27min 30s). Updates for this repository will not be applied.

i need help 

good Days 

Cordialement

Ysurac, 2020-06-22:
There is an invalid on the server or OVH Debian mirror is not up to date.
You can use another mirror in Debian, or use Debian 10 directly if it's a new install.

NoobLinux, 2020-06-22:
I only have debian 9 on my vps, I need to find a solution but I do not master Linux

NoobLinux, 2020-06-22:
I come back to what I said, I have access to debian 10.

I will reinstall the server thank you

NoobLinux, 2020-06-22:
Ysurac thx godd day

closed 2020-06-22 by NoobLinux