Ysurac/openmptcprouter #1079: wan/eth interfaces mixup upon reboot when usi USB2Ethernet adapters

vistman, 2020-06-25:
## Expected Behavior
wan/eth binding to stay intact upon reboot.

## Current Behavior
I'm using USB2Ethernet adapters with OpenMPTCProuter and so far I had 2 of them. Recently I've added the the 3rd USB2Ethernet and another LTE router, so now I have 3 USB2Ethernet adapters connected.
After every reboot (and after 2 days brainstorming and trying out every possible combination) I've found out my wan2/eth2 and wan3/eth3 bindings got mixed up so I have to manually change wan2 to eth3 and wan3 to eth2 ...

## Possible Solution
Don't know. Some sort of startup wan/eth check

## Steps to Reproduce the Problem

  1. Connect 3 USB2Ethernet adapters and reboot OpenMPTCProuter

## Context (Environment)
Upon every reboot on wan2 and wan3 I get errors reporting GATEWAY down ...

## Specifications

  - OpenMPTCProuter version:  v0.54.7
  - OpenMPTCProuter VPS version: 0.1015 4.19.104
  - OpenMPTCProuter platform: RPI4b/x86_64)

Ysurac, 2020-06-25:
Why are you using USB2Ethernet adapters ?
I just ordered 2 adapters and I will make test on how I can fix them on interface easily.

vistman, 2020-06-25:
Well, I'm experimenting - getting into networking and have upgraded my network equippment to managed routers/switches. Along with that I'm testing various LTE providers/routers/antennas to get optimal Internet access solution. Major advantage of USB2Eth adapters is when you have them connected to OpenMPTCProuter they provide direct access to connected xDSL/LTE routers. If I connect OpenMPTCProuter and xDSL/LTE routers via switch I need to do subnetting and DNS servers on my main router which is not very suitable in testing phase ... Someday when I get everything to work stable on OpenMPTCHrouter side  and get bored I'll probably connect them via switch ...

vistman, 2020-06-25:
Just an update ... all wans and eths are affected by this mixup upon reboot. I've tried sorting things out by manually reassigning wans and eths under Interfaces but it just didn't seem to work even I've tried matching them by MAC addresses. At the end only thing that helped is deleting all wans and reconnecting USB2Eth adapters one by one ...

Ysurac, 2020-07-01:
I just received my adapters, I begin to test and check how I can solve this problem.

Ysurac, 2020-07-02:
A solution (I hope) will be available in next release.

vistman, 2020-07-02:
Great, thanks!

asla2000, 2020-07-17:
I have the same issue with 4 x USBeth (USB LTE)
Works fine only when turning them on one by one in a proper order. If a reboot occurs (or if they are turned on in wrong order), all wans are in Geteway down state.

> A solution (I hope) will be available in next release.
Ysurac, could you please tell the expected next release date? THX!

asla2000, 2020-07-17:
Is there any temp solution before the next release arrives?

asla2000, 2020-07-17:
### **UPD: Solution!**  @vistman @Ysurac 

1. Set all USB eths (USB LTE) in DHCP mode but **different subnets**
2. Clear all wans in OMR, reboot
3. Connect 1 USB LTE (no matter wich one)
4. Wizard -> Select new eth1 -> Add interface -> Change protocol from Static to **OTHER**. Save & Apply.
5. Network -> Interfaces -> WAN1 -> Edit -> Protocol DHCP client, check **"Bring up on boot"**. Save.
6. Apply all changes. Now 1 USB LTE is up and using multipath as expected.
7. Repeat steps 3-6 for all other modems
8. Reboot
9. Check Status, Enjoy! )

closed 2020-11-19 by github-actions[bot]