Ysurac/openmptcprouter #1080: Can't get public IP address from ShadowSocks

ccmks, 2020-06-26:
## Expected Behavior
To have the OMR function properly with ShadowSocks

## Current Behavior
Can't get public IP address from ShadowSocks
[image]

The only workaround is to reboot VPS and everything is working again for around 20 mins or so then it will behave same thing.

## Possible Solution
Looking for advise

## Steps to Reproduce the Problem

 Install OpenMPTCPRouter on x86 based system
Install OpenMPTCPRouter on VPS Ubuntu on Vultr Singapore Region
Everything is working normally for the first 20 - 30 mins then the issue appear until VPS is being rebooted

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.54.7 r0+12509-04a21c26a0
  - OpenMPTCProuter VPS version: 0.1015 4.19.104-mptcp
  - OpenMPTCProuter platform: x86_64

ccmks, 2020-06-26:
Here is the system log

Fri Jun 26 15:53:42 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:53:46 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:53:51 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:53:56 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:54:01 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:54:02 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:54:06 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:54:11 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:54:16 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:54:21 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:54:22 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:54:26 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:54:31 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:54:36 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:54:41 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:54:42 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:54:46 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:54:51 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:54:57 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:55:02 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:55:02 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:55:07 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:55:12 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:55:17 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:55:22 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:55:22 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:55:27 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:55:32 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:55:37 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:55:42 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:55:42 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:55:47 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:55:52 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:55:57 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:56:02 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:56:02 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:56:07 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:56:12 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:56:17 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:56:22 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:56:22 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:56:27 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:56:32 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:56:37 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:56:42 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:56:43 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:56:48 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:56:53 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:56:58 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:57:02 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:57:03 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:57:08 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:57:13 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:57:18 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:57:22 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:57:23 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 15:57:28 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
[... cut ...]
Fri Jun 26 16:39:10 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 16:39:13 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 16:39:15 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 16:39:20 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 16:39:22 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected
Fri Jun 26 16:39:25 2020 daemon.err /usr/bin/ss-local[2324]: getpeername: Socket not connected

ccmks, 2020-06-26:
Here is my setting

dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].boguspriv='1'
dhcp.@dnsmasq[0].filterwin2k='0'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].resolvfile='/tmp/resolv.conf.d/resolv.conf.auto'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].server='127.0.0.1#5353' '/use-application-dns.net/'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.force='1'
dhcp.lan.ra_slaac='1'
dhcp.lan.ra_flags='managed-config' 'other-config'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].RootPasswordAuth='on'
dropbear.@dropbear[0].Port='22'
dscp.@classify[0]=classify
dscp.@classify[0].direction='both'
dscp.@classify[0].proto='icmp'
dscp.@classify[0].class='cs5'
dscp.@classify[0].comment='ICMP'
dscp.@classify[1]=classify
dscp.@classify[1].direction='both'
dscp.@classify[1].proto='udp'
dscp.@classify[1].class='cs5'
dscp.@classify[1].src_port='53,123,5353'
dscp.@classify[1].comment='DNS udp and NTP'
dscp.@classify[2]=classify
dscp.@classify[2].direction='both'
dscp.@classify[2].proto='tcp'
dscp.@classify[2].class='cs5'
dscp.@classify[2].src_port='53,5353'
dscp.@classify[2].comment='DNS tcp'
dscp.@classify[3]=classify
dscp.@classify[3].direction='both'
dscp.@classify[3].proto='tcp'
dscp.@classify[3].class='cs6'
dscp.@classify[3].dest_port='65001,65301,65011'
dscp.@classify[3].comment='OMR vpn'
dscp.@classify[4]=classify
dscp.@classify[4].direction='both'
dscp.@classify[4].proto='udp'
dscp.@classify[4].class='cs6'
[... 1375 lines cut ...]
upnpd.config.enable_natpmp='1'
upnpd.config.enable_upnp='1'
upnpd.config.secure_mode='1'
upnpd.config.log_output='0'
upnpd.config.download='1024'
upnpd.config.upload='512'
upnpd.config.internal_iface='lan'
upnpd.config.port='5000'
upnpd.config.upnp_lease_file='/var/run/miniupnpd.leases'
upnpd.config.igdv1='0'
upnpd.@perm_rule[0]=perm_rule
upnpd.@perm_rule[0].action='allow'
upnpd.@perm_rule[0].ext_ports='1024-65535'
upnpd.@perm_rule[0].int_addr='0.0.0.0/0'
upnpd.@perm_rule[0].int_ports='1024-65535'
upnpd.@perm_rule[0].comment='Allow high ports'
upnpd.@perm_rule[1]=perm_rule
upnpd.@perm_rule[1].action='deny'
upnpd.@perm_rule[1].ext_ports='0-65535'
upnpd.@perm_rule[1].int_addr='0.0.0.0/0'
upnpd.@perm_rule[1].int_ports='0-65535'
upnpd.@perm_rule[1].comment='Default deny'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='eth1' 'eth2' 'eth3' 'eth4' 'eth5'
vnstat.@vnstat[0].backup='0'

Ysurac, 2020-06-26:
Check that the master interface is the interface with the most stable connection.
You can check on the VPS the log about ss-server in ```/var/log/daemon.log```
For shadowsocks, I would need ss-redir log on the router (ss-local is only used for connection test).

In Network->MPTCP, you can test BLEST as MPTCP scheduler and increase to 3 Multipath TCP SYN retries. The cubic congestion control can also be tested.

If the problem is always after same delay, it's possible that ISP or VPS provider close some connection after a delay.

ccmks, 2020-06-27:
For shadowsocks, I would need ss-redir log on the router (ss-local is only used for connection test).

How do I generate the log from above?

ccmks, 2020-06-27:
Just want to give you update,

After I followed your advice, I have been running this for more than 6 hours and still haven't encountered any issue ever since. Will keep monitoring it.

Thank you

ccmks, 2020-06-28:
Hello, I noticed that if one of the WAN (not master) goes down, it will behave same issue.

I tried to open the log on the VPS per your instruction "You can check on the VPS the log about ss-server in /var/log/daemon.log" and couldn't find the file listed in my VPS.

Also for this one "For shadowsocks, I would need ss-redir log on the router (ss-local is only used for connection test)." How do I do it?

Thank you

Ysurac, 2020-06-28:
If you use Debian, it's in /var/log/daemon.log on the VPS else in /var/log directory.
For ss-redir log, it's in same State->System Log, before log about ss-local. Log file size is limited by default, so old log are removed after some time.
In Network->MPTCP and MPTCP fullmesh tab, you have all your IPs ?

ccmks, 2020-06-28:
VPS else in /var/log directory
What is the name of the log file?

ccmks, 2020-06-28:
[image]

Ysurac, 2020-06-28:
I don't remember on Ubuntu, maybe in /var/log/messages.

ccmks, 2020-06-28:
This is the one from ubuntu

[image]

Ysurac, 2020-06-28:
So it's in syslog

ccmks, 2020-06-29:
[syslog.txt](https://github.com/Ysurac/openmptcprouter/files/4843669/syslog.txt)

Here is the attached syslog

ccmks, 2020-07-02:
Here is the new system log from OMR when the Shadow socks can't get public IP address again. I noticed that when I am downloading with full speed, the behavior happened again

Thu Jul  2 10:59:47 2020 user.notice post-tracking-post-tracking: Get status and settings for wan4... Failed
Thu Jul  2 11:00:01 2020 user.notice post-tracking-post-tracking: eth4 switched off because ping from 192.168.0.30 error (1.0.0.1,80.67.169.12)
Thu Jul  2 11:00:01 2020 user.notice post-tracking-post-tracking: Delete default route via 192.168.0.1 dev eth4
Thu Jul  2 11:00:09 2020 daemon.err /usr/bin/ss-redir[17413]: server recv: Host is unreachable
Thu Jul  2 11:00:39 2020 user.notice post-tracking-post-tracking: Set server vps (139.180.220.146) route via 192.168.0.1 metric 7
Thu Jul  2 11:00:39 2020 user.notice post-tracking-post-tracking: Get status and settings for wan4...
Thu Jul  2 11:00:42 2020 user.notice post-tracking-post-tracking: Get status and settings for wan4... Failed
Thu Jul  2 11:00:56 2020 user.notice post-tracking-post-tracking: eth4 switched off because ping from 192.168.0.30 error (1.0.0.1,80.67.169.12)
Thu Jul  2 11:00:56 2020 user.notice post-tracking-post-tracking: Delete default route via 192.168.0.1 dev eth4
Thu Jul  2 11:01:40 2020 user.notice post-tracking-post-tracking: Set server vps (139.180.220.146) route via 192.168.0.1 metric 7
Thu Jul  2 11:01:40 2020 user.notice post-tracking-post-tracking: Get status and settings for wan4...
Thu Jul  2 11:01:43 2020 user.notice post-tracking-post-tracking: Get status and settings for wan4... Failed
Thu Jul  2 11:01:57 2020 user.notice post-tracking-post-tracking: eth4 switched off because ping from 192.168.0.30 error (1.0.0.1,80.67.169.12)
Thu Jul  2 11:01:57 2020 user.notice post-tracking-post-tracking: Delete default route via 192.168.0.1 dev eth4
Thu Jul  2 11:02:40 2020 user.notice post-tracking-post-tracking: Set server vps (139.180.220.146) route via 192.168.0.1 metric 7
Thu Jul  2 11:02:40 2020 user.notice post-tracking-post-tracking: Get status and settings for wan4...
Thu Jul  2 11:02:43 2020 user.notice post-tracking-post-tracking: Get status and settings for wan4... Failed
Thu Jul  2 11:02:57 2020 user.notice post-tracking-post-tracking: eth4 switched off because ping from 192.168.0.30 error (1.0.0.1,80.67.169.12)
Thu Jul  2 11:02:57 2020 user.notice post-tracking-post-tracking: Delete default route via 192.168.0.1 dev eth4
Thu Jul  2 11:03:38 2020 user.notice post-tracking-post-tracking: Set server vps (139.180.220.146) route via 192.168.0.1 metric 7
Thu Jul  2 11:03:38 2020 user.notice post-tracking-post-tracking: Get status and settings for wan4...
Thu Jul  2 11:03:41 2020 user.notice post-tracking-post-tracking: Get status and settings for wan4... Failed
Thu Jul  2 11:03:55 2020 user.notice post-tracking-post-tracking: eth4 switched off because ping from 192.168.0.30 error (1.0.0.1,80.67.169.12)
Thu Jul  2 11:03:55 2020 user.notice post-tracking-post-tracking: Delete default route via 192.168.0.1 dev eth4
Thu Jul  2 11:04:35 2020 user.notice post-tracking-post-tracking: Set server vps (139.180.220.146) route via 192.168.0.1 metric 7
Thu Jul  2 11:04:35 2020 user.notice post-tracking-post-tracking: Get status and settings for wan4...
Thu Jul  2 11:04:38 2020 user.notice post-tracking-post-tracking: Get status and settings for wan4... Failed
Thu Jul  2 11:04:52 2020 user.notice post-tracking-post-tracking: eth4 switched off because ping from 192.168.0.30 error (1.0.0.1,80.67.169.12)
Thu Jul  2 11:04:52 2020 user.notice post-tracking-post-tracking: Delete default route via 192.168.0.1 dev eth4
Thu Jul  2 11:05:30 2020 user.notice post-tracking-post-tracking: Set server vps (139.180.220.146) route via 192.168.0.1 metric 7
Thu Jul  2 11:05:30 2020 user.notice post-tracking-post-tracking: Get status and settings for wan4...
Thu Jul  2 11:05:33 2020 user.notice post-tracking-post-tracking: Get status and settings for wan4... Failed
Thu Jul  2 11:05:47 2020 user.notice post-tracking-post-tracking: eth4 switched off because ping from 192.168.0.30 error (1.0.0.1,80.67.169.12)
Thu Jul  2 11:05:47 2020 user.notice post-tracking-post-tracking: Delete default route via 192.168.0.1 dev eth4
Thu Jul  2 11:06:30 2020 user.notice post-tracking-post-tracking: Set server vps (139.180.220.146) route via 192.168.0.1 metric 7
Thu Jul  2 11:06:30 2020 user.notice post-tracking-post-tracking: Get status and settings for wan4...
Thu Jul  2 11:06:33 2020 user.notice post-tracking-post-tracking: Get status and settings for wan4... Failed
Thu Jul  2 11:06:47 2020 user.notice post-tracking-post-tracking: eth4 switched off because ping from 192.168.0.30 error (1.0.0.1,80.67.169.12)
Thu Jul  2 11:06:47 2020 user.notice post-tracking-post-tracking: Delete default route via 192.168.0.1 dev eth4
Thu Jul  2 11:07:26 2020 user.notice post-tracking-post-tracking: Set server vps (139.180.220.146) route via 192.168.0.1 metric 7
Thu Jul  2 11:07:26 2020 user.notice post-tracking-post-tracking: Get status and settings for wan4...
Thu Jul  2 11:07:29 2020 user.notice post-tracking-post-tracking: Get status and settings for wan4... Failed
Thu Jul  2 11:07:43 2020 user.notice post-tracking-post-tracking: eth4 switched off because ping from 192.168.0.30 error (1.0.0.1,80.67.169.12)
Thu Jul  2 11:07:43 2020 user.notice post-tracking-post-tracking: Delete default route via 192.168.0.1 dev eth4
[... cut ...]
Thu Jul  2 12:08:00 2020 daemon.err /usr/bin/ss-local[2590]: getpeername: Socket not connected
Thu Jul  2 12:08:02 2020 daemon.err /usr/bin/ss-local[2590]: getpeername: Socket not connected
Thu Jul  2 12:08:05 2020 daemon.err /usr/bin/ss-local[2590]: getpeername: Socket not connected
Thu Jul  2 12:08:10 2020 daemon.err /usr/bin/ss-local[2590]: getpeername: Socket not connected
Thu Jul  2 12:08:14 2020 daemon.err /usr/bin/ss-local[2590]: getpeername: Socket not connected
Thu Jul  2 12:08:15 2020 daemon.err /usr/bin/ss-local[2590]: getpeername: Socket not connected

Ysurac, 2020-07-02:
This may be a problem with a connection. Did you try to remove wan after wan ?

ccmks, 2020-07-02:
No, I only replace one of the WAN since the WAN5 previously was no longer in use. I changed to new one

ccmks, 2020-07-02:
Just to give you the update. This behavior always happened when I am using Internet Download Manager to test the aggregation. The test was only last for several seconds then I got the error that can't get public IP from shadowsocks. When that happened, I have to reboot the VPS to get everything normal again.

ccmks, 2020-07-06:
I changed my VPS to Amazon lightsail and the problem still happened when we are reaching maximum speed for more than 30 seconds. That would be really hassle to keep rebooting VPS when the problem happened. Is there any way we can speed up the Glorytun VPN to have same level of speed with ShadowSocks?

Ysurac, 2020-07-06:
It's not possible to get same speed as shadowsocks with a VPN (else I would not use shadowsocks).
You can enable SQM in the wizard, it's made to solve this problem.
You can also try latest 0.55 beta if you want.

ccmks, 2020-07-06:
Do I need to enable SQM on each my WAN interface? Just enable it, no need to setup download and upload speed right?

Also where can I download the latest 0.55 beta?

Thank you

Ysurac, 2020-07-06:
This should be enabled on each WAN and speeds should be set when possible but you can try with and without.

0.55 beta is pinned in issues, so it's https://github.com/Ysurac/openmptcprouter/issues/959

ccmks, 2020-07-06:
Tried with enable SQM, doesn't improve much with or without speed being set

Ysurac, 2020-07-06:
This should. At least if VPS speed is higher than aggregated speed.

vistman, 2020-07-06:
> 
> 
> This should be enabled on each WAN and speeds should be set when possible but you can try with and without.
> 
> 0.55 beta is pinned in issues, so it's #959

This might be a stupid question, but just to make sure - shout SQM be enabled on LTE, even it says "You should disable SQM for LTE or any interfaces with variable speed."?

ccmks, 2020-07-07:
VPS is around 1 Gbps, so my aggregate speed is less then my VPS speed

ccmks, 2020-07-10:
Just to give you update, I am no longer experiencing the issue after I moved my VPS to Linode. I am not sure why it happened on Vultr and Amazon Lightsail? However, I will keep monitoring it on how it goes.

Thank you

closed 2020-11-13 by github-actions[bot]