Ysurac/openmptcprouter #1081: Slow/impossible Samba communication on Pi4

robertoglxda, 2020-06-26:
## Expected Behavior
Transferring file with Samba should be possible and it should work at a relativity good speed (assuming a good ethernet cable and a good USB3 hard drive).
## Current Behavior
Using Samba is almost impossible.
Every (big) file I try to copy from a Windows PC, it starts with a great speed (~112MB/s) then is decreases to 0MB/s after 3 or 4 seconds. After that, the pi4 appears offline (no ping either) for some time (a couple of minutes) and I can't copy the file anymore. Note that I can run iperf for hours without issues.

## Possible Solution
No solution yet.

## Steps to Reproduce the Problem

  1. Install Samba4 (System/Software on the OMR Web UI)
  2. Copy a file from a machine on the same local network
  3. See it fails.

## Context (Environment)
I've setup on my home a pi4 (4gb ram) running OMR,
I've also installed samba on it, and I've attached a USB3 hard drive to it. These are the results of hdparm and iperf.
hdparm from the pi4 on the usb3 hard drive:
```
root@OpenMPTCProuter:/# hdparm -tT --direct /dev/sdb
/dev/sdb:
 Timing O_DIRECT cached reads:   648 MB in  2.00 seconds = 323.62 MB/sec
 Timing O_DIRECT disk reads: 522 MB in  3.00 seconds = 173.83 MB/sec
```
iperf between my Windows pc and the pi4:
```
root@OpenMPTCProuter:/# iperf3 -s
-----------------------------------------------------------
Server listening on 5201
-----------------------------------------------------------
Accepted connection from 192.168.8.113, port 54966
[  5] local 192.168.8.3 port 5201 connected to 192.168.8.113 port 54967
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec   106 MBytes   890 Mbits/sec
[  5]   1.00-2.00   sec   111 MBytes   929 Mbits/sec
[  5]   2.00-3.00   sec   110 MBytes   923 Mbits/sec
[  5]   3.00-4.00   sec   111 MBytes   932 Mbits/sec
[  5]   4.00-5.00   sec   110 MBytes   925 Mbits/sec
[  5]   5.00-6.00   sec   110 MBytes   919 Mbits/sec
[  5]   6.00-7.00   sec   112 MBytes   937 Mbits/sec
[  5]   7.00-8.00   sec   111 MBytes   929 Mbits/sec
[  5]   8.00-9.00   sec   112 MBytes   938 Mbits/sec
[  5]   9.00-10.00  sec   111 MBytes   935 Mbits/sec
[  5]  10.00-10.04  sec  4.62 MBytes   942 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-10.04  sec  1.08 GBytes   926 Mbits/sec                  receiver
```

I'm not even sure this is the right place. If I ask for help in a Pi4 community, they will send me here probably :)

## Specifications

  - OpenMPTCProuter version: 0.54.7
  - OpenMPTCProuter VPS version: 0.1015 4.19.104-mptcp
  - OpenMPTCProuter platform: RPI4 with ext4 image 'openmptcprouter-v0.54.7-r012509-04a21c26a0-bcm27xx-bcm2711-rpi-4-ext4-factory.img.gz'
  - Samba4-server: 4.12.3-4

Ysurac, 2020-06-29:
Can be related to network cache issue. You can try latest OpenMPTCProuter 0.55 beta if you want.

robertoglxda, 2020-07-10:
Thanks, I think I'll wait for the stable versione.
However, now samba is working fine on my setup. I was able to tweak smb a bit, and formatting to ext4. I think it would work great also in ntfs after the tweaks.

closed 2020-11-07 by robertoglxda