Ysurac/openmptcprouter #1082: Question about VPS key exchange with multiple VPSes

vistman, 2020-06-29:
## Expected Behavior
Don't know ... My guess is all VPSes entered after first VPS will exchange it's keys no matter they all got their own keys ...

## Current Behavior
Don't know ...

## Possible Solution
I'd just like to get an explanation to be sure how kez exchange is actually working ...

## Steps to Reproduce the Problem
No steps needed ;-)

## Context (Environment)
I have multiple VPSes and sometimes VPN Tunnel goes down when exchanging them so I'd like to understand the process of handling multiple VPS keys ...
For example on my "oldest" VPS no matter I'm running latest OMR VPS version I don't have Dead simple VPS key ...

## Specifications
- OpenMPTCProuter version: v0.54.7
- OpenMPTCProuter VPS version: 0.1015 4.19.104
- OpenMPTCProuter platform: RPI4b/x86_64)

Ysurac, 2020-06-29:
I don't understand what is exactly the issue. What you mean by "when exchanging them" ?
In next release the way used for multiple VPS is modified, nginx-ha will not be used anymore so this use less CPU and should be better.

vistman, 2020-06-29:
What I mean is - there is an option to add server name, server username and server key for each VPS, but only one entry for ShadowSocks key, Glorytun key, A Dead Simple VPN key and MLVPN password (under Common server settings ->Advanced settings) no matter how manz VPSes have been added. As each VPS gets all of these keys (newly generated) after installation script has completed installation process, my guess (question if I'm right) is that ShadowSocks key, Glorytun key, A Dead Simple VPN key and MLVPN password are common to all VPSes no matter all VPSes get them upon installation script has completed. But what I'd like to know how OMR decides which ones will be used ...

Ysurac, 2020-06-29:
If you use multiple VPS, keys of first VPS (or any previous defined key) are sent to all others.

vistman, 2020-06-30:
Thanks. And what happens if I delete this first VPS from which keys are propagated? Will keys from second VPS be propagated on others? I'm sorry but this is important to me since I'm trying to find the most reliable VPS and therefore am quite often adding new and deleting new VPSes and then it happens I'm getting VPS tunnel down so I'm trying to pinpoint the problem.

Ysurac, 2020-06-30:
No, keys of any others VPS are replaced by first VPS key. So if you remove any VPS, keys are still the same.
For the tunnel down, this can be related to nginx HA that is used in current stable, a simpler mechanism is used in next release.

vistman, 2020-06-30:
Thanks. So it is a very good idea - mandatory to always keep the primary generated keys ... This explains some of the problems that occurred while I was experimenting with VPSes.
I had a case where my primary added VPS did not have A Dead Simple VPN key generated no matter it has been updated/upgraded ...

vistman, 2020-07-01:
Just one thing ... Today I've decided to reinstall my fastest VPS with latest stable OMR version, making sure fail2ban is uninstalled, but I was constantly getting VPN tunnel DOWN whenever some increased network activity occurred (ie. gaming) which is kind of funny, because when I switched to another VPS everything is working fine without VPN tunnel going down on the same OMR version ...
Then I've decided to try OMR latest beta and VPN tunnel DOWN hasn't occurred yet, so whatever you changed seems to be working better ...

closed 2020-07-05 by vistman