Ysurac/openmptcprouter #1084: Hardware to be supported in the 0.55 release

Corrado56, 2020-07-02:
Hi, Just a question about the hardware to be supported in version 0.55
Linksys wrt3200ACM will still be supported ?
Any new hardware ?

Regards

Ysurac, 2020-07-03:
Hi,
All hardware supported in 0.54 will still be supported in 0.55.
No new hardware supported, but if you have some suggestion...

closed 2020-07-03 by Ysurac

Corrado56, 2020-07-03:
[image]

I asked because my Raspberry is a bit stressed, I prefere RJ45 ports native on hardware, Linksys is not cheap, and I want to be sure will be supported in the future.

Thanks

Ysurac, 2020-07-03:
If you have a switch, you don't need to use adapters on the Raspberry.

Corrado56, 2020-07-03:
Does not work for me,  see #1075

Ysurac, 2020-07-03:
yes true, sorry.
Even if I still don't understand why this work with adapters ;)

bambuca, 2020-07-03:
I had problems with switch. ISP1 was using DHCP and the second (ISP2) don't. When I connected both into one switch then the DHCP was propagated to ISP2 and he was a bit angry.