Ysurac/openmptcprouter #1086: modemmanager wan not working after reconnect

Peyoz, 2020-07-09:
## Expected Behavior
WAN via LTE card configured through Modemmanager should be kept online even after a disconnect 

## Current Behavior
Interface is fine through gui, but traffic is not flowing.
mmli reports state as registered which means it is ready for connection, but not connected

## Possible Solution
not sure

## Steps to Reproduce the Problem

  1. Configure a WAN interface as modemmanager device
  2. Wait for a disconnection event
  3. Check with mmcli state as registered and not Connected

## Context (Environment)
Just keep up connections

## Specifications

  - OpenMPTCProuter version: 0.54.7
  - OpenMPTCProuter VPS version:  0.1102
  - OpenMPTCProuter platform: RPi4I

Peyoz, 2020-07-09:
I'm attaching a full log of the event, please have a look at timestamps around 10:30 (now they are fine).
Looks like hotplug events are correctly managed, but there is something else which gets in middle of it..
Let me know how I can assist you on this :)
Many thanks in advance 
[modemmanager.log](https://github.com/Ysurac/openmptcprouter/files/4897406/modemmanager.log)

Peyoz, 2020-07-13:
Hi, I made some further checks, and the problem is a lack of integration between modemmanager and ubus, so events are not generated on modem disconnect, thus no action is taken.
Is there any kind of watchdog to ensure the wan connection is up?
As a side note, I made an hotplug script which triggers an ifup/ifdown when a modemmanager based interface monitored by mwan3 is down, do you think we can port it to this project?

Ysurac, 2020-07-13:
omr-tracker do an ifdown/ifup when a connection using modemmanager is detected as down.

Peyoz, 2020-07-13:
Hi, I did a check and it works, unfortunately my gateway is not pingable, so it just keeps on restarting the connection.
Is there a way to inhibit the gateway ping from the omr-tracker?

Ysurac, 2020-07-14:
Yes, in System->OpenMPTCProuter and advanced parameters tab. But gateway ping is not the only test done, it also test to ping external IPs from the interface. You can check log in State->System log to find why an interface is show as down.

Peyoz, 2020-09-04:
Works fine with 0.55, closing

closed 2020-09-04 by Peyoz