Ysurac/openmptcprouter #1087: Luci MPTCP bandwidth incoherent values

gnosoz, 2020-07-09:
I am having a problem with the Luci graphics under MPTCP menu as it shows incorrect information.

As per below picture you can see what follows:

## Expected Behavior
    speedtest 630Mbps aggregated
    iptraf-ng  shows correct values for wan1 and wan2 which if summed give roughly eth0 throughput
    LuCi shows instead values which are incoherent as the total sum 630Mbps is shown for wan1 and the same look appears for wan2
[image]

## Current Behavior
The tests I run show the following:

    no MPTCP and only one wan active = correct graph values
    multiple wans shows both always roughly 1.53 of the real value taken by iptraf and confirmed by speedtests

This scenario was confirmed on multiple occasions as it was showing for an ADSL limited at 15Mbps that it was going at roughly 25-30 which is physically impossible.

[image]

## Possible Solution
A) verify  bandwidth.js script to see why it performs in such a odd way, maybe it picks the wrong field and the graph shows the total throughput by default
B) modify .js script to use iptraf-ng values or bmon to generate the graphics
C) when more than one connection is being shown it devides the data by 1.53

## Specifications

  - OpenMPTCProuter version: 0.47
  - OpenMPTCProuter VPS version: 0.47
  - OpenMPTCProuter platform : x86_64

Ysurac, 2020-07-09:
The graph show total speed and the colors are part of each connections inside the total speed.

gnosoz, 2020-07-10:
yes that is understood but my point is that iptraf-ng and speedtests are similar except for the protocol overhead whilst the graphs are showing different values which normally amount to 50% per line.

I tried several different tests and it seems to be constantly producing details which appear incorrect even when compared with the graphs shown in the menu luci/admin/status/realtime/bandwidth.

It seems to be showing for wan1 the aggregated value of wan1+wan2 or maybe the br-lan value as it is identical to that if you compare it with either iptraf results or realtime graphs.

For example 
         iptraf eth1+eth2 = eth0 - overhead
         realtime graphs eth1+eth2 = eth0 - overhead
         MPTCP graphics eth1+eth2 = eth0 + between 53 to 80% more

Ysurac, 2020-07-10:
```luci-bwc``` is used as source, it's the same source that is used for State->Realtime graph
I have maybe an error in my js in ```/usr/lib/lua/luci/view/mptcp/multipath.htm``` but no idea where.
I don't really the way it's done, but I didn't find something good that doesn't use too much CPU and give a parsable output (or an openwrt application that already combine interfaces on a graph).

gnosoz, 2020-07-10:
I can imagine as the script is quite complex :)
To narrow the problem down I did some further research and if you look at below picture the in depth inspection shows that data is correct but is just the drawing which has an offset for some reasons.
[image]

This is what I can see from the picture above:
   - the drawing reports doesn't match peaks / current values reported below the graph
   - the data generated on the html source is showing the correct values on the righthand side (I chose to see the current Kbps)
   - the data generated is matching both iptraf and external speedtests

So this means the data which is fed to the script is correct but then there is probably a small drawing calculation which is offsetting the drawing.

If my calculations are correct it seems that the problem might be in these lines located in the file:
/usr/lib/lua/luci/view/mptcp/multipath.htm  at the position 691/829 83%

```
var size = Math.floor(Math.log2(data_max)),                                                                                                                                                  
div = Math.pow(2, size - (size % 10)),                                                                                                                                                   
mult = data_max / div,                                                                                                                                                                   
mult = (mult < 5) ? 2 : ((mult < 50) ? 10 : ((mult < 500) ? 100 : 1000));                                                                                                                

data_max = data_max + (mult * div) - (data_max % (mult * div));                                                                                                                              
```

Let me know if you recognize them and if they are the ones involved I will see if I can correct and send the new code over for the develop version :)

Ysurac, 2020-07-10:
This JS part is only for individual interface graph.

closed 2020-11-13 by github-actions[bot]