Ysurac/openmptcprouter #1090: Aggregation change depending on master

PxxBxx, 2020-07-13:
## Expected Behavior
All WAN should be used whatever the master interface is

## Current Behavior
Nothing changed these last 2-3 months (same VPS version 1015, same router version 0.54.7, same infra/hardware), but :
* when selecting WAN2 as master, only WAN2 is used for aggretation
* when selection WAN1 or WAN3 as master, only WAN1+WAN3 are used for aggregation

In April/May, with same setup, all my WANs where used, whichever master I chose (or was chosen automaticaly)

## Possible Solution
I tried to change path_manager, scheduler, congestion_control, but nothing made it work like before.

## Steps to Reproduce the Problem

n/a

## Context (Environment)

*Any clue for this behaviour ? a change in MPTCP support on one of my ISP ? is there any log/debug I can send you ?*

## Specifications

  - OpenMPTCProuter version: 0.54.7
  - OpenMPTCProuter VPS version: 1015
  - OpenMPTCProuter platform: x86_64 non-UEFI

Ysurac, 2020-07-17:
Possible that wan2 doesn't support MPTCP anymore.
What is the ISP ?

PxxBxx, 2020-07-17:
Hello @Ysurac , you're probably right, here is the output of the support-check : 
```
TCP::SrcPort TCP::DstPort TCP::SeqNumber TCP::AckNumber TCP::DataOffset TCP::Flags TCP::WindowsSize TCP::CheckSum IP::TotalLength IP::Identification IP::TTL IP::CheckSum IP::SourceIP IP::DestinationIP +TCPOptionMaxSegSize -TCPOptionMPTCPCapable
```
Sadly there is the `-TCPOptionMPTCPCapable`

=> I just discovered the option to use MPTCP over VPN, I just activated it for wan2 and now it works again.

Thanks a lot, and thanks to you for making this option available !

PxxBxx, 2020-07-17:
Sorry, forgot to answer the question "which ISP" => it was SFR, with their 4GBox
(first, it sucks because hard-capping at 200GB/mo., second, they dropped MPTCP support :'( )

Ysurac, 2020-07-17:
ok thanks, I will make test on a SFR connection (without a 4GBOX) to check if I have same result.

Ysurac, 2020-07-18:
I tested, tracebox report the "-TCPOptionMPTCPCapable" but it was already the case before from SFR. But I can add it to the aggregation and it's working. I'm using a Huawei USB LTE dongle.

The tracebox report, for comparison:
```
root@OpenMPTCProuter:~# omr-tracebox eth1
tracebox to 130.104.228.140 (multipath-tcp.org): 64 hops max
1: 192.168.105.1 36ms 
2: *
3: 10.4.0.13 64ms  [PARTIAL] +PartialTCP < PartialTCP (8 bytes) :: SrcPort = 25190 , DstPort = 80 , SeqNumber = 2067660161 , >
 IP::TTL (3 -> 1) IP::CheckSum (0x4818 -> 0x4a18) 
4: 130.104.228.140 52ms TCP::SrcPort (25190 -> 80) TCP::DstPort (80 -> 25190) TCP::SeqNumber (2067660161 -> 3865212004) TCP::AckNumber (0 -> 2067660162) TCP::DataOffset (10 -> 7) TCP::Flags (( SYN ) -> ( SYN ACK )) TCP::WindowsSize (5840 -> 23360) TCP::CheckSum (0x5015 -> 0x618b) IP::DiffServicesCP (0 -> 10) IP::TotalLength (60 -> 48) IP::Identification (0x9ea2 -> 0xf5f5) IP::TTL (4 -> 251) IP::CheckSum (0x4718 -> 0xf8a7) IP::SourceIP (192.168.105.100 -> 130.104.228.140) IP::DestinationIP (130.104.228.140 -> 192.168.105.100) -TCPOptionMPTCPCapable < TCPOptionMPTCPCapable (12 bytes) :: Kind = 30 , Length = 12 , Subtype = 0 , Version = 0 , Checksum = 1 (Checksum Enabled) , Flags = 0 , Crypto = 1 (HMAC-SHA1) , Sender's Key = Sender's Key = 4669631797620057590 , >
 TCPOptionWindowScale::Shift (14 -> 4)
```

closed 2020-08-12 by Ysurac