Ysurac/openmptcprouter #1091: Problem with IPv6 VPN

AdrienDssy, 2020-07-14:
## Current Behavior
Hello, when I have configured v6 in my VPS and activate v6 in my router I get this :
[image]
What's the problem ?

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter platform:

Ysurac, 2020-07-15:
The IPv6 tunnel is down. What is the VPN used ? What do you have in State->System log about 6in4 ?

AdrienDssy, 2020-07-15:
OpenVPN,

> Tue Jul 14 19:53:09 2020 daemon.err /usr/bin/ss-redir[32474]: remote recv: Connection reset by peer
Tue Jul 14 19:53:10 2020 daemon.err /usr/bin/ss-redir[32475]: remote recv: Connection reset by peer
Tue Jul 14 19:53:11 2020 daemon.err /usr/bin/ss-redir[32475]: remote recv: Connection reset by peer
Tue Jul 14 19:53:11 2020 daemon.err /usr/bin/ss-redir[32475]: remote recv: Connection reset by peer
Tue Jul 14 19:53:11 2020 daemon.err /usr/bin/ss-redir[32474]: remote recv: Connection reset by peer
Tue Jul 14 19:53:12 2020 daemon.err /usr/bin/ss-redir[32474]: remote recv: Connection reset by peer
Tue Jul 14 19:53:22 2020 user.notice post-tracking-post-tracking: Get status and settings for wan2... Done
Tue Jul 14 19:53:47 2020 daemon.err /usr/bin/ss-redir[32474]: remote recv: Connection reset by peer
Tue Jul 14 19:53:56 2020 daemon.err /usr/bin/ss-redir[32474]: remote recv: Operation timed out
Tue Jul 14 19:54:03 2020 daemon.err /usr/bin/ss-redir[32474]: remote recv: Connection reset by peer
Tue Jul 14 19:54:03 2020 daemon.err /usr/bin/ss-redir[32474]: remote recv: Connection reset by peer
Tue Jul 14 19:54:05 2020 daemon.err /usr/bin/ss-redir[32475]: remote recv: Connection reset by peer
Tue Jul 14 19:54:05 2020 daemon.err /usr/bin/ss-redir[32474]: remote recv: Connection reset by peer
Tue Jul 14 19:54:15 2020 daemon.err /usr/bin/ss-redir[32474]: remote recv: Operation timed out
Tue Jul 14 19:54:23 2020 daemon.err /usr/bin/ss-redir[32475]: remote recv: Operation timed out
Tue Jul 14 19:55:24 2020 daemon.err /usr/bin/ss-redir[32475]: remote recv: Operation timed out
Tue Jul 14 19:56:05 2020 daemon.err /usr/bin/ss-redir[32474]: remote recv: Operation timed out
Tue Jul 14 19:56:17 2020 daemon.err /usr/bin/ss-redir[32475]: remote recv: Operation timed out
Tue Jul 14 19:56:17 2020 daemon.err /usr/bin/ss-redir[32474]: remote recv: Operation timed out
Tue Jul 14 19:56:27 2020 daemon.err /usr/bin/ss-redir[32475]: remote recv: Operation timed out
Tue Jul 14 19:57:06 2020 daemon.err /usr/bin/ss-redir[32475]: remote recv: Operation timed out
Tue Jul 14 19:57:06 2020 daemon.err /usr/bin/ss-redir[32474]: remote recv: Operation timed out
Tue Jul 14 19:57:06 2020 daemon.err /usr/bin/ss-redir[32474]: remote recv: Operation timed out
Tue Jul 14 19:57:06 2020 daemon.err /usr/bin/ss-redir[32475]: remote recv: Operation timed out
Tue Jul 14 19:57:06 2020 daemon.err /usr/bin/ss-redir[32474]: remote recv: Operation timed out
Tue Jul 14 19:57:06 2020 daemon.err /usr/bin/ss-redir[32475]: remote recv: Operation timed out
Tue Jul 14 19:57:06 2020 daemon.err /usr/bin/ss-redir[32475]: remote recv: Operation timed out
Tue Jul 14 19:57:06 2020 daemon.err /usr/bin/ss-redir[32475]: remote recv: Operation timed out
Tue Jul 14 19:57:08 2020 daemon.err /usr/bin/ss-redir[32475]: remote recv: Operation timed out
Tue Jul 14 19:58:04 2020 daemon.err /usr/bin/ss-redir[32474]: remote recv: Operation timed out
Tue Jul 14 19:59:33 2020 daemon.err /usr/bin/ss-redir[32475]: remote recv: Operation timed out
Tue Jul 14 19:59:37 2020 daemon.err /usr/bin/ss-redir[32475]: remote recv: Connection reset by peer
Tue Jul 14 20:00:00 2020 cron.err crond[6132]: USER root pid 25834 cmd /etc/init.d/vnstat_backup backup
Tue Jul 14 20:00:10 2020 daemon.err /usr/bin/ss-redir[32474]: remote recv: Operation timed out
Tue Jul 14 20:01:28 2020 daemon.err /usr/bin/ss-redir[32474]: remote recv: Connection reset by peer
Tue Jul 14 20:01:42 2020 user.notice post-tracking-post-tracking: eth2 switched off because ping from 192.168.10.108 error (9.9.9.9,1.1.1.1)
Tue Jul 14 20:01:42 2020 user.notice post-tracking-post-tracking: Delete default route via 192.168.10.1 dev eth2
Tue Jul 14 20:01:46 2020 user.notice post-tracking-post-tracking: Set server vps (54.38.188.175) default route via 192.168.10.1
Tue Jul 14 20:01:46 2020 user.notice post-tracking-post-tracking: Set server vps (54.38.188.175) route via 192.168.10.1 metric 4
Tue Jul 14 20:01:46 2020 user.notice post-tracking-post-tracking: Get status and settings for wan2...
Tue Jul 14 20:01:48 2020 daemon.err /usr/bin/ss-local[32473]: remote_recv_cb_recv: Connection reset by peer
Tue Jul 14 20:01:59 2020 user.notice post-tracking-post-tracking: Get status and settings for wan2... Done
Tue Jul 14 20:02:08 2020 daemon.err /usr/bin/ss-redir[32474]: remote recv: Operation timed out
Tue Jul 14 20:02:19 2020 daemon.err /usr/bin/ss-redir[32475]: remote recv: Operation timed out
Tue Jul 14 20:02:37 2020 daemon.err /usr/bin/ss-redir[32475]: remote recv: Connection reset by peer
Tue Jul 14 20:03:01 2020 daemon.info dnsmasq-dhcp[12171]: DHCPREQUEST(eth4) 192.168.100.230 00:26:f2:b8:33:cd
Tue Jul 14 20:03:01 2020 daemon.info dnsmasq-dhcp[12171]: DHCPACK(eth4) 192.168.100.230 00:26:f2:b8:33:cd
Tue Jul 14 20:04:16 2020 daemon.err /usr/bin/ss-redir[32474]: remote recv: Connection reset by peer
Tue Jul 14 20:04:56 2020 daemon.err /usr/bin/ss-redir[32475]: remote recv: Connection reset by peer
Tue Jul 14 20:05:07 2020 daemon.err /usr/bin/ss-redir[32475]: remote recv: Operation timed out
Tue Jul 14 20:07:40 2020 daemon.err /usr/bin/ss-redir[32474]: remote recv: Operation timed out
Tue Jul 14 20:07:58 2020 daemon.err /usr/bin/ss-redir[32474]: remote recv: Connection reset by peer
Tue Jul 14 20:08:00 2020 daemon.err /usr/bin/ss-local[32473]: getpeername: Socket not connected
Tue Jul 14 20:09:27 2020 daemon.err /usr/bin/ss-redir[32474]: remote recv: Operation timed out
[... cut ...]
Wed Jul 15 08:03:50 2020 daemon.err /usr/bin/ss-local[32473]: getpeername: Socket not connected
Wed Jul 15 08:04:06 2020 daemon.err /usr/bin/ss-redir[32474]: remote recv: Operation timed out
Wed Jul 15 08:04:11 2020 daemon.err uhttpd[6675]: luci: accepted login on / for root from 192.168.100.212
Wed Jul 15 08:04:16 2020 user.notice OMR-VPS: Can't get token, try later
Wed Jul 15 08:04:22 2020 daemon.err /usr/bin/ss-redir[32474]: remote recv: Operation timed out
Wed Jul 15 08:04:23 2020 daemon.err /usr/bin/ss-redir[32474]: remote recv: Operation timed out

Ysurac, 2020-07-15:
What is the result on the server of ```ip a show omr-6in4-user0``` ?

AdrienDssy, 2020-07-15:
[image]

Ysurac, 2020-07-15:
What is the result of ```ip a show 6in4-omr6in4``` on the router via SSH ?

AdrienDssy, 2020-07-15:
[image]

Ysurac, 2020-07-15:
I don't think peer is correct on the router part. I will make some tests and I will fix that. I think this only happen with OpenVPN.

closed 2020-07-15 by AdrienDssy