Ysurac/openmptcprouter #1094: Unable to connect on Virtualbox

rifaterdemsahin, 2020-07-19:
## Expected Behavior
I expect to be able to connect to OMR using internal networks in Virtualbox.
I expect to be able to connect to OMR using hostonly networks in my network

## Current Behavior
I can ping the box inside OMR
I cant reach from outside. it is a new setup there is not dhcp leases

## Possible Solution
If I can resolve the issue I would create a video on to do a fast setup in the virtual box where people can use it 

## Steps to Reproduce the Problem
Attached all the screenshots

## Context (Environment)
I am running the virtualbox in a skull canyon intel

## Specifications
I have 5 wan connections ( fiber,dsl,3 lte)

  - OpenMPTCProuter version: 0.54.7
  - OpenMPTCProuter VPS version:

Ysurac, 2020-07-19:
How many interfaces did you set in virtualbox ?
Did you set interface as Server Intel PRO/1000 MT (82545EM) set to bridge with promiscous mode enabled ?

rifaterdemsahin, 2020-07-19:
mate thanks for the ultrafast response . I did setup 5 interfaces
1 for internal network
1 host only network
3 for wan

**I have realized Missed the last letters on the nic type** / Intel PRO/1000 MT (82545EM) 
I moved from intel 82540 to 82545 now ( dont know why might be related to multi path tcp maybe )

Updated the scripts for the external card in the virtualbox with

> ` VBoxManage modifyvm "OpenMPTCP" --nic5 hostonly
VBoxManage modifyvm "OpenMPTCP" --nictype5 82545EM
VBoxManage modifyvm "OpenMPTCP" --hostonlyadapter5 enx000ec651112d
`

Still i am unable to access it.

rifaterdemsahin, 2020-07-19:
I changed the network to 192.168.200.1

 i would revert that change. How can I debug OMR to see the issue?

[image]

rifaterdemsahin, 2020-07-19:
I have decided to start from scratch and add the cards one by one. Would that help ?
[image]

rifaterdemsahin, 2020-07-19:
Maybe the internal network is causing the issue
[image]

Ysurac, 2020-07-19:
On network adapter, the lan interface need to be attached to internal network if it's what you want.

rifaterdemsahin, 2020-07-19:
I have created an internal network and i was expecting the client to see the server but that did not work.

I suspect it is related to hostonly adapter issue in virtualbox after the update i can only mention the network card on the cli and the UI throws the error

[image]

If i want to bond like 600MBits with which hardware should i go for to have the least issues?

rifaterdemsahin, 2020-07-19:
maybe today is not my day i am giving a break .
I found the thread here https://ltehacks.com/viewtopic.php?f=21&t=1078

Project looks promising. Keep up the great work and keep pedalling it would be great to be able to bond connections

Ysurac, 2020-07-19:
I use virtualbox with internal network, one interface attached to internal network, others interfaces as bridge for a direct access to modem/box.

I don't have 600Mb/s to test speed for now, but I would say any x86_64 with AES hardware support will have no problem at all.

rifaterdemsahin, 2020-07-19:
do you setup the host network adapter when you want to use the virtualbox as the router in the network ?
[image]

Ysurac, 2020-07-19:
I don't use a host private network, only an internal network.

rifaterdemsahin, 2020-07-19:
5 th time is the charm
[image]

I have understood that if we do not setup correctly at the start the restart does not read the updated config.
So I created the machine from start and it worked ( that part needs to be updated i suppose as there can be hardware changes )
[image]

Ysurac, 2020-07-19:
I don't know what you mean, the config can be updated.

rifaterdemsahin, 2020-07-19:
I have tried many times with the original one. If I do not create the machine with 82545EM than i couldnt change and expect the router page to load. That was my case.

closed 2020-07-25 by rifaterdemsahin