Ysurac/openmptcprouter #1096: Unreliable connection due to one of the connections being unstable

enricodetoma, 2020-07-20:
I have the attched configuration, where I have 3 internet connections.
The 1st and the 3rd are ADSL and are quite stable.
The 2nd (middle) connection is 4G and it often gets disconnected with IP change.
For VPS I use Hetzner VPS with 1 vcpu and 2Gb RAM.

## Expected Behavior
In presence of a disconnection of the 4G internet connection, all TCP connections should stay alive thank to the two other ADSL connections.

## Current Behavior
TCP connections go down anyway, even if the ADSLs are still connected.

## Possible Solution
I would like to know how to change the configuration to make my overall connection stable, even if it is necessary to waste some bandwidth.
I already tried to put `net.ipv4.tcp_retries2=3` manually in `/etc/sysctl.d/shadowsocks.conf` and `IPv4 TCP SYN retries` is equal to 3, but it didn't improve the stability when the 4G disconnects.

## Specifications

  - OpenMPTCProuter version: 0.55beta16
  - OpenMPTCProuter VPS version: 0.1015 5.4.0-mptcp
  - OpenMPTCProuter platform: x86 with 4 ethernet ports

[image]

Ysurac, 2020-07-20:
You should update to latest beta and use snapshot VPS script.
Increase in Network->MPTCP the value of Multipath TCP SYN retries until it's ok.

enricodetoma, 2020-07-23:
Thank you, now the stability is quite improved.

closed 2020-07-23 by enricodetoma