Ysurac/openmptcprouter #1097: Vps ip assigned to another internal router

dettofatto, 2020-07-22:
## Expected Behavior

I would give my vps ip directly to my router (fritzbox). So I can use my fritzbox as router/dhcp server/firewall

## Current Behavior

My pi4 (192.168.2.1) give internal ip to  my fritzbox's wan port (192.168.2.2). So I can't use port forwarding.
Also if I add a rule in port forwarding for redirecting all traffic to my fritzbox I can't have any success.
I would like have my external ip assigned to my fritzbox, or it's better put my fritzbox into dmz zone?

## Possible Solution

Put fritzbox in dmz (how is possible?)

## Steps to Reproduce the Problem

## Context (Environment)

## Specifications
I have 2  huawei modem with 4g, that are connected to rpi4. 
On mi rpi4 I've disabled dhcp, and give internet to my fritzbox through fritzbox's wan port. So actually I have 2 router.

  - OpenMPTCProuter version:  0.55 beta 15
  - OpenMPTCProuter VPS version:  0.55 beta 15
  - OpenMPTCProuter platform: 
Rpi4

Ysurac, 2020-07-31:
OpenMPTCProuter can't be used in a bridge mode, so a public IP can't be directly attributed to any internal network.

closed 2020-07-31 by Ysurac

Adorfer, 2020-07-31:
@dettofatto actually you do not want to use the mptcp-router as modem (e.g. Layer2-bridge)
perhaps you want it to behave as a pppoe-server on vlan 7?

Or is it just that you want to have a/one of the VPS-IPs on the internal network? 
In the last case, this would be a duplicate issue with https://github.com/Ysurac/openmptcprouter/issues/1006

leoogermenia, 2022-10-02:
Hi,
I'm in the same situation.
With OMR, I can't use my VPN client.
I have 2 VDSL and one 4G connection connected to OMR. Plus I have 2 VPS (one in Aruba, another one at my second house where I have a FTTH connection)
The fritz can establish a connection with another fritz (the LAN - LAN VPN) with different IPs for each fritz connected (VPS IP, WAN1 IP, etc.). 
But the VPN Client, for connect a single device to the fritz LAN, not work at all. I want to redirect the VPN connection to the 4G connection, because is the faster one. I tried to create a Port Forwarding and a OMR-Bypass, but not of this action work.