Ysurac/openmptcprouter #1099: Top Speeds with MpTCP

rifaterdemsahin, 2020-07-26:
## Expected Behavior
Aggregate all the bandwidth

## Current Behavior
Some form of aggregation is working

## Possible Solution
Have a panel on the dashboard to show active consumption of the bandwidth
I need to see if the VPS is the bottleneck.

## Steps to Reproduce the Problem

  Cloud Server at GCP
[image]

5 Broadbands
1.Fiber Virgin FTTC ( 320 )
2.LTE Vodaphone ( 40 )
3.LTE Smarty ( 50 )
4.LTE Three ( 50 )
5.DSL Post Office  ( 10 )

Expected output 460 but getting 340
[image]

## Context (Environment)
Running it on skull canyon intel with ESXI

## Specifications

  - OpenMPTCProuter version: Powered by openmptcprouter v0.54.7 r0+12509-04a21c26a0
  - OpenMPTCProuter VPS version:

Adorfer, 2020-07-27:
please have look onto (cpu-) load and io-delay on the vps during testing (and make some tests e.g. via iperf3 (from a client-pc in lan and iperf3 hosted close to the your VPS), lasting for several mininutes, in order to get stable results.

rifaterdemsahin, 2020-07-27:
Adorfer the KS for the update. I realized the VPS is in mountain view I should have selected London DC as I am in Cambridge uk.

Today wan2 stopped working. I have started getting errors
[image]

Ysurac, 2020-07-27:
wan2 gateway IP doesn't answer to ping. This can be a problem or not, some hardware block ICMP ping request.

user747, 2020-07-27:
> io-delay on the vps during testing
what tool do you use for this

rifaterdemsahin, 2020-07-28:
https://www.youtube.com/watch?v=5URNj8YexsY

Here is today.
I cant seem to see to utilise all the bandwidth

Ysurac, 2020-07-28:
The VPN is used by default only for traffic that is not TCP. Enabling/disabling VPN should always be done via the wizard (else you can have more than one VPN enabled, and this can't work well).
Connection should also have low latency with the VPS if you want to get a good speed, it's not really the case here.

To optimize, you can try in Network->MPTCP to set scheduler to BLEST and Congestion control to Cubic, sometimes this help.

rifaterdemsahin, 2020-07-28:
The VPN is used by default only for traffic that is not TCP.:: understood.I suspect 80 percent of traffic is TCP.

Enabling/disabling VPN should always be done via the wizard ::: ok

 Network->MPTCP to set scheduler to BLEST and Congestion control to Cubic, sometimes this help.

::; I will check this out. http://192.168.16.113/cgi-bin/luci/admin/network/mptcp

http://192.168.16.113/cgi-bin/luci/admin/services/iperf
this page displays 
Upload - Server: bouygues.iperf.fr - Sender: 316.61Mb/s - Receiver: 324.77Mb/s

But i do not see this in fast.com ^^^

Ysurac, 2020-07-28:
Some test speed always give bad results. The best way to test is to download a file, on the router, via SSH, there is ```omr-test-speed``` (ctrl+c to stop test). It download a file from OVH.
I will add an internal speed test in OpenMPTCProuter in a next release.

rifaterdemsahin, 2020-07-29:
Ysurac thanks for all the great work on this. If i add 3rd line would i improve due to the algorithm maybe 2 lines with mptcp is not the best. Is there a study on how many lines gives the best possible speed ?

Ysurac, 2020-07-29:
The best is one line with max speed, but you can add up to 8 lines. Latency is important.

rifaterdemsahin, 2020-07-29:
in the morning the speed was at 100 at night it goes to 400 ( using 2 lines)
[image]

how do i debug the reason ?

kevin39, 2020-07-29:
Hi,
Do not use fast.com.

As @Ysurac said, check your speed using CLI with `omr-test-speed`

While testing you can go to Network > MTCP > Bandwidth to Check which connections are used.

rifaterdemsahin, 2020-07-29:
Kevin thanks for the response.

much better result on this page
http://192.168.16.113/cgi-bin/luci/admin/network/mptcp/bandwidth

[image]

Adorfer, 2020-11-27:
Just for reference: 
[image]

TalalMash, 2020-12-31:
@rifaterdemsahin You can try Vulr hourly dedicated server to see if it's VPS processing bottleneck, no commitment or fixed pricing. I've only tested up to 300mbit.

closed 2021-04-06 by github-actions[bot]