Ysurac/openmptcprouter #1100: How to install nextdns to openmptcp router

rifaterdemsahin, 2020-07-27:
Should I do command line install or just use the ips ?

Ysurac, 2020-07-27:
You can add the IPs in Network->DHCP & DNS the same way as "127.0.0.1#5353" (it redirect to local Unbound DNS server who query root DNS servers).

user747, 2020-07-27:
if using pihole already and want to see or try nextdns blocklist

git clone --depth 1 https://github.com/nextdns/metadata && cd metadata && grep -r '.txt'

gwarf, 2020-07-27:
Looking at nextdns.io instructions for routers it seems that to avoid manually installing an app that could cause troubles with OpenMPTCPRouter you may have to use their DNS, and linking your VPS IP:
- https://help.nextdns.io/en/articles/3972204-which-setup-type-to-use
- https://help.nextdns.io/en/articles/3941230-what-is-linked-ip

Out of curiosity I've tried to add the DNS they provided me after singing up, but OpenMPTCPRouter doesn't accept it's format:
[image]
Wondering about `/use-application-dns.net/`entry I saw https://support.mozilla.org/fr/kb/configurer-reseaux-desactiver-dns-via-https which seems to means that it warns applications like Firefox that DNS over HTTPS won't work. It's meant to enforce using the router's DNS (unbound)?

Ysurac, 2020-07-27:
@gwarf you use latest stable release ? I don't remember if I solved the issue in current release or after the release.
Yes the ```/use-application-dns.net/``` disable DNS over HTTPS. This force the use of OpenMPTCProuter DNS, it's needed for OMR-ByPass.

gwarf, 2020-07-27:
Yep sorry I missed to add the version before, it's now it, it should be latest stable.

gwarf, 2020-07-27:
Is there a way to manually overwrite this conf via SSH and edition of some locale file?

Ysurac, 2020-07-27:
Via SSH on the router: ```uci add_list dhcp.@dnsmasq[0].server='8.8.8.8'```

gwarf, 2020-07-27:
So it seems I also need `uci commit dhcp` to save the changes. Can/should I also delete the local/unbound entry to enforce using the new DNS server? Would it affect/break some specific features of OpenMPTCPRouter?

Ysurac, 2020-07-27:
Yes you have to delete default 127.0.0.1#5353 and this will not break anything in OpenMPTCProuter.

rifaterdemsahin, 2020-07-28:
so i need to do is 

1. add /use-application-dns.net/
2.XX.XX.XX.XX#53

XX.XX.XX.XX is the nextdns server and 53 is the port.

and it should not kill the router right?

gwarf, 2020-07-29:
So use-applicaiton-dns.net should already be there, you should see it in the `Network->DHCP & DNS` section.
What I did was:
```sh
# once ssh keys have been configured via the web interface
ssh root@192.168.100.1
# check what is there, to be used to rever if needed
uci get dhcp.@dnsmasq[0].server
# Add the nextdns DNS server, no need to specify the port if it's on 53
uci add_list dhcp.@dnsmasq[0].server='X.X.X.X'
# Save changes
uci commit dhcp
```
Then as I was unsure about the appropriate I went to `Network->DHCP & DNS` to delete the unbound entry `127.0.0.1#5353`, saved and reload then all went fine.

I'm testing nextdns since then, no problem at all. They have a great dashboard, it's also possible to request to get data hosted in the EC and to configure logs retention. And the stats/analytics are quite intereseting, it's what looks the more interesting to me.
Only but gig drawback is that it's all in the hands of an American startup, while I would prefer to be sure that this data will remain private...
So will probably stop in a few days, once I will have a definitive feeling about the service.

A similar dashboard in OpenMPTPCPRouter with something like pihole would be a killer feature
[image]
[image]
[image]
[image]
[image]

.

gwarf, 2020-07-29:
Also as here the DNS server is configured on the router it removes the possibility to have per-device stats, it's an aggregation of all the devices from the local network.

gwarf, 2020-08-03:
I've now moved to Pi-Hole and didn't realised before that nexdns is that close/similar to Pi-hole so OpenMPTCPRouter is really killing everythign else :)

closed 2020-12-07 by github-actions[bot]