Ysurac/openmptcprouter #1102: The vps does not restarts after the refresh install on GCP

rifaterdemsahin, 2020-07-28:
## Expected Behavior
i should be able to connect.

this is what i get
[image]

## Specifications

  - OpenMPTCProuter VPS version: todays download....i cant locate the version not clear in the download and the log/

After the install developer should point the version number

  /!\ You need to reboot to enable MPTCP, shadowsocks, glorytun and shorewall /!\
------------------------------------------------------------------------------------
 After reboot, check with uname -a that the kernel name contain mptcp.
 Else, you may have to modify GRUB_DEFAULT in /etc/defaut/grub
====================================================================================

^^^ after this line

rifaterdemsahin, 2020-07-28:
the vps is dead and i cant even ssh into it as well
[image]

Ysurac, 2020-07-28:
Are you sure it is dead ? You seems to try to connect to it port 22, but after script install SSH is moved to port 65222 as said in documentation and at end of the script install.

rifaterdemsahin, 2020-07-28:
Ysurac thanks for the fast response.

1. i should have opened the issue here https://github.com/Ysurac/openmptcprouter-vps
2.I did not read the documentation in that detail. (: lol i have killed 3 servers

rifaterdemsahin, 2020-07-28:
now it started working that was my process all along
https://www.youtube.com/watch?v=FtgFZOplq0s&feature=youtu.be

Could be useful for someone who is more visual ^^^

rifaterdemsahin, 2020-07-28:
this one is the part ( upload in progress) that i have solve  the connection issue

Ysurac, 2020-07-28:
Videos seems to be marked as private ;)

rifaterdemsahin, 2020-07-28:
i have unlisted them from private
[image]

this is the main video https://youtu.be/sVyzgs3HbCo

closed 2020-07-28 by rifaterdemsahin