Ysurac/openmptcprouter #1103: Can I compile OpenMPTCProuter with a missing driver for a Wifi AP ?

Nexus83510, 2020-07-29:
Hi,
Excuse me for my bad english.
I want to integrate a driver of an added Wifi dongle to the image of OpenMPTCProuter for RPI 4.

Is it possible to compile an image for the router to work with my dongle ?

Thank you in advance.

Cordially,
Nexus8.510

Ysurac, 2020-07-29:
You can follow https://github.com/Ysurac/openmptcprouter/wiki/Create-image-for-unsupported-platform to compiole your own image.
What is the driver that you need ?

Nexus83510, 2020-07-29:
Hello !
Thank you very much for reply !
I need : rtl88x2bu-dkms driver.
I will follow your advice and create my own image ^^
Cordially

Nexus83510, 2020-07-29:
Just a question ? Can I compile my image on another system for example with Debian 10 then recover the compiled image ? And where is the compiled image ?
Or must I compile my image on Raspbian directly on my RPI 4 with the additional driver ?
Thank you in advance.
Cordially

Ysurac, 2020-07-29:
You must use another system under Debian 10 (or what you want), the images will be under x86_64/sources/bin/ subdirectory after compialtion.

Nexus83510, 2020-07-29:
Thank you very much for your awsome work and your support ;-)

Ysurac, 2020-07-29:
I'm not able to find rtl88x2bu-dkms driver for OpenWrt, what is the chipset ?

Nexus83510, 2020-07-29:
:-)
Here is the link : https://www.amazon.fr/BrosTrend-1200Mbps-Adaptateurs-Raspberry-AC1L/dp/B07FCNP2VL.
I don't know the chipset exactly. I hope the driver exists for OpenWrt.
Thanks ^^

Nexus83510, 2020-07-29:
Here is the link of managing the product : https://deb.trendtechcn.com/

Nexus83510, 2020-07-29:
Is there any way to install rtl88x2bu-dkms.deb directly on your deployed RPI 4 image please ?

Ysurac, 2020-07-29:
No, OpenMPTCProuter is based on OpenWrt not on debian/raspbian.
There is no driver under OpenWrt for the chipset (rtl8812bu), maybe rtw88 may work but I think it's only for PCI-E device.

Nexus83510, 2020-07-29:
Maybe here ? : https://forum.openwrt.org/t/drivers-for-rtl88x2bu/43299

Nexus83510, 2020-07-29:
What Wifi dongle would you advice to use with RPI 4 ?

Ysurac, 2020-07-29:
I don't really like wifi dongle, but if i refer to https://forum.openwrt.org/t/wi-fi-adapter-for-raspberry-pi-4-b/56969 dongle based on MediaTek MT7612U seems to work well.

kevin39, 2020-07-30:
Can I know why do you want to use a Wifi dongle ? 
RPI4 has an integrated chip and wifi is not required to use your differents routers.

Nexus83510, 2020-07-30:
I think you're right in fact. I will use my routers. Thank you !

closed 2020-07-30 by Nexus83510