Ysurac/openmptcprouter #1104: vpn server

wolfpcgr, 2020-07-29:
I would like to know if there is a way to create a vpn server for my network directly on the mptcp router and if so how do I do it. I would like to use open vpn if possible. 

## Specifications

  - OpenMPTCProuter version: < 0.54.7>
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter platform: <!x86_64 (RPI2/RPI3/x86/x86_64) -->

Ysurac, 2020-07-29:
Yes it's possible using the web interface, create a new VPN and don't use tun0 as interface.

tentious, 2020-07-30:
I also needed to accomplish this and also needed to route certain traffic through this VPN. To accomplish this, I added the domain to the OMR-Bypass with tun1 as interface, which worked to force the desired traffic through that VPN. Good luck!

closed 2020-12-03 by github-actions[bot]