Ysurac/openmptcprouter #1105: Added the 3 rd connection and not seeing it online

rifaterdemsahin, 2020-07-29:
## Expected Behavior
Go online

## Current Behavior

## Possible Solution
`Wed Jul 29 23:24:14 2020 user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)
Wed Jul 29 23:24:15 2020 daemon.notice openvpn(WAN3)[19402]: OpenVPN 2.4.8 x86_64-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD]
Wed Jul 29 23:24:15 2020 daemon.notice openvpn(WAN3)[19402]: library versions: OpenSSL 1.1.1d  10 Sep 2019, LZO 2.10
Wed Jul 29 23:24:15 2020 daemon.warn openvpn(WAN3)[19402]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Wed Jul 29 23:24:15 2020 daemon.notice openvpn(WAN3)[19402]: TCP/UDP: Preserving recently used remote address: [AF_INET]34.105.154.195:65301
Wed Jul 29 23:24:15 2020 daemon.err openvpn(WAN3)[19402]: TCP/UDP: Socket bind failed on local address [AF_INET]192.168.16.115:65301: Address not available (errno=99)
Wed Jul 29 23:24:15 2020 daemon.notice openvpn(WAN3)[19402]: Exiting due to fatal error
Wed Jul 29 23:24:18 2020 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.254.1 dev tun0
Wed Jul 29 23:24:19 2020 user.notice post-tracking-post-tracking: eth3 switched off because gateway down and ping from 192.168.1.3 error (4.2.2.1,8.8.8.8)
Wed Jul 29 23:24:19 2020 user.notice post-tracking-post-tracking: Delete default route via 192.168.1.1 dev eth3
Wed Jul 29 23:24:20 2020 daemon.notice openvpn(WAN3)[19999]: OpenVPN 2.4.8 x86_64-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD]
Wed Jul 29 23:24:20 2020 daemon.notice openvpn(WAN3)[19999]: library versions: OpenSSL 1.1.1d  10 Sep 2019, LZO 2.10
Wed Jul 29 23:24:20 2020 daemon.warn openvpn(WAN3)[19999]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Wed Jul 29 23:24:20 2020 daemon.notice openvpn(WAN3)[19999]: TCP/UDP: Preserving recently used remote address: [AF_INET]34.105.154.195:65301
Wed Jul 29 23:24:20 2020 daemon.err openvpn(WAN3)[19999]: TCP/UDP: Socket bind failed on local address [AF_INET]192.168.16.115:65301: Address not available (errno=99)
Wed Jul 29 23:24:20 2020 daemon.notice openvpn(WAN3)[19999]: Exiting due to fatal error
Wed Jul 29 23:24:25 2020 daemon.notice openvpn(WAN3)[20464]: OpenVPN 2.4.8 x86_64-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD]
Wed Jul 29 23:24:25 2020 daemon.notice openvpn(WAN3)[20464]: library versions: OpenSSL 1.1.1d  10 Sep 2019, LZO 2.10
Wed Jul 29 23:24:25 2020 daemon.warn openvpn(WAN3)[20464]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Wed Jul 29 23:24:25 2020 daemon.notice openvpn(WAN3)[20464]: TCP/UDP: Preserving recently used remote address: [AF_INET]34.105.154.195:65301
Wed Jul 29 23:24:25 2020 daemon.err openvpn(WAN3)[20464]: TCP/UDP: Socket bind failed on local address [AF_INET]192.168.16.115:65301: Address not available (errno=99)
Wed Jul 29 23:24:25 2020 daemon.notice openvpn(WAN3)[20464]: Exiting due to fatal error
Wed Jul 29 23:24:26 2020 user.notice post-tracking-post-tracking: Set server GCP (34.105.154.195) default route via 192.168.0.1`
## Steps to Reproduce the Problem

  1.add the 3rd connection
  2.DSL connection

[image]

## Context (Environment)
unable to add a new connection

## Specifications

  - OpenMPTCProuter version:
Powered by openmptcprouter v0.54.7 r0+12509-04a21c26a0
platfrom esxi 7

Ysurac, 2020-07-30:
You seems to have enabled MPTCP over VPN. Why ?

rifaterdemsahin, 2020-07-30:
`You seems to have enabled MPTCP over VPN. Why ?`

I am trying different variations to see if it work. I have also unchecked it. Maybe it is not saving with the wizard
[image]

I have changed with LTE instead of DSL and the 3 rd connections still does not work
[image]

Ysurac, 2020-07-30:
LTE and WAN2 are in same network, it's not a good idea.
For interfaces you should check via SSH that you can ping gateway, if it's not the case check your host configuration. If you use macvlan (so if you don't create virtuals interface in your VM manager), check that interface is set as promiscuous.

rifaterdemsahin, 2020-07-30:
I have updated the network address of the LTE not to be an issue

192.168.32.254

rifaterdemsahin, 2020-07-30:
I have deleted the node with the wizard and updated the network still the same result
[image]

`hu Jul 30 08:46:42 2020 daemon.notice openvpn(WAN3)[12921]: library versions: OpenSSL 1.1.1d  10 Sep 2019, LZO 2.10
Thu Jul 30 08:46:42 2020 daemon.warn openvpn(WAN3)[12921]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Thu Jul 30 08:46:42 2020 daemon.notice openvpn(WAN3)[12921]: TCP/UDP: Preserving recently used remote address: [AF_INET]34.105.154.195:65301
Thu Jul 30 08:46:42 2020 daemon.err openvpn(WAN3)[12921]: TCP/UDP: Socket bind failed on local address [AF_INET]192.168.16.115:65301: Address not available (errno=99)
Thu Jul 30 08:46:42 2020 daemon.notice openvpn(WAN3)[12921]: Exiting due to fatal error
Thu Jul 30 08:46:47 2020 daemon.notice openvpn(WAN3)[13227]: OpenVPN 2.4.8 x86_64-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD]
Thu Jul 30 08:46:47 2020 daemon.notice openvpn(WAN3)[13227]: library versions: OpenSSL 1.1.1d  10 Sep 2019, LZO 2.10
Thu Jul 30 08:46:47 2020 daemon.warn openvpn(WAN3)[13227]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Thu Jul 30 08:46:47 2020 daemon.notice openvpn(WAN3)[13227]: TCP/UDP: Preserving recently used remote address: [AF_INET]34.105.154.195:65301
Thu Jul 30 08:46:47 2020 daemon.err openvpn(WAN3)[13227]: TCP/UDP: Socket bind failed on local address [AF_INET]192.168.16.115:65301: Address not available (errno=99)
Thu Jul 30 08:46:47 2020 daemon.notice openvpn(WAN3)[13227]: Exiting due to fatal error
Thu Jul 30 08:46:52 2020 daemon.notice openvpn(WAN3)[13612]: OpenVPN 2.4.8 x86_64-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD]
Thu Jul 30 08:46:52 2020 daemon.notice openvpn(WAN3)[13612]: library versions: OpenSSL 1.1.1d  10 Sep 2019, LZO 2.10
Thu Jul 30 08:46:52 2020 daemon.warn openvpn(WAN3)[13612]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Thu Jul 30 08:46:52 2020 daemon.notice openvpn(WAN3)[13612]: TCP/UDP: Preserving recently used remote address: [AF_INET]34.105.154.195:65301
Thu Jul 30 08:46:52 2020 daemon.err openvpn(WAN3)[13612]: TCP/UDP: Socket bind failed on local address [AF_INET]192.168.16.115:65301: Address not available (errno=99)
Thu Jul 30 08:46:52 2020 daemon.notice openvpn(WAN3)[13612]: Exiting due to fatal error
Thu Jul 30 08:46:57 2020 daemon.notice openvpn(WAN3)[14053]: OpenVPN 2.4.8 x86_64-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD]
Thu Jul 30 08:46:57 2020 daemon.notice openvpn(WAN3)[14053]: library versions: OpenSSL 1.1.1d  10 Sep 2019, LZO 2.10
Thu Jul 30 08:46:57 2020 daemon.warn openvpn(WAN3)[14053]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Thu Jul 30 08:46:57 2020 daemon.notice openvpn(WAN3)[14053]: TCP/UDP: Preserving recently used remote address: [AF_INET]34.105.154.195:65301
Thu Jul 30 08:46:57 2020 daemon.err openvpn(WAN3)[14053]: TCP/UDP: Socket bind failed on local address [AF_INET]192.168.16.115:65301: Address not available (errno=99)
Thu Jul 30 08:46:57 2020 daemon.notice openvpn(WAN3)[14053]: Exiting due to fatal error`

Ysurac, 2020-07-30:
Did you check that you can ping gateway via SSH ?
For the Openvpn dirty log, disable it in VPN->OpenVPN.
What is the dongle used ? Did you configure it if needed ?

rifaterdemsahin, 2020-07-30:
I have also performed a reboot after remove and readd without checking the MPTCP over VPN.

Still the 3 rd line with different access points are not going online. Is there a reset system where i can restart ?

rifaterdemsahin, 2020-07-30:
and why is there an openvpn thing going on WAN3 ?

`
Thu Jul 30 08:50:32 2020 daemon.info dnsmasq-dhcp[18232]: read /etc/ethers - 0 addresses
Thu Jul 30 08:50:32 2020 user.notice omr-bypass: OMR-ByPass is running
Thu Jul 30 08:50:35 2020 user.notice post-tracking-post-tracking: Set server GCP (34.105.154.195) default route via 192.168.0.1
Thu Jul 30 08:50:37 2020 daemon.notice openvpn(WAN3)[21384]: OpenVPN 2.4.8 x86_64-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD]
Thu Jul 30 08:50:37 2020 daemon.notice openvpn(WAN3)[21384]: library versions: OpenSSL 1.1.1d  10 Sep 2019, LZO 2.10
Thu Jul 30 08:50:37 2020 daemon.warn openvpn(WAN3)[21384]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Thu Jul 30 08:50:37 2020 daemon.notice openvpn(WAN3)[21384]: TCP/UDP: Preserving recently used remote address: [AF_INET]34.105.154.195:65301
Thu Jul 30 08:50:37 2020 daemon.err openvpn(WAN3)[21384]: TCP/UDP: Socket bind failed on local address [AF_INET]192.168.16.115:65301: Address not available (errno=99)
Thu Jul 30 08:50:37 2020 daemon.notice openvpn(WAN3)[21384]: Exiting due to fatal error
Thu Jul 30 08:50:42 2020 daemon.notice openvpn(WAN3)[21691]: OpenVPN 2.4.8 x86_64-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD]
Thu Jul 30 08:50:42 2020 daemon.notice openvpn(WAN3)[21691]: library versions: OpenSSL 1.1.1d  10 Sep 2019, LZO 2.10
Thu Jul 30 08:50:42 2020 daemon.warn openvpn(WAN3)[21691]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Thu Jul 30 08:50:42 2020 daemon.notice openvpn(WAN3)[21691]: TCP/UDP: Preserving recently used remote address: [AF_INET]34.105.154.195:65301
Thu Jul 30 08:50:42 2020 daemon.err openvpn(WAN3)[21691]: TCP/UDP: Socket bind failed on local address [AF_INET]192.168.16.115:65301: Address not available (errno=99)
Thu Jul 30 08:50:42 2020 daemon.notice openvpn(WAN3)[21691]: Exiting due to fatal error
Thu Jul 30 08:50:47 2020 daemon.notice openvpn(WAN3)[22087]: OpenVPN 2.4.8 x86_64-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD]
Thu Jul 30 08:50:47 2020 daemon.notice openvpn(WAN3)[22087]: library versions: OpenSSL 1.1.1d  10 Sep 2019, LZO 2.10
Thu Jul 30 08:50:47 2020 daemon.warn openvpn(WAN3)[22087]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Thu Jul 30 08:50:47 2020 daemon.notice openvpn(WAN3)[22087]: TCP/UDP: Preserving recently used remote address: [AF_INET]34.105.154.195:65301
Thu Jul 30 08:50:47 2020 daemon.err openvpn(WAN3)[22087]: TCP/UDP: Socket bind failed on local address [AF_INET]192.168.16.115:65301: Address not available (errno=99)
Thu Jul 30 08:50:47 2020 daemon.notice openvpn(WAN3)[22087]: Exiting due to fatal error
Thu Jul 30 08:50:52 2020 daemon.notice openvpn(WAN3)[22452]: OpenVPN 2.4.8 x86_64-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD]
Thu Jul 30 08:50:52 2020 daemon.notice openvpn(WAN3)[22452]: library versions: OpenSSL 1.1.1d  10 Sep 2019, LZO 2.10
Thu Jul 30 08:50:52 2020 daemon.warn openvpn(WAN3)[22452]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Thu Jul 30 08:50:52 2020 daemon.notice openvpn(WAN3)[22452]: TCP/UDP: Preserving recently used remote address: [AF_INET]34.105.154.195:65301
Thu Jul 30 08:50:52 2020 daemon.err openvpn(WAN3)[22452]: TCP/UDP: Socket bind failed on local address [AF_INET]192.168.16.115:65301: Address not available (errno=99)
Thu Jul 30 08:50:52 2020 daemon.notice openvpn(WAN3)[22452]: Exiting due to fatal error
Thu Jul 30 08:50:57 2020 daemon.notice openvpn(WAN3)[22748]: OpenVPN 2.4.8 x86_64-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD]
Thu Jul 30 08:50:57 2020 daemon.notice openvpn(WAN3)[22748]: library versions: OpenSSL 1.1.1d  10 Sep 2019, LZO 2.10
Thu Jul 30 08:50:57 2020 daemon.warn openvpn(WAN3)[22748]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Thu Jul 30 08:50:57 2020 daemon.notice openvpn(WAN3)[22748]: TCP/UDP: Preserving recently used remote address: [AF_INET]34.105.154.195:65301
Thu Jul 30 08:50:57 2020 daemon.err openvpn(WAN3)[22748]: TCP/UDP: Socket bind failed on local address [AF_INET]192.168.16.115:65301: Address not available (errno=99)
Thu Jul 30 08:50:57 2020 daemon.notice openvpn(WAN3)[22748]: Exiting due to fatal error
Thu Jul 30 08:51:02 2020 daemon.notice openvpn(WAN3)[23070]: OpenVPN 2.4.8 x86_64-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD]
Thu Jul 30 08:51:02 2020 daemon.notice openvpn(WAN3)[23070]: library versions: OpenSSL 1.1.1d  10 Sep 2019, LZO 2.10
Thu Jul 30 08:51:02 2020 daemon.warn openvpn(WAN3)[23070]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Thu Jul 30 08:51:02 2020 daemon.notice openvpn(WAN3)[23070]: TCP/UDP: Preserving recently used remote address: [AF_INET]34.105.154.195:65301
Thu Jul 30 08:51:02 2020 daemon.err openvpn(WAN3)[23070]: TCP/UDP: Socket bind failed on local address [AF_INET]192.168.16.115:65301: Address not available (errno=99)
Thu Jul 30 08:51:02 2020 daemon.notice openvpn(WAN3)[23070]: Exiting due to fatal error
[... cut ...]
Thu Jul 30 08:51:27 2020 daemon.warn openvpn(WAN3)[25038]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Thu Jul 30 08:51:27 2020 daemon.notice openvpn(WAN3)[25038]: TCP/UDP: Preserving recently used remote address: [AF_INET]34.105.154.195:65301
Thu Jul 30 08:51:27 2020 daemon.err openvpn(WAN3)[25038]: TCP/UDP: Socket bind failed on local address [AF_INET]192.168.16.115:65301: Address not available (errno=99)
Thu Jul 30 08:51:27 2020 daemon.notice openvpn(WAN3)[25038]: Exiting due to fatal error`

rifaterdemsahin, 2020-07-30:
somehow it is left there and now i remove the openvpn setting
[image]

rifaterdemsahin, 2020-07-30:
now i got rid of the openvpn error this is the last state

`Thu Jul 30 08:53:39 2020 daemon.info dnsmasq-dhcp[18497]: read /etc/ethers - 0 addresses
Thu Jul 30 08:53:40 2020 daemon.notice procd: /etc/rc.d/S99openmptcprouter-vps: {"result":"done","reason":"no changes","route":"shadowsocks"}
Thu Jul 30 08:53:45 2020 daemon.info procd: - init complete -
Thu Jul 30 08:53:45 2020 daemon.info vnstatd[19105]: No interfaces found in database, adding available interfaces...
Thu Jul 30 08:53:45 2020 daemon.info vnstatd[19105]: Interface "ifb4tun0" added with 1000 Mbit bandwidth limit.
Thu Jul 30 08:53:45 2020 daemon.info vnstatd[19105]: Interface "eth1" added with 1000 Mbit bandwidth limit.
Thu Jul 30 08:53:45 2020 daemon.info vnstatd[19105]: Interface "wan2" added with 1000 Mbit bandwidth limit.
Thu Jul 30 08:53:45 2020 daemon.info vnstatd[19105]: Interface "eth3" added with 1000 Mbit bandwidth limit.
Thu Jul 30 08:53:45 2020 daemon.info vnstatd[19105]: Interface "eth0" added with 1000 Mbit bandwidth limit.
Thu Jul 30 08:53:45 2020 daemon.info vnstatd[19105]: Interface "ifb1" added with 1000 Mbit bandwidth limit.
Thu Jul 30 08:53:45 2020 daemon.info vnstatd[19105]: Interface "teql0" added with 1000 Mbit bandwidth limit.
Thu Jul 30 08:53:45 2020 daemon.info vnstatd[19105]: Interface "tun0" added with 10 Mbit bandwidth limit.
Thu Jul 30 08:53:45 2020 daemon.info vnstatd[19105]: Interface "eth2" added with 1000 Mbit bandwidth limit.
Thu Jul 30 08:53:45 2020 daemon.info vnstatd[19105]: Interface "ifb0" added with 1000 Mbit bandwidth limit.
Thu Jul 30 08:53:45 2020 daemon.info vnstatd[19105]: -> 10 new interfaces found.
Thu Jul 30 08:53:45 2020 daemon.info vnstatd[19105]: Limits can be modified using the configuration file. See "man vnstat.conf".
Thu Jul 30 08:53:45 2020 daemon.info vnstatd[19105]: Unwanted interfaces can be removed from monitoring with "vnstat --remove".
Thu Jul 30 08:53:45 2020 daemon.info vnstatd[19105]: Info: vnStat daemon 2.4 started. (pid:19105 uid:0 gid:0 64-bit)
Thu Jul 30 08:53:45 2020 daemon.info vnstatd[19105]: Info: Monitoring (10): wan2 (1000 Mbit) tun0 (10 Mbit) teql0 (1000 Mbit) ifb4tun0 (1000 Mbit) ifb1 (1000 Mbit) ifb0 (1000 Mbit) eth3 (1000 Mbit) eth2 (1000 Mbit) eth1 (1000 Mbit) eth0 (1000 Mbit)
Thu Jul 30 08:53:46 2020 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.251.1 dev tun0
Thu Jul 30 08:53:47 2020 user.notice post-tracking-post-tracking: wan2 switched off because gateway down and ping from 192.168.32.32 error (4.2.2.1,8.8.8.8)
Thu Jul 30 08:53:47 2020 user.notice post-tracking-post-tracking: Delete default route via 192.168.32.254 dev wan2
Thu Jul 30 08:53:53 2020 daemon.err uhttpd[6608]: luci: accepted login on / for root from 192.168.16.122`

rifaterdemsahin, 2020-07-30:
I have disabled the firewall on the device
[image]

rifaterdemsahin, 2020-07-30:
I am able to ping the router ( tp link mobile router ) which has a usb dongle
[image]

rifaterdemsahin, 2020-07-30:
And If i connect my laptop i can also ping the modem as well
[image]

Ysurac, 2020-07-30:
Can you ping it from OpenMPTCProuter via SSH ?

rifaterdemsahin, 2020-07-30:
`Thu Jul 30 08:53:34 2020 user.notice mptcp: Reloading mptcp config due to ifup of WAN1 (eth1)
Thu Jul 30 08:53:34 2020 user.notice mptcp: Reloading mptcp config due to ifup of WAN2 (eth2)
Thu Jul 30 08:53:34 2020 user.notice firewall: Reloading firewall due to ifup of lan (eth0)
Thu Jul 30 08:53:35 2020 daemon.info dsvpn[11265]: Gateway changed from [192.168.32.254] to [192.168.8.1]
Thu Jul 30 08:53:35 2020 daemon.err dsvpn[11265]: RTNETLINK answers: Permission denied
Thu Jul 30 08:53:35 2020 daemon.info dsvpn[11265]: Connecting to 34.105.154.195:65401...
Thu Jul 30 08:53:35 2020 user.notice mptcp: Reloading mptcp config due to ifup of lan (eth0)
Thu Jul 30 08:53:35 2020 daemon.info dsvpn[11265]: Connected
Thu Jul 30 08:53:35 2020 user.notice post-tracking-post-tracking: Master up : Replace default route by 192.168.0.1 dev eth2
Thu Jul 30 08:53:35 2020 user.notice post-tracking-post-tracking: Set server GCP (34.105.154.195) route via 192.168.0.1 metric 8
Thu Jul 30 08:53:35 2020 user.notice post-tracking-post-tracking: Can't find Shadowsocks, restart it...
Thu Jul 30 08:53:35 2020 daemon.notice procd: /etc/rc.d/S95done: Failed to initialize libusb: LIBUSB_ERROR_OTHER
Thu Jul 30 08:53:35 2020 user.notice Shadowsocks: Rules DOWN
Thu Jul 30 08:53:35 2020 user.notice post-tracking-post-tracking: Set server GCP (34.105.154.195) route via 192.168.8.1 metric 7
Thu Jul 30 08:53:35 2020 user.err vnstat_backup: cannot restore, backup file does not exist (yet)
Thu Jul 30 08:53:35 2020 user.notice adb-enablemodem: INFO: waiting for device
Thu Jul 30 08:53:35 2020 daemon.notice procd: /etc/rc.d/S99omr-6in4: net.ipv6.conf.all.disable_ipv6 = 1
Thu Jul 30 08:53:35 2020 user.notice omr-6in4: Disable IPv6
Thu Jul 30 08:53:35 2020 user.notice post-tracking-post-tracking: Can't find Shadowsocks, restart it...
Thu Jul 30 08:53:35 2020 user.notice omr-bypass: Starting OMR-ByPass...
Thu Jul 30 08:53:36 2020 user.notice Shadowsocks: Rules UP`

rifaterdemsahin, 2020-07-30:
> Can you ping it from OpenMPTCProuter via SSH ?

yes from my laptop
[image]

rifaterdemsahin, 2020-07-30:
Debug
Gateway changed from [192.168.32.254] to [192.168.8.1]

[192.168.32.254]  : is my tplink portable router
192.168.8.1 is the usb modem ip that is connected to the tplink

rifaterdemsahin, 2020-07-30:
I suspect here is the issue
[image]

Ysurac, 2020-07-30:
It's not the issue.
Please try to ping via SSH on the OpenMPTCProuter the modem/box that is not working (not via a portable or anything else).
Give me also the result of ```ip a``` and ```ip r```.

rifaterdemsahin, 2020-07-30:
`Please try to ping via SSH on the OpenMPTCProuter the modem/box that is not working (not via a portable or anything else).`

the new one is not reachable
[image]

I have changed the networks one to 32 ( uses 8 in the back on usb dongle) and 64
[image]

rifaterdemsahin, 2020-07-30:
the issue is i can ping from my laptop but not from mptcp router machine

rifaterdemsahin, 2020-07-30:
`Give me also the result of ip a and ip r.`

[image]

[image]

rifaterdemsahin, 2020-07-30:
latest logs

> hu Jul 30 09:19:01 2020 daemon.info dnsmasq[28442]: read /etc/hosts - 4 addresses
Thu Jul 30 09:19:01 2020 daemon.info dnsmasq[28442]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Thu Jul 30 09:19:01 2020 daemon.info dnsmasq-dhcp[28442]: read /etc/ethers - 0 addresses
Thu Jul 30 09:19:03 2020 user.notice omr-bypass: Starting OMR-ByPass...
Thu Jul 30 09:19:03 2020 user.notice omr-bypass: Restart dnsmasq...
Thu Jul 30 09:19:03 2020 daemon.info dnsmasq[28442]: exiting on receipt of SIGTERM
Thu Jul 30 09:19:03 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Thu Jul 30 09:19:03 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Thu Jul 30 09:19:03 2020 user.notice dnsmasq: Allowing RFC1918 responses for domain plex.direct
Thu Jul 30 09:19:03 2020 daemon.info dnsmasq[29805]: started, version 2.81rc3 cachesize 150
Thu Jul 30 09:19:03 2020 daemon.info dnsmasq[29805]: DNS service limited to local subnets
Thu Jul 30 09:19:03 2020 daemon.info dnsmasq[29805]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC no-ID loop-detect inotify dumpfile
Thu Jul 30 09:19:03 2020 daemon.info dnsmasq-dhcp[29805]: DHCP, IP range 192.168.16.113 -- 192.168.16.192, lease time 12h
Thu Jul 30 09:19:03 2020 daemon.info dnsmasq[29805]: using only locally-known addresses for domain test
Thu Jul 30 09:19:03 2020 daemon.info dnsmasq[29805]: using only locally-known addresses for domain onion
Thu Jul 30 09:19:03 2020 daemon.info dnsmasq[29805]: using only locally-known addresses for domain localhost
Thu Jul 30 09:19:03 2020 daemon.info dnsmasq[29805]: using only locally-known addresses for domain local
Thu Jul 30 09:19:03 2020 daemon.info dnsmasq[29805]: using only locally-known addresses for domain invalid
Thu Jul 30 09:19:03 2020 daemon.info dnsmasq[29805]: using only locally-known addresses for domain bind
Thu Jul 30 09:19:03 2020 daemon.info dnsmasq[29805]: using only locally-known addresses for domain use-application-dns.net
Thu Jul 30 09:19:03 2020 daemon.info dnsmasq[29805]: using nameserver 127.0.0.1#5353
Thu Jul 30 09:19:03 2020 daemon.info dnsmasq[29805]: using only locally-known addresses for domain lan
Thu Jul 30 09:19:03 2020 daemon.info dnsmasq[29805]: read /etc/hosts - 4 addresses
Thu Jul 30 09:19:03 2020 daemon.info dnsmasq[29805]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Thu Jul 30 09:19:03 2020 daemon.info dnsmasq-dhcp[29805]: read /etc/ethers - 0 addresses
Thu Jul 30 09:19:03 2020 user.notice omr-bypass: OMR-ByPass is running
Thu Jul 30 09:19:12 2020 user.notice post-tracking-post-tracking: eth1 switched off because ping from 192.168.64.111 error (4.2.2.1,8.8.8.8)
Thu Jul 30 09:19:12 2020 user.notice post-tracking-post-tracking: Delete default route via 192.168.64.1 dev eth1
Thu Jul 30 09:19:14 2020 user.notice post-tracking-post-tracking: wan2 switched off because gateway down and ping from 192.168.32.32 error (4.2.2.1,8.8.8.8)
Thu Jul 30 09:19:14 2020 user.notice post-tracking-post-tracking: Delete default route via 192.168.32.254 dev wan2
Thu Jul 30 09:22:47 2020 auth.info login[2445]: root login on 'tty1'
Thu Jul 30 09:30:29 2020 daemon.info dnsmasq-dhcp[29805]: DHCPDISCOVER(eth0) d0:03:df:f8:6c:00
Thu Jul 30 09:30:29 2020 daemon.info dnsmasq-dhcp[29805]: DHCPOFFER(eth0) 192.168.16.115 d0:03:df:f8:6c:00
Thu Jul 30 09:30:29 2020 daemon.info dnsmasq-dhcp[29805]: DHCPDISCOVER(eth0) d0:03:df:f8:6c:00
Thu Jul 30 09:30:29 2020 daemon.info dnsmasq-dhcp[29805]: DHCPOFFER(eth0) 192.168.16.115 d0:03:df:f8:6c:00
Thu Jul 30 09:30:29 2020 daemon.info dnsmasq-dhcp[29805]: DHCPREQUEST(eth0) 192.168.16.115 d0:03:df:f8:6c:00
Thu Jul 30 09:30:29 2020 daemon.info dnsmasq-dhcp[29805]: DHCPACK(eth0) 192.168.16.115 d0:03:df:f8:6c:00 localhost
Thu Jul 30 09:30:29 2020 daemon.warn dnsmasq-dhcp[29805]: not giving name localhost.lan to the DHCP lease of 192.168.16.115 because the name exists in /etc/hosts with address 127.0.0.1
Thu Jul 30 09:30:29 2020 daemon.warn dnsmasq-dhcp[29805]: not giving name localhost to the DHCP lease of 192.168.16.115 because the name exists in /etc/hosts with address 127.0.0.1
Thu Jul 30 09:31:15 2020 daemon.err /usr/bin/ss-redir[22062]: send: Connection reset by peer
Thu Jul 30 09:32:58 2020 daemon.err /usr/bin/ss-redir[22062]: server recv: Host is unreachable
Thu Jul 30 09:33:39 2020 daemon.info dnsmasq-dhcp[29805]: DHCPREQUEST(eth0) 192.168.16.144 00:11:32:b4:68:b2
Thu Jul 30 09:33:39 2020 daemon.info dnsmasq-dhcp[29805]: DHCPACK(eth0) 192.168.16.144 00:11:32:b4:68:b2 SINGULARITY
Thu Jul 30 09:33:39 2020 daemon.warn dnsmasq-dhcp[29805]: not giving name localhost.lan to the DHCP lease of 192.168.16.115 because the name exists in /etc/hosts with address 127.0.0.1
Thu Jul 30 09:33:39 2020 daemon.warn dnsmasq-dhcp[29805]: not giving name localhost to the DHCP lease of 192.168.16.115 because the name exists in /etc/hosts with address 127.0.0.1
Thu Jul 30 09:38:13 2020 daemon.err /usr/bin/ss-redir[22061]: server recv: Host is unreachable
Thu Jul 30 09:46:25 2020 daemon.info dnsmasq-dhcp[29805]: DHCPREQUEST(eth0) 192.168.16.143 00:11:32:b4:68:b1
Thu Jul 30 09:46:25 2020 daemon.info dnsmasq-dhcp[29805]: DHCPACK(eth0) 192.168.16.143 00:11:32:b4:68:b1 SINGULARITY
Thu Jul 30 09:46:25 2020 daemon.warn dnsmasq-dhcp[29805]: not giving name localhost.lan to the DHCP lease of 192.168.16.115 because the name exists in /etc/hosts with address 127.0.0.1
Thu Jul 30 09:46:25 2020 daemon.warn dnsmasq-dhcp[29805]: not giving name localhost to the DHCP lease of 192.168.16.115 because the name exists in /etc/hosts with address 127.0.0.1
Thu Jul 30 10:00:00 2020 cron.err crond[6045]: USER root pid 14806 cmd /etc/init.d/vnstat_backup backup

rifaterdemsahin, 2020-07-30:
I suppose too many deletions created the issue
[image]

Ysurac, 2020-07-30:
You should do a fresh install.

rifaterdemsahin, 2020-07-30:
Thu Jul 30 15:05:43 2020 daemon.info dsvpn[11268]: Trying to reconnect
**Thu Jul 30 15:05:44 2020 daemon.info dsvpn[11268]: Gateway changed from [192.168.64.1] to [192.168.0.1]**
Thu Jul 30 15:05:45 2020 daemon.err dsvpn[11268]: RTNETLINK answers: Permission denied
Thu Jul 30 15:05:45 2020 daemon.info dsvpn[11268]: Connecting to 34.105.154.195:65401...
Thu Jul 30 15:05:45 2020 daemon.info dsvpn[11268]: Connected
Thu Jul 30 15:05:49 2020 user.notice post-tracking-post-tracking: Tunnel up : Replace default route b

The huwaie 5g router has a card inside ...and I can only change the shell not the core
shell ip is 192.168.64.1
core ip is 192.168.0.1 and it gets overlapped with my fiber network.

How can i solve that issue.

My router
https://consumer.huawei.com/en/routers/5g-cpe-pro/

My usecase at the moment working on it.
Here is how i got stuck
https://www.wevideo.com/view/1781230112

Ysurac, 2020-07-30:
Why are you using dsvpn ?
While it's not working, you should keep all defaults settings. 

The gateway is one of your defined IP. If IPs are already used, change the subnet.

rifaterdemsahin, 2020-07-30:
`Why are you using dsvpn ?`

no idea (: and i dont know what the  default vpn is

Ysurac, 2020-07-30:
glorytun tcp is the default VPN.

rifaterdemsahin, 2020-07-30:
[image]

When i moved the 5g one from 192.168.8.1 to 192.168.64.1 with 255.255.255.0  it stopped working now. I have 2 connections. will retry to do add the 3 rd one.

rifaterdemsahin, 2020-07-30:
> The gateway is one of your defined IP. If IPs are already used, change the subnet.

[image]

like this ^^^ I have also changed it in the usb dongle as well

Ysurac, 2020-07-30:
Gateway need to be in subnet, with your netmask it's not the case.

rifaterdemsahin, 2020-07-30:
`Gateway need to be in subnet, with your netmask it's not the case.
`
realized it was too small but how can i 2 subnet in the same network with different subnet masks but not overlapping each other ( I suppose basic networking question )

I can ping the device on my laptop but not over the ssh in the esxi that is the issue i suppose. As it is colliding with the first connection. 

```
u Jul 30 16:39:16 2020 user.notice post-tracking-post-tracking: wan1 switched off because gateway down and ping from 192.168.0.111 error (4.2.2.1,8.8.8.8)
Thu Jul 30 16:39:16 2020 user.notice post-tracking-post-tracking: Delete default route via 192.168.0.254 dev wan1
Thu Jul 30 16:41:32 2020 daemon.info dnsmasq-dhcp[9088]: DHCPREQUEST(eth0) 192.168.16.114 18:e8:29:2c:c7:05
Thu Jul 30 16:41:32 2020 daemon.info dnsmasq-dhcp[9088]: DHCPACK(eth0) 192.168.16.114 18:e8:29:2c:c7:05
```

I changed the network on the usb dongle to 192.168.3.1 with 255.255.255.0 subnet mask.

rifaterdemsahin, 2020-07-30:
```
hu Jul 30 17:04:02 2020 user.notice omr-bypass: OMR-ByPass is running
Thu Jul 30 17:04:14 2020 user.notice post-tracking-post-tracking: wan1 switched off because gateway down and ping from 192.168.3.13 error (4.2.2.1,8.8.8.8)
Thu Jul 30 17:04:14 2020 user.notice post-tracking-post-tracking: Delete default route via 192.168.3.254 dev wan1
```

rifaterdemsahin, 2020-07-30:
for some reason i lost the port in esxi maybe when i rollbacked the snapshot
[image]

rifaterdemsahin, 2020-07-30:
now the card is added
[image]

Thu Jul 30 17:14:28 2020 daemon.info dnsmasq[30273]: using nameserver 127.0.0.1#5353
Thu Jul 30 17:14:28 2020 daemon.info dnsmasq[30273]: using only locally-known addresses for domain lan
Thu Jul 30 17:14:28 2020 daemon.info dnsmasq[30273]: read /etc/hosts - 4 addresses
Thu Jul 30 17:14:28 2020 daemon.info dnsmasq[30273]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Thu Jul 30 17:14:28 2020 daemon.info dnsmasq-dhcp[30273]: read /etc/ethers - 0 addresses
Thu Jul 30 17:14:41 2020 user.notice post-tracking-post-tracking: eth3 switched off because gateway down and ping from 192.168.3.101 error (4.2.2.1,8.8.8.8)
Thu Jul 30 17:14:41 2020 user.notice post-tracking-post-tracking: Delete default route via 192.168.3.254 dev eth3

[image]

I got tired i am giving up for today.

rifaterdemsahin, 2020-07-30:
@Ysurac would you mind if i create a video tutorial on how to use openmptcp and i would also learn at the same time as well?

rifaterdemsahin, 2020-07-30:
[image]

they ok look for me but might be missing something basic.

rifaterdemsahin, 2020-07-30:
I suspect maybe that network interface (usb is problematic)
[image]

rifaterdemsahin, 2020-07-30:
i moved to a different network card and it worked i suppose the usb card is broken some how (👍 
[image]

closed 2020-07-30 by rifaterdemsahin

rifaterdemsahin, 2020-07-30:
I have figured out the issue it is not because the network card is broken if we install the same model of the network card i think esxi can not differerantiate between each other. When i installed 3 different brands than they worked
[image]

rifaterdemsahin, 2020-07-30:
Second find out if i use 5 metre cable to extend the usb dongle this fails due to 1 milisecond error
[image]