Ysurac/openmptcprouter #1106: VPN TUNNEL DOWN

philoun8, 2020-07-30:
Nom d'hôte	OpenMPTCProuter
Modèle	       Raspberry Pi 4 Model B Rev 1.2
Architecture	?
Version du micrologiciel	openmptcprouter v0.54.7 r0+12509-04a21c26a0 / LuCI Master git-20.069.54281-3fb8e3d
Version du noyau	4.19.108

Bonjour a tous

je suis tres content du system qui fonctionne tres bien et qui correspond a mes attentes.
Habitant en pleine campagne, je fais tourner un modem huawei en 4G connecté a une antenne exterieur. debit tres correct qui oscille entre 25 et 80Mbits/s et un modem freebox en adsl 2Mbits/s,obligatoire pour avoir le data 4G en illimité.  
Donc openmptcprouteur me permet d'avoir la vitesse de l'adsl et la stabilité + l'ouverture des ports.

[image]

maintenant, au soucis! hihihi

J'ai voulu rajouter des tel en mode modem pour augmenter l'agregation mais des que je mets les tels, je passe en VPN TUNNEL DOWN et donc je perd l'acces de l'exterieur car les ports ne s'ouvrent plus

Ysurac, 2020-07-30:
Il faut faire attention que l'interface en master soit bien l'interface la plus stable. Au besoin dans Système->OpenMPTCProuter et onglet "Paramètres avancés" mettre Sélection de l'interface maître à "Pas de changement" puis dans Réseau->MPTCP régler l'interface maitre à celle désirée.
Sinon il faut voir dans Etat->Journal système les erreurs qu'il peut y avoir.

philoun8, 2020-07-30:
je ne peux pas mettre en maitre mais uniquement en client le reseau sans fil sinon je n'ai plus rien sur le wifi

Ysurac, 2020-07-30:
Je parle de la configuration Multipath, pas de wifi.

philoun8, 2020-07-30:
ok, désolé

closed 2020-12-03 by github-actions[bot]