Ysurac/openmptcprouter #1108: Hardware Selection for OpenMpTCP

rifaterdemsahin, 2020-07-30:
Currently i use a skull canyon to host 5 connections and it is messy with usb adapters on esxi.

When i check the mhz it looks like no more than 400 Mhz when i process 400MBits/sec

Can I have the assumption that a device like this would work ok without issues as it is 1.4 Ghz till 1.4GBit/sec

https://www.amazon.co.uk/Kettop-Mi3865L-Untangle-Celeron-Gigabit/dp/B07FBZ87NJ/ref=sr_1_1?dchild=1&keywords=openwrt+intel&qid=1596132475&sr=8-1

How do i know if the device would work which openmptcp ?

Ysurac, 2020-07-30:
Something like this should work and it have AES-NI support and this really help, but it's not really cheap ;)

rifaterdemsahin, 2020-07-30:
`it's not really cheap ;)`
I agree my focus to is to improve the internet speed and stability

`AES-NI support`
The AES-NI instruction set extensions are used to optimize encryption and decryption algorithms on select Intel and AMD processors. Intel announced AES-NI in 2008 and released supported CPUs late 2010 with the Westmere architecture. AMD announced and shipped AES-NI support in 2010, starting with Bulldozer.

I am getting aggravate by esxi losing the network adapters on skull canyon. it is really not build to be a router.

I am comparing 
https://www.supermicro.com/en/products/chassis/1u/505/SC505-203B as a board
https://www.amazon.co.uk/gp/product/B01H6NE4X2/ref=ox_sc_act_title_4?smid=ADD8GDUWB40L7&psc=1 and get as a network card

and 
https://www.amazon.co.uk/Kettop-Mi3865L-Untangle-Celeron-Gigabit/dp/B07FBZ87NJ/ref=sr_1_1?dchild=1&keywords=openwrt+intel&qid=1596132475&sr=8-1

my goal is to get a sustained 600 MBits this year and have the infra to be able to increase speed by 30 percent every year on the same architecture. OpenMpTCP is very promising

rifaterdemsahin, 2020-07-30:
I am also considering installing on Edgerouter 12p
https://eu.store.ui.com/products/edgerouter-12p

I dont know if i can

gnosoz, 2020-07-30:
Guys, the main problem you are facing is based on the chipset speed of the
network card rather than processors speed / ram. Regardless of RJ45 port
number it all depends on the chipset processing the interrupts. separate
RJ45 cards gives overall better speed in my humble opinion and tests.
most routers and switches aren't built with MPCTP in mind and therefore the
less ports the better. More lan ports means less overall throughput, I am
basing this assumption on pure tests only.

the majority of cards for home are with 1.5gbps throughput altogether so if
you have a lot of traffic in the lan wan1 wan2 wan3 the total in any given
moment will never be greater than 1.5gbps.
if in this condition you have 3 fiber optics giving you 1gbps each in the
lan your max will still be 4-600Mbps given the fact you are using at the
same time the same chip for the wans.  this is pure physics I am afraid and
the only way out is to bypass that chipset adding separate network cards.
The USB can be another wayout but has the same principle as it all depends
on the usb chipset there...

Ysurac, 2020-07-31:
Edgerouter 12p is not supported by OpenWrt, so it will not be supported by OpenMPTCProuter.

Processor speed is important for encryption, at least if there is no AES-NI support.
Else internal switch speed can be important if you want a big aggregated speed. There is now some with 2.5gbps (like BPI-R64, not yet supported).

rifaterdemsahin, 2020-08-01:
`Edgerouter 12p is not supported by OpenWrt, so it will not be supported by OpenMPTCProuter.
`
bummer! Now i am thinking to use that device as the fallback router if OMR fails.

`Else internal switch speed can be important if you want a big aggregated speed. There is now some with 2.5gbps (like BPI-R64, not yet supported).
`
I am thinking to use 12p and connect to OMR with one cable and creating 5 virtual nics in ESXI which would end up in different wan connections. I don't know if i can I wanna try out.

it looks like this

[image]

my rationale : I wanna be able to use OMR as the main gateway but when it fails i wanna fall back on the classic 12p ( sticky connections based action )

PS: Plugging to many usb ethernet cards is not a good idea for ESXI as it loses the connection at restarts.I wanna use the dedicated only one ethernet connection from the ESXI and divide it into 5 virtual ports where they would end up using only one connection from 12P. I am not sure that kind of forwarding could be setup using OMR and if it can detect the signals from the modem.( that part is not complete in my mind )

rifaterdemsahin, 2020-08-01:
https://www.youtube.com/watch?v=UqUE1xiuKEY

> it looks like this

jhackeur, 2020-08-26:
Great thread!

What do you think about this one:
https://fr.aliexpress.com/item/4000480680892.html?spm=a2g0o.productlist.0.0.4ef545a3xQVEwQ&algo_pvid=e4cf825a-597a-4946-a325-27438a1549c9&algo_expid=e4cf825a-597a-4946-a325-27438a1549c9-1&btsid=0b0a119a15984337175543505e7674&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_
160€

Should works ?

Adorfer, 2020-08-26:
Concerning the "Kettop" (they are sold as e.g. "NRG Systems IPU661"), i'd go for an i3. 

As gnosoz pointed out: 

 >   Regardless of RJ45 port number it all depends on the chipset processing the interrupts. separate RJ45 cards gives overall better speed in my humble opinion and tests.

the layout of the Kettop is "one rj45 goes to one dedicated intel nic which is exclusivly asinged a PCIe lane", which sounds very promising. 
(e.g. if you like an IRQ/PCIe on the minipcie slot for wifi: requiring you to reflash the bios to disable eth6, since they do not like to share lanes/irqs)

https://www.amazon.co.uk/Kettop-Mi7100L6-Firewall-Support-Barebone/dp/B07L5QMW83

jhackeur, 2020-08-26:
> thank you for the very unrefined, unsanitized url, splease be bit more specific what you mean by "this one"
> [image]

Indeed, sorry for the URL - solved.
I mean Pentuim 4405U with 4GB RAM

Adorfer, 2020-08-27:
For evaluation i ordered this one. i will come back with results. 
[image]

jhackeur, 2020-08-27:
> 
> 
> For evaluation i ordered this one. i will come back with results.
> [image]

I'm looking forward for your feedback :)

rifaterdemsahin, 2020-09-04:
i gave up working on my skull canyon the usb ethernet cards and ESXI is not stable so I am planning to buy
[image]

I dont know if this one would work
https://www.amazon.co.uk/gp/product/B076B6SWG5/ref=as_li_qf_asin_il_tl?ie=UTF8&tag=rifaterdemsah-21&creative=6738&linkCode=as2&creativeASIN=B076B6SWG5&linkId=60fd0178aa505784d9006093b8d000f1

jhackeur, 2020-09-05:
> i gave up working on my skull canyon the usb ethernet cards and ESXI is not stable so I am planning to buy
> [image]
>
> I dont know if this one would work [...]

Realy expansive compare to device founded on Aliexpress.
If device ordered by "Adorfer" works fine, that's the better option !

Adorfer, 2020-09-10:
> 
> 
> For evaluation i ordered this one. i will come back with results.
> [image]

not to keep you waiting for too long: 

  - unit arrived in good condition including decent PSU
  - there was a windows10 preinstalled, which boots within seconds due to EFI
  - bios is really good with all sorts of settings you would expect in a server board (redirect console/bios to COM-port etc)
  - install of OMR worked with a simple "dd" to the installed kingston-SSD-modules
  - LAN1-LAN5 recognized as eth0-eth4, 
  - no luck with LAN6 yet. documentation says "one PCIe-lane per interface", and this "last one" is reserverd for the pcie-slot (wifi....), there is an alternative bios to be flashed to use the lan6 (as far as i understood so far)

no time for a full setup so far, but at the moment it looks totally fine and working as expected.
i keep you posted. for benchmark. 

[image]

rifaterdemsahin, 2020-09-17:
thanks mate what the max speed did you get combining ?

having windows on it can be a plus for the multi purpose usecase.

`install of OMR worked with a simple "dd" to the installed kingston-SSD-modules
`

this is great news. Any ali baba link for this device ?

rifaterdemsahin, 2020-09-17:
price gets higher with a better core
[image]

I need to use this device and build a 2 GBit / sec network on top of it. Pentium would not cut it. ( Assumption )

Adorfer, 2020-09-17:
Pentium4405U, 4GB RAM, 64GB SSD, 
3 WAN: 200/20, 50/10, 50/10. 
Getting alsways maximum speed on speedtests, a little shy of 300/40

CPU never coming out of idle: 
[image]
RAM use... irrelevant
[image]
Filesystem: i have not resized the default-partition of 512MB... since: no need. 
[image]

In other words: I do not want to discourage you from purchasing an i3 or even i5 with large SSD and several GB of RAM. 
For ORM on 1000TX LAN: Definitly not needed! thie 4405U would probably be able to handle more than 2GBit, if it had an interface for that or i would be able to figure out how to LAG towards the switch.

openaspace, 2020-10-06:
> Pentium4405U, 4GB RAM, 64GB SSD,
> 3 WAN: 200/20, 50/10, 50/10.
> Getting alsways maximum speed on speedtests, a little shy of 300/40
> 
> CPU never coming really out of idle (you see the load during daytime hourse when there are people in the office. In plus there a constant upstream of about 12MBit/s 24/7 for a Cloud-CCTV system.)
> [image]
> RAM use... irrelevant
> [image]
> Filesystem: i have not resized the default-partition of 512MB... since: no need.
> [image]
> 
> In other words: I do not want to discourage you from purchasing an i3 or even i5 with large SSD and several GB of RAM.
> For ORM on 1000TX LAN: Definitly not needed! thie 4405U would probably be able to handle more than 2GBit, if it had an interface for that or i would be able to figure out how to LAG towards the switch.

Hi,  a courtesy please, you can write me the exact motherboard model?
Really thank you.

Adorfer, 2020-10-06:
no i can not, but here you have a photo from the device: 
[image]

rifaterdemsahin, 2020-10-09:
` For ORM on 1000TX LAN: Definitly not needed! thie 4405U would probably be able to handle more than 2GBit, if it had an interface for that or i would be able to figure out how to LAG towards the switch.`

I am considering  CPU: Intel Dual Core i5 7200U, 64 bit, up to 3.1GHz, 3MB Smart Cache, Intel AES-NI hardware support 

the reason is if the OMR does not work i could turn it into a pfsense box. I am focused on getting the best possible upload speeds to upload content to youtube.

Adorfer, 2020-11-21:
just for note, i have now the 3rd of the is p4405u in use, this was a test on 800/50 + 100/25
[image]

Idle with the supplied PSU is around 10W, on upload it peaks to 16W, 12W seems to be typical. 
[image]

closed 2021-03-03 by github-actions[bot]