Ysurac/openmptcprouter #1109: kobo

philoun8, 2020-07-30:
Bonsoir a tous

est ce normal que certains sites ne sont pas accessibles, comme kobo.com? même apres avoir reglé dans Service / OMR-Bypass

[image]

Ysurac, 2020-07-30:
Il faut plutôt mettre juste "kobo.com" dans les domaines à by-passer, ça permettra de bypasser aussi tous les sous-domaines (donc www. et autres). Pour certains sites d'autres domaines peuvent êtres à by-passer selon ce que donne les options développeur de ton navigateur web.

Quelques sites bloquent les IPs de la majorité des hébergeurs de VPS/dédié pour des raisons économiques/légales (par exemple netflix), suite à des abus (par exemple jeuxvideo.com) ou par incompétence des administrateurs (par exemple boulanger.com mais ça concerne pas mal de sites).

philoun8, 2020-07-30:
netflix j'ai réussi a le contourner (pour le moment); juste de temps en temps un message comme quoi je possède un proxy!
par contre kobo, j'arrive a l'ouvrir un fois sur dix a peu près. ou alors je dérègle mes paramètres pc pour repasser en ip static sur le modem 4G

Ysurac, 2020-07-30:
Pour kobo, en mettant juste "kobo.com" dans les domaines j'ai accès au site depuis un VPS scaleway.

philoun8, 2020-07-30:
mon vps est chez ovh, je ne sais pas si ca change quelque chose!

Ysurac, 2020-07-30:
Pas vraiment. Par contre il faut bien vérifier que le DNS utilisé sur l'ordinateur client est bien l'adresse IP d'OpenMPTCProuter.
Si l'adresse est en DHCP pas de soucis normalement.

philoun8, 2020-07-30:
je viens de regarder boulanger.com et par contre chez moi aucun soucis

Ysurac, 2020-07-30:
Ils bannissent pas toutes les plages d'IPs.
Pour que omr-bypass fonctionne il faut que la résolution du domaine se fasse en passant par OpenMPTCProuter, donc ça peut ne pas fonctionner juste après le changement du fait du cache DNS du navigateur.

kevin39, 2020-07-30:
Il a laissé interface "Par défaut". Ca fait quoi l'option par défaut dans OMR Bypass ? 
J'ai tjrs sélectionné un WAN précis quand je l'ai pris...

philoun8, 2020-07-30:
par défaut ca sélectionne toutes les interfaces

philoun8, 2020-07-30:
par contre dans "Protocoles et services", je choisi qu'elle modem je veux selectionner.

[image]

philoun8, 2020-07-31:
Bonjour
donc rapport ce matin, apres modif des parametres dns = les sites apparaissent bien.
merci

closed 2020-07-31 by Ysurac