Ysurac/openmptcprouter #1111: proxmox

philoun8, 2020-07-31:
Bonjour

Est il possible de mettre openmptcprouter sur une VM ou CT proxmox? 
je possède un serveur proliant et j'ai test l'installation d'une image openmptcprouter wmvare mais je n'ai pas réussi a le faire fonctionner

[image]

Ysurac, 2020-07-31:
Oui ça fonctionne sous proxmox, à condition d'utiliser KVM et pas LXC.

philoun8, 2020-07-31:
avec une image vmware?

Ysurac, 2020-07-31:
Je sais que ça fonctionne avec une image raw, le reste je n'ai pas testé.

Adorfer, 2020-07-31:
create a VM with a raw hdd (in the filesystem, not in ZFS etc)
you have to use an x64 images and unpack the "img.gz" file.
stop the vm (if started by accident), cp the unpacked image file over the /var/lib/vz/$id/$rawimagename. 
(you have to determine the $id from the VM number and the rawimage is the file you hopefully find exsting. Do not care about filesizes, just plain copy it over, the appropriate size will be detected by kvm during next start.

faktori, 2020-08-02:
Bonjour,
Tu peux installer ubuntu 18.04 puis tu fait l'installation du script et voila :)

philoun8, 2020-08-02:
Bonjour
de quel script tu parles Faktori?

faktori, 2020-08-03:
Si vous avez choisi Debian :
wget -O - http://www.openmptcprouter.com/server/debian9-x86_64.sh | sh

Si vous avez choisi Ubuntu :
wget -O - http://www.openmptcprouter.com/server/ubuntu18.04-x86_64.sh | sh

kevin39, 2020-08-03:
Non ne fais pas cela. C'est pour installer la partie serveur et non la partie routeur.
Si tu veux installer ton routeur dans une VM Proxmox, tu dois utiliser une VM (kvm).
Sois tu charge un disque raw sois tu convertis l'image ova/vmware. Ca devrait suffir.

Le wiki de Proxmox explique comment charger un disque ova/vmware.

philoun8, 2020-08-03:
Bonjour

j'ai essayé avec un fichier kvm et en suivant les explications proxmox (de plusieurs façons) mais ca ne tourne pas normalement...

faktori, 2020-08-03:
Ha oui je parle de la version serveur moi pas routeur 
Je pensais que tu parlais pour faire l'agregation 

désolé

Adorfer, 2020-08-03:
o.k., we are now into the n-th loop of a) finding out out what the initial request was and b) explaining things which was not initally requested. 

@philoun8 for clairities sake: 
1) select a decent title! this is a bug report? a feature request? or (as i assume) a help/support request (you can still edit that)
2) please state what you try to do (setting up the routeur part in a proxmox vm or the server part in a proxmox vm?)
3) if it's not working please be specific with wout error message. _"does not turn"_ is not helpful. otherwise people will suggest: _"have you tried turning it off and on again?"_

Eliasdevlb, 2020-12-06:
unpack the gz so you can have a .img file. Upload the image into proxmox using winscp or whatever you feel comfortable using. cd into the directory where you placed your file then use the following shell command (I am using openmptcprouter.img as the image name and I have vm id 101, please change accordingly)
qm importdisk 101 openmptcprouter.img local-lvm

philoun8, 2020-12-07:
Merci beaucoup Eliasdevlb

closed 2021-03-13 by github-actions[bot]

philoun8, 2021-11-19:
bonsoir a tous
j'ai repris mes investigations sur une vm proxmox en kvm et ca tourne nickel (enfin)…
par contre j'ai un soucis au niveau des carte physique reseau. je m'explique.
quand le serveur est démarré, je lance la vm openmptcprouter et je débranche mon raspberry qui contenait openmptcp jusqu'à maintenant. la vm prend la relève du rasp et tout fonctionne impeccable, même mieux qu'avec le rasp.
par contre si je reboot le serveur (proliant dl380gen6), la vm openmptcp prend le control de la carte physique et me bloque l'accès au webif et même ssh du serveur (qui est pourtant sur la même plage ip).
vu que le serveur a plusieurs cartes réseau j'ai donc attribuée une carte reseau juste a la vm openmptcprouter mais rien n'y fait!!!
si quelqu'un a une idée!!! merci

Eliasdevlb, 2022-07-22:
You should set your boot order and boot delay. That should fix your issue.