Ysurac/openmptcprouter #1112: DMZ Problem

florianlog, 2020-07-31:
Hello everyone, I have a problem with a fresh installation, the connection, speed, everything works correctly, my problem being that the DMZ redirection unfortunately does not work.

Here is my connection:

VPS (MilkyWan) -> OMR (WRT3200ACM) coupled with 2 fibers and a 4G FREE -> PFSENSE -> LAN.

All of this works wonderfully.

So I uncommented the 2 lines in "/ etc / shorewall / rules" and created a DMZ in "Network / Firewall / port forwarding" all the ports from the VPN to the ip of the pfsense and nothing, impossible to connect to my network from outside.

I reinstalled the VPS as well as my router without success

I'm not sure what to do except remove OMR which I don't want.

  - OpenMPTCProuter version: 0.54.7
  - OpenMPTCProuter VPS version: 0.1015
  - OpenMPTCProuter platform:  WRT3200ACM

Ysurac, 2020-07-31:
You don't need to edit manually the file ```/etc/shorewall/rules```, you only need to check the box in System->OpenMPTCProuter then Advanced settings tab. But the DMZ in port forwarding should do the same think if correctly done ( check on https://github.com/Ysurac/openmptcprouter/wiki/Port-forwarding ).
Only ports from 0 to 64999 can be redirected (ports > 65000 are used by OpenMPTCProuter).

florianlog, 2020-08-03:
Thank you for your answer, so I put the "rules" file back to its initial state and create the firewall rules as described in the tutorial.
I still have no access, I have the impression the redirection of the vps to the router is not done.
Is there a way to check the firewall entries in OMR?

Ysurac, 2020-08-03:
Redirections are done over the VPN. You can check on ```/etc/shorewall/rules``` if you see rules created automatically when you added redirections on OMR.
On OMR via SSH, you can check rules using ```iptables-save``` and check if data come to the router when you access a port on the VPS with ```tcpdump```.

florianlog, 2020-08-03:
Thank you, I just went to check,

Shorewall / Rules:

DNAT net vpn: $ OMR_ADDR tcp 1-64999 # OMR openmptcprouter redirect router 1-64999 port tcp
DNAT net vpn: $ OMR_ADDR udp 1-64999 # OMR openmptcprouter redirect router 1-64999 port udp

Iptables:

-A net-vpn -d 10.255.255.2/32 -p tcp -m tcp --dport 1: 64999 -j ACCEPT
-A net-vpn -d 10.255.255.2/32 -p udp -m udp --dport 1: 64999 -j ACCEPT

and testing with tcpdump i can see the traffic coming, but still unreachable, I guess the problem is elsewhere, maybe on the pfsense.

Thank you anyway for the answers !!

closed 2020-12-07 by github-actions[bot]