Ysurac/openmptcprouter #1113: Creation d'un wan3

faktori, 2020-08-02:
Bonjour,

Je test en ce moment le produit est le resultat sur 2 ADSL est parfait. J'ai réussi à mettre Netflix qui fonctionne 👍 
Par contre je voudrais rajouter un wan3 quand je le rajoute dans  Resau/interfaces et que je clique dans parametre physique je ne trouve pas comment créer l'interface physique Wan3

Je suis actuellement sur un Rasp PI 3B et un reseau local avec un switch 

Merci de votre aide

Ysurac, 2020-08-02:
Pour créer un wan3 de type macvlan, le plus simple est de la faire via le wizard (Système->OpenMPTCProuter), en bas de la page basé sur eth0 (ça doit être le défaut).

faktori, 2020-08-02:
Merci Ysurac mais le soucis j'ai testé me donne cela : ( voir photo ) je veux vraiment avoir 3 Wan alors que la , il pointe sur Lan et non un nouveau Wan(3)
[image]

faktori, 2020-08-02:
j ai mis deux connexions sur le meme wan cela fonctionne 👍

Ysurac, 2020-08-02:
C'est pas une bonne idée. Il faut créer les macvlan via le wizard ou via Réseau->Macvlan (ça sera différent dans la prochaine version).

closed 2020-12-05 by github-actions[bot]