Ysurac/openmptcprouter #1114: interface status

zulu14, 2020-08-02:
hello, my interface status no show graph , when i manyapy request the interface_status , the json output is absolutly not correct: 

the ouput are:     {"_":0}

do you have an idea ? (this is not a new instance, anything has been edited,

Ysurac, 2020-08-03:
Please use the template and be more precise: which page ? Do you have any other working wan interface ?

zulu14, 2020-08-03:
manual force the request http://192.168.100.1/cgi-bin/luci/admin/system/openmptcprouter/interfaces_status
the output was : {"_":0}

the theme are not changed, and he has any result obviously

zulu14, 2020-08-09:
manual force the request http://192.168.100.1/cgi-bin/luci/admin/system/openmptcprouter/interfaces_status
the output was : {"_":0}

the theme are not changed, and he has any result

closed 2020-08-09 by Ysurac