Ysurac/openmptcprouter #1116: Connecting openmptcprouter without direct connection to the modem

rifaterdemsahin, 2020-08-04:
## Expected Behavior
I have a router in the middle which has 5 incoming connections.
Openmptcp machine has 2 network cards

## Current Behavior
I have added 5 virtual cards to the esxi mapping to one pyshical network card
[image]

## Possible Solution
Is it possible to use the same network over the switch and let eth0 on the edge router to serve to omr that i can realize this is dedicated to one connection.

My rationale I need to fall back on the Edgerouter incase the OMR does not work. I am planning to add the OMR as a the main WAN port on the Edgerouter which consumes 5 connections over the edge router.

[image]

mehmetakifaksoy, 2020-08-04:
I am also interested to learn if that is possible

Ysurac, 2020-08-04:
I don't understand exactly what you want, but if you want to have all connection on same switch then you need to have each modem/box in own subnet else you have to use vlan.
I don't understand what the edgerouter is made for in your configuration and how it can be used at fallback.

Adorfer, 2020-08-05:
this sounds more like a question for a tutorial "How to virtualize network cards over vlan for VMs". 

In other words: No job for the openmptcprouter here: 
Everthing in terms of vlan tagging and vlan untagging sould be done in the hypervisor and the external vlan-switch.

rifaterdemsahin, 2020-08-10:
`this sounds more like a question for a tutorial "How to virtualize network cards over vlan for VMs".
`
MAKES SENSE finally there is a usecase for vlans. if i can do it over the hypervisor that would be the dream come true,

https://kb.vmware.com/s/article/1004088
https://support.us.ovhcloud.com/hc/en-us/articles/360002411530-How-to-Deploy-VLANs-with-VMware-ESXi-6-5

rifaterdemsahin, 2020-08-10:
[image]

i neeed to find a way to attach a different vlan to each interface

rifaterdemsahin, 2020-08-10:
late night could not sleep
[image]

rifaterdemsahin, 2020-08-10:
Attached
[image]

[image]

rifaterdemsahin, 2020-08-10:
could not see the vlans in openmptcp
[image]
[image]

rifaterdemsahin, 2020-08-10:
each one looks like a different card
[image]

rifaterdemsahin, 2020-08-10:
Too tired After setting the Vlan IDs i cant login to the mptcp interface
[image]

rifaterdemsahin, 2020-08-10:
> and the external vlan-switch.
[image]

is this the external swith (the open openmptcp or are we talking about my layer 3 switch where i connect the cables for the router to the network

rifaterdemsahin, 2020-08-10:
https://help.ui.com/hc/en-us/articles/360046144234-UniFi-Using-VLANs-with-UniFi-Switching#:~:text=VLANs%20can%20be%20used%20with,a%20network%20as%20VLAN%20only.

rifaterdemsahin, 2020-08-10:
[image]

rifaterdemsahin, 2020-08-10:
still not sure if this is the way to go
[image]

As how do i mention the vlan in openmptcp ?

rifaterdemsahin, 2020-08-10:
[image]

rifaterdemsahin, 2020-08-10:
[image]

rifaterdemsahin, 2020-08-10:
[image]

rifaterdemsahin, 2020-08-10:
now i cant login to openmptcp
[image]

[image]

3.23 am i give up for today

rifaterdemsahin, 2020-08-10:
I might need to tag the wan networks with vlans as well maybe but that is another question than me not being able to login to openmptcp

[image]

Adorfer, 2020-08-10:
thanks for clarivication: 
You are looking for a _"how to virtualize network cards from an external VLAN switch into a VM"._ 

1) that is a generic concept which is not special to MPTCProuter
2) depending on the "detail level of howto" 
   -  that is totally specific for a hypervisor (kvm/proxmox is different than hyperv, is differnt than virtualbox etc
   -  that it totally specific for a given switch model/make

Ysurac, 2020-08-10:
For the related OpenMPTCProuter part, I can only say, for hardware that doesn't have an internal switch, if you want a vlan, you need to use as physical interface, for example, eth0.3 to use vlan 3 of interface eth0.

rifaterdemsahin, 2020-08-10:
`, if you want a vlan, you need to use as physical interface, for example, eth0.3 to use vlan 3 of interface eth0.`

thanks mate

rifaterdemsahin, 2020-08-10:
I found this https://github.com/Ysurac/openmptcprouter/issues/168

I would try to edit /etc/config/network and /etc/init.d/network reload

Put added LAN interface in tagged port (id = 11)

config device 'LAN_VLAN42_dev'            
        option name 'LAN_VLAN42'          
        option type '8021q'         
        option ifname 'eth0'        
        option macaddr 'ce:20:fb:ad:90:ef'     
        option vid '11'

Ysurac, 2020-08-10:
There is multiple way to add vlan. I gave the easier. In Network->Interfaces edit an interface and in physical settings you can set what you want.

rifaterdemsahin, 2020-09-04:
i gave up working on my skull canyon the usb ethernet cards and ESXI is not stable so I am planning to buy
[image]

rifaterdemsahin, 2020-09-17:
the cost is high if it does not work on this than it is a bummer
[image]

Adorfer, 2020-09-18:
Can you please rephrase what this issue is actually about. 
In plus it would help to edit the title AND the first post of the issue. 
Just in order to get people on the right track, looking into this the first time.

closed 2021-01-04 by github-actions[bot]