Ysurac/openmptcprouter #1119: Question wan creation

Ilya1963, 2020-08-05:
In System - OpenMPTCProuter - Wizzard - under interface settings we have wan1 and wan2. How to create wan3, wan4 and so on?
In Network - Inrefaces - wan 1 - physical settings = eth1 
                                     - wan2  - physical settings = eth2 
In drop down list I have eth1, eth2 and so on, but do not have interface eth3.

Please, help to understand how to create wan3-wan8 in same manner as wan1 and wan2

Ysurac, 2020-08-05:
What is the hardware used ?
If there is physical eth0 to eth4 then at first boot lan and wan1 to wan4 are created.
If there is only eth0 than lan is created and wan1 and wan2 are created as macvlan interface based on eth0.
If you hare only 2 physical interface and want more wan, you need to choose an interface and add macvlan interfaces based on it. You can add interfaces in Network->MacVlan for current release.

Ilya1963, 2020-08-05:
The hardware is Rasperry PI 4.
Thank you very much
 
Ilya

> 
> 
>What is the hardware used ?
>If there is physical eth0 to eth4 then at first boot lan and wan1 to wan4 are created.
>If there is only eth0 than lan is created and wan1 and wan2 are created as macvlan interface based on eth0.
>If you have only 2 physical interface and want more wan, you need to choose an interface and add macvlan interfaces based on it. You can add interfaces in Network->MacVlan for current release.

Ilya1963, 2020-08-05:
Is it any chance to get help by Teamviewer?
248 833 562
862vce
 
Ilya

> What is the hardware used ?
> If there is physical eth0 to eth4 then at first boot lan and wan1 to wan4 are created.
> If there is only eth0 than lan is created and wan1 and wan2 are created as macvlan [...]

Ysurac, 2020-08-05:
On RPI4 you have one interface if you don't use any Ethernet to USB adapter.
So WAN1 and WAN2 should be macvlan on eth0.
If you use current stable, then macvlan management is in Network->MacVLAN, and you should be able to create interfaces in wizard.
If you use latest beta/release candidate, all can be donne in Network->Interface in Physical parameters tab of each interface.

Ysurac, 2020-08-05:
I don't use teamviewer, and you shouldn't put pass on public issues ;)

Ilya1963, 2020-08-05:
Hardware is Raspberry PI4 

I tried to add interface Network->MacVLAN. 
wan3 with interface eth0 - is it correct? 
In network -  interfaces
wan1 - eth1
wan2 - eth2
wan3 - wan3  - is it correct?

Ysurac, 2020-08-05:
For wan3 yes, for wan2 I don't understand how you can get a eth2.

Ilya1963, 2020-08-05:
https://yadi.sk/i/soxdyowYO_yKbA

for wan2 got eth2 by default

Ilya1963, 2020-08-05:
How many wan I can create in total? I am trying to create wan1-wan8 for usb 4G modems. How many WiFi connections I can add?

Ysurac, 2020-08-05:
Really strange...
wan3 is correct, you only need to add it to firewall wan zone in interface parameters.
For USB modems ? ok so I understand why wan1 and wan2 use an eth interface... I was speaking on adding a virtual port for an ethernet modem/box.
If they have enough power and same type, you should see eth interfaces for all USB modems (some give an ethernet interface like huawei in hilink mode, some need QMI or modemmanager and you need to create new interface).

OpenMPTCProuter support up to 8 WANs (it's an MPTCP hard limit).
If WIFI connections are not used as wan, you can add as many as you want (else the total is still 8).

Ilya1963, 2020-08-05:
https://yadi.sk/i/fCraGOafJQXWCA is it correct with firewall zone?
Is it possible to connect 4G router by LAN connection (DHCP usually) to openMPTCProuter and if yes, how I can do it?

Ysurac, 2020-08-05:
Yes it's correct.
If you talk about an ethernet 4G router, It's possible but not with DHCP (or by using a VLAN if you have a manageable switch), because DHCP with conflict with OpenMPTCProuter DHCP. It's also possible if you use an USB to Ethernet adapter.

Ilya1963, 2020-08-05:
USB to Ethernet adapter is much more easy:) thank you very much - great support - amazing program!!!!

closed 2020-08-05 by Ilya1963