Ysurac/openmptcprouter #1121: can't contact server admin script (no answer)

administrator22, 2020-08-06:
the info of the vps is can't contact server admin script (no answer),what can I do

Ysurac, 2020-08-06:
Complete the template. I need VPS version, OpenMPTCProuter version,...
Did you rebooted the VPS ? And check log on OpenMPTCProuter in State->System log.

administrator22, 2020-08-06:
> Complete the template. I need VPS version, OpenMPTCProuter version,...
> Did you rebooted the VPS ? And check log on OpenMPTCProuter in State->System log.

Yes of course ,I reboot the VPS ,the VPS version is 0.1015.4.19 and the OpenMPTCProuter version is 0.54.7,the system log says Thu Aug  6 08:26:49 2020 kern.notice kernel: [ 8247.874731] nf_ct_snmp: dropping packet: parser failed
Thu Aug  6 08:26:49 2020 kern.notice kernel: [ 8247.874731]  IN= OUT= SRC=192.168.120.196 DST=192.168.120.12 LEN=87 TOS=0x00 PREC=0x00 TTL=126 ID=29455 PROTO=UDP SPT=51435 DPT=161 LEN=67
Thu Aug  6 08:26:55 2020 kern.notice kernel: [ 8253.840218] nf_ct_snmp: dropping packet: parser failed
Thu Aug  6 08:26:55 2020 kern.notice kernel: [ 8253.840218]  IN= OUT= SRC=192.168.120.196 DST=192.168.120.12 LEN=87 TOS=0x00 PREC=0x00 TTL=126 ID=29528 PROTO=UDP SPT=59129 DPT=161 LEN=67
Thu Aug  6 08:27:01 2020 kern.notice kernel: [ 8259.855629] nf_ct_snmp: dropping packet: parser failed
Thu Aug  6 08:27:01 2020 kern.notice kernel: [ 8259.855629]  IN= OUT= SRC=192.168.120.196 DST=192.168.120.12 LEN=87 TOS=0x00 PREC=0x00 TTL=126 ID=29570 PROTO=UDP SPT=59132 DPT=161 LEN=67
Thu Aug  6 08:27:07 2020 kern.notice kernel: [ 8265.839268] nf_ct_snmp: dropping packet: parser failed
Thu Aug  6 08:27:07 2020 kern.notice kernel: [ 8265.839268]  IN= OUT= SRC=192.168.120.196 DST=192.168.120.12 LEN=87 TOS=0x00 PREC=0x00 TTL=126 ID=29651 PROTO=UDP SPT=60173 DPT=161 LEN=67
Thu Aug  6 08:27:13 2020 kern.notice kernel: [ 8271.857098] nf_ct_snmp: dropping packet: parser failed
Thu Aug  6 08:27:13 2020 kern.notice kernel: [ 8271.857098]  IN= OUT= SRC=192.168.120.196 DST=192.168.120.12 LEN=87 TOS=0x00 PREC=0x00 TTL=126 ID=29719 PROTO=UDP SPT=60177 DPT=161 LEN=67
Thu Aug  6 08:27:19 2020 kern.notice kernel: [ 8277.856853] nf_ct_snmp: dropping packet: parser failed
Thu Aug  6 08:27:19 2020 kern.notice kernel: [ 8277.856853]  IN= OUT= SRC=192.168.120.196 DST=192.168.120.12 LEN=87 TOS=0x00 PREC=0x00 TTL=126 ID=29795 PROTO=UDP SPT=60183 DPT=161 LEN=67
Thu Aug  6 08:27:21 2020 daemon.warn dnsmasq-dhcp[3235]: no address range available for DHCP request via eth1
Thu Aug  6 08:27:24 2020 daemon.warn dnsmasq-dhcp[3235]: no address range available for DHCP request via eth1
Thu Aug  6 08:27:25 2020 kern.notice kernel: [ 8283.891949] nf_ct_snmp: dropping packet: parser failed
Thu Aug  6 08:27:25 2020 kern.notice kernel: [ 8283.891949]  IN= OUT= SRC=192.168.120.196 DST=192.168.120.12 LEN=87 TOS=0x00 PREC=0x00 TTL=126 ID=29938 PROTO=UDP SPT=51578 DPT=161 LEN=67
Thu Aug  6 08:27:31 2020 kern.notice kernel: [ 8289.845730] nf_ct_snmp: dropping packet: parser failed
Thu Aug  6 08:27:31 2020 kern.notice kernel: [ 8289.845730]  IN= OUT= SRC=192.168.120.196 DST=192.168.120.12 LEN=87 TOS=0x00 PREC=0x00 TTL=126 ID=30006 PROTO=UDP SPT=51583 DPT=161 LEN=67
Thu Aug  6 08:27:37 2020 kern.notice kernel: [ 8295.870386] nf_ct_snmp: dropping packet: parser failed

Ysurac, 2020-08-06:
I ask about errors related to the issue, so what openmptcprouter-vps script return in log when you validate wizard. It should say something like "OMR-VPS can ping server, no server API answer".

closed 2020-12-09 by github-actions[bot]