Ysurac/openmptcprouter #1123: How to debug high latency (spikes)?

alexciarlillo, 2020-08-06:
## Expected Behavior
Latency should be relatively stable and only slightly increased over normal interface latency.

## Current Behavior
I see spikes in latency and overall generally higher latency when using OpenMPTCProuter vs just the regular device.

## Possible Solution
It feels like this is an issue on the router itself, not the VPS. I have enabled OMR ByPass for the desktop computer I am testing from (by source lan IP). I have also disabled OMRVPN completely. I have tried various congestion algorithms and schedulers.

I say this feels like an issue on the router because while running a ping test I rebooted the VPS. Pings continued to go out and had the same behavior.

Another possibility is that I am currently only using 1 WAN interface. I am still in the testing/configuration phase so I have not added a second interface yet - could this be part of the problem?

## Steps to Reproduce the Problem

  1. 4g modem running ROOter/GoldenOrb on RPi 3b+
  2. Expose modem as a gateway on VLAN3, 192.168.3.1
  3. Add WAN interface on router, eth0.3, static address 192.168.3.30
  4. Perform tests both when directly connected to modem device and when connected to router

## Context (Environment)
When testing using only the modem device, my ping is not great, but is fairly stable in the 60ms range. Of course under heavy bandwidth use I see ICMP being de-prioritized and it rises. All tests performed with no background downloads occurring.

I have 2 VPS in the same datacenter. I am only using one VPS for OpenMPTCP router, I am just using the second as a ping target.

Ping from OpenMPTCP VPS to other VPS. As expected, sub millisecond latency inside the same DC.

```
64 bytes from 97.107.XXX.XXX: icmp_seq=27 ttl=63 time=1.55 ms
64 bytes from 97.107.XXX.XXX: icmp_seq=28 ttl=63 time=1.70 ms
64 bytes from 97.107.XXX.XXX: icmp_seq=29 ttl=63 time=0.442 ms
64 bytes from 97.107.XXX.XXX: icmp_seq=30 ttl=63 time=0.462 ms
64 bytes from 97.107.XXX.XXX: icmp_seq=31 ttl=63 time=0.473 ms
64 bytes from 97.107.XXX.XXX: icmp_seq=32 ttl=63 time=1.59 ms
64 bytes from 97.107.XXX.XXX: icmp_seq=33 ttl=63 time=0.483 ms
64 bytes from 97.107.XXX.XXX: icmp_seq=34 ttl=63 time=0.467 ms
64 bytes from 97.107.XXX.XXX: icmp_seq=35 ttl=63 time=1.61 ms
64 bytes from 97.107.XXX.XXX: icmp_seq=36 ttl=63 time=0.538 ms
64 bytes from 97.107.XXX.XXX: icmp_seq=37 ttl=63 time=0.400 ms
64 bytes from 97.107.XXX.XXX: icmp_seq=38 ttl=63 time=0.558 ms
^C
--- 97.107.XXX.XXX ping statistics ---
38 packets transmitted, 38 received, 0% packet loss, time 704ms
rtt min/avg/max/mdev = 0.400/0.690/1.701/0.391 ms
```

Ping from Windows computer over OpenMPTCP Router. This is an exampled excerpt taken from a much longer test:

```
64 bytes from 97.107.XXX.XXX: icmp_seq=220 ttl=64 time=78.0 ms
64 bytes from 97.107.XXX.XXX: icmp_seq=221 ttl=64 time=48.1 ms
64 bytes from 97.107.XXX.XXX: icmp_seq=222 ttl=64 time=753 ms
64 bytes from 97.107.XXX.XXX: icmp_seq=223 ttl=64 time=349 ms
64 bytes from 97.107.XXX.XXX: icmp_seq=224 ttl=64 time=203 ms
64 bytes from 97.107.XXX.XXX: icmp_seq=225 ttl=64 time=47.2 ms
64 bytes from 97.107.XXX.XXX: icmp_seq=226 ttl=64 time=86.0 ms
64 bytes from 97.107.XXX.XXX: icmp_seq=227 ttl=64 time=500 ms
64 bytes from 97.107.XXX.XXX: icmp_seq=228 ttl=64 time=1281 ms
64 bytes from 97.107.XXX.XXX: icmp_seq=229 ttl=64 time=473 ms
64 bytes from 97.107.XXX.XXX: icmp_seq=230 ttl=64 time=84.1 ms
64 bytes from 97.107.XXX.XXX: icmp_seq=231 ttl=64 time=83.1 ms
64 bytes from 97.107.XXX.XXX: icmp_seq=232 ttl=64 time=42.0 ms
64 bytes from 97.107.XXX.XXX: icmp_seq=233 ttl=64 time=492 ms
64 bytes from 97.107.XXX.XXX: icmp_seq=246 ttl=64 time=75.9 ms
^C
--- 97.107.XXX.XXX ping statistics ---
246 packets transmitted, 246 received, 0% packet loss, time 245147ms
rtt min/avg/max/mdev = 42.094/522.540/2560.598/525.835 ms, pipe 3
```

There is generally a pattern of 3-4 seconds of "normal" pings followed by 3-4 seconds of high latency.

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.54.7 r0+12509-04a21c26a0 / LuCI Master git-20.069.54281-3fb8e3d
  - OpenMPTCProuter VPS version: Version 0.1015 4.19.104-mptcp
  - OpenMPTCProuter platform: x86_64 Intel(R) Core(TM) i3-4030U CPU @ 1.90GHz (QOTOM industrial firewall appliance)

Ysurac, 2020-08-06:
ICMP Ping doesn't general latency but ICMP ping latency (and that is always low priority). ICMP use the VPN (TCP use shadowsocks).
ICMP ping is more stable on next release 0.55.

alexciarlillo, 2020-08-06:
Ah okay that is good to know. Do you know of a reliable way to test/verify TCP latency then?

Ysurac, 2020-08-06:
nmap or netperf can be used for that.
curl should also be usable ( https://help.heroku.com/NY64S5NL/how-do-i-debug-latency-issues-using-curl )

alexciarlillo, 2020-08-06:
I tried using `tcping` for windows and am still getting similar results:

Direct on ROOter device:

```
PS C:\Users\alexc\Downloads> .\tcping.exe -n 20 www.REDACTEDcom

Probing 97.107.XXX.XXX:80/tcp - Port is open - time=66.189ms
Probing 97.107.XXX.XXX:80/tcp - Port is open - time=41.858ms
Probing 97.107.XXX.XXX:80/tcp - Port is open - time=35.769ms
Probing 97.107.XXX.XXX:80/tcp - Port is open - time=29.000ms
Probing 97.107.XXX.XXX:80/tcp - Port is open - time=57.078ms
Probing 97.107.XXX.XXX:80/tcp - Port is open - time=35.436ms
Probing 97.107.XXX.XXX:80/tcp - Port is open - time=36.502ms
Probing 97.107.XXX.XXX:80/tcp - Port is open - time=27.473ms
Probing 97.107.XXX.XXX:80/tcp - Port is open - time=32.648ms
Probing 97.107.XXX.XXX:80/tcp - Port is open - time=32.387ms
Probing 97.107.XXX.XXX:80/tcp - Port is open - time=48.430ms
Probing 97.107.XXX.XXX:80/tcp - Port is open - time=34.426ms
Probing 97.107.XXX.XXX:80/tcp - Port is open - time=48.446ms
Probing 97.107.XXX.XXX:80/tcp - Port is open - time=29.475ms
Probing 97.107.XXX.XXX:80/tcp - Port is open - time=39.432ms
Probing 97.107.XXX.XXX:80/tcp - Port is open - time=75.471ms
Probing 97.107.XXX.XXX:80/tcp - Port is open - time=28.466ms
Probing 97.107.XXX.XXX:80/tcp - Port is open - time=50.440ms
Probing 97.107.XXX.XXX:80/tcp - Port is open - time=40.469ms
Probing 97.107.XXX.XXX:80/tcp - Port is open - time=37.441ms

Ping statistics for 97.107.XXX.XXX:80
     20 probes sent.
     20 successful, 0 failed.  (0.00% fail)
Approximate trip times in milli-seconds:
     Minimum = 27.473ms, Maximum = 75.471ms, Average = 41.342ms
```

Using MPTCP router:
```
PS C:\Users\alexc\Downloads> .\tcping.exe -n 20 www.REDACTED.com

Probing 97.107.XXX.XXX:80/tcp - Port is open - time=1592.453ms
Probing 97.107.XXX.XXX:80/tcp - Port is open - time=429.574ms
Probing 97.107.XXX.XXX:80/tcp - Port is open - time=41.388ms
Probing 97.107.XXX.XXX:80/tcp - Port is open - time=48.420ms
Probing 97.107.XXX.XXX:80/tcp - Port is open - time=68.392ms
Probing 97.107.XXX.XXX:80/tcp - Port is open - time=166.503ms
Probing 97.107.XXX.XXX:80/tcp - Port is open - time=185.529ms
Probing 97.107.XXX.XXX:80/tcp - Port is open - time=24.311ms
Probing 97.107.XXX.XXX:80/tcp - Port is open - time=39.399ms
Probing 97.107.XXX.XXX:80/tcp - Port is open - time=569.046ms
Probing 97.107.XXX.XXX:80/tcp - Port is open - time=376.057ms
Probing 97.107.XXX.XXX:80/tcp - Port is open - time=305.069ms
Probing 97.107.XXX.XXX:80/tcp - Port is open - time=23.410ms
Probing 97.107.XXX.XXX:80/tcp - Port is open - time=63.375ms
Probing 97.107.XXX.XXX:80/tcp - Port is open - time=342.036ms
Probing 97.107.XXX.XXX:80/tcp - Port is open - time=1610.030ms
Probing 97.107.XXX.XXX:80/tcp - Port is open - time=57.173ms
Probing 97.107.XXX.XXX:80/tcp - Port is open - time=44.865ms
Probing 97.107.XXX.XXX:80/tcp - Port is open - time=429.018ms
Probing 97.107.XXX.XXX:80/tcp - Port is open - time=83.421ms

Ping statistics for 97.107.XXX.XXX:80
     20 probes sent.
     20 successful, 0 failed.  (0.00% fail)
Approximate trip times in milli-seconds:
     Minimum = 23.410ms, Maximum = 1610.030ms, Average = 324.973ms
```

Any other ideas of where I can investigate?

Ysurac, 2020-08-06:
```
./tcping -c 20 google.com
Ping tcp://google.com:80(172.217.22.142:80) - Connected - time=949.059µs
Ping tcp://google.com:80(172.217.22.142:80) - Connected - time=1.318112ms
Ping tcp://google.com:80(172.217.22.142:80) - Connected - time=851.783µs
Ping tcp://google.com:80(172.217.22.142:80) - Connected - time=1.076152ms
Ping tcp://google.com:80(172.217.22.142:80) - Connected - time=1.525528ms
Ping tcp://google.com:80(172.217.22.142:80) - Connected - time=1.873214ms
Ping tcp://google.com:80(172.217.22.142:80) - Connected - time=950.398µs
Ping tcp://google.com:80(172.217.22.142:80) - Connected - time=709.436µs
Ping tcp://google.com:80(172.217.22.142:80) - Connected - time=707.796µs
Ping tcp://google.com:80(172.217.22.142:80) - Connected - time=865.521µs
Ping tcp://google.com:80(172.217.22.142:80) - Connected - time=966.807µs
Ping tcp://google.com:80(172.217.22.142:80) - Connected - time=836.666µs
Ping tcp://google.com:80(172.217.22.142:80) - Connected - time=856.062µs
Ping tcp://google.com:80(172.217.22.142:80) - Connected - time=1.130583ms
Ping tcp://google.com:80(172.217.22.142:80) - Connected - time=1.067045ms
Ping tcp://google.com:80(172.217.22.142:80) - Connected - time=689.112µs
Ping tcp://google.com:80(172.217.22.142:80) - Connected - time=1.105041ms
Ping tcp://google.com:80(172.217.22.142:80) - Connected - time=862.732µs
Ping tcp://google.com:80(172.217.22.142:80) - Connected - time=1.01452ms
Ping tcp://google.com:80(172.217.22.142:80) - Connected - time=793.526µs

Ping statistics tcp://google.com:80
        20 probes sent.
        20 successful, 0 failed.
Approximate trip times:
        Minimum = 689.112µs, Maximum = 1.873214ms, Average = 1.007454ms
```
This is using tcping golang version for Linux, so I have no issue (but I'm using 0.55).
Shadowsocks is not disabled ?

Ysurac, 2020-08-06:
And did you try tcping on the IP directly instead of the domain ?

alexciarlillo, 2020-08-06:
Same result when using the IP directly.

Poking around the shadowsocks page I see a lot of things I can disable/enable on all 3 tabs. I haven't touched anything here, but just want to make sure I answer correctly about it being enabled or disabled.

Ysurac, 2020-08-06:
If it's disabled, it's written in OpenMPTCProuter status page.
You can try to set in Network->MPTCP, BLEST MPTCP scheduler and cubic TCP congestion control.
If no change, try 0.55RC or wait for release (should be beginning of next week).

alexciarlillo, 2020-08-06:
Here is a tracert:

```
PS C:\Users\alexc\Downloads> tracert 97.107.xxx.xxx

Tracing route to li63-xxx.members.linode.com [97.107.xxx.xxx]
over a maximum of 30 hops:

  1    <1 ms    <1 ms    <1 ms  HOUSTON.lan [192.168.100.1]
  2    15 ms     5 ms   524 ms  192.168.3.1
  3  1193 ms    67 ms    54 ms  li63-xx.members.linode.com [97.107.xxx.xxx]
```

Ysurac, 2020-08-06:
What is 192.168.3.1 ? This seems to exit directly without using VPS.

alexciarlillo, 2020-08-06:
192.168.3.1 is the modem device. It is running ROOter (based on OpenWRT) specifically for use with cellular modems and that interface is on VLAN3. When I plug directly into this device the ping is fine. 

It is not showing the VPS because I added this machine to OMR ByPass while I try to track down this problem. 

I am now wondering if it has something to do with the vlan gateway configuration. When I plug into the device directly, I get the non-VLAN interface (192.168.1.1 and DHCP). But when I go through the OpenMPTCP router I have to use the VLAN gateway and this is when the issue presents.

alexciarlillo, 2020-08-06:
To clarify, here are the interfaces for the 192.168.1.3 device:

[image]

and OpenMPTCP router:

[image]

Ysurac, 2020-08-06:
macvlan was disabled on wan1 ? this can make conflict (in Network->Macvlan on 0.54)
When you ping from OpenMPTCProuter via SSH 192.168.3.1, there is no issues ?
What to you have in OpenMPTCProuter State->kernel log ? nothing strange with interface ?

alexciarlillo, 2020-08-06:
Aha! There is an issue when pinging from OpenMPTCProuter via SSH to 192.168.3.1:

```
PING 192.168.3.1 (192.168.3.1) 56(84) bytes of data.
64 bytes from 192.168.3.1: icmp_seq=1 ttl=64 time=364 ms
64 bytes from 192.168.3.1: icmp_seq=2 ttl=64 time=0.572 ms
64 bytes from 192.168.3.1: icmp_seq=3 ttl=64 time=0.439 ms
64 bytes from 192.168.3.1: icmp_seq=4 ttl=64 time=364 ms
64 bytes from 192.168.3.1: icmp_seq=5 ttl=64 time=306 ms
64 bytes from 192.168.3.1: icmp_seq=6 ttl=64 time=174 ms
64 bytes from 192.168.3.1: icmp_seq=7 ttl=64 time=0.417 ms
64 bytes from 192.168.3.1: icmp_seq=8 ttl=64 time=0.476 ms
64 bytes from 192.168.3.1: icmp_seq=9 ttl=64 time=373 ms
64 bytes from 192.168.3.1: icmp_seq=10 ttl=64 time=844 ms
64 bytes from 192.168.3.1: icmp_seq=11 ttl=64 time=1093 ms
64 bytes from 192.168.3.1: icmp_seq=12 ttl=64 time=53.3 ms
64 bytes from 192.168.3.1: icmp_seq=13 ttl=64 time=0.485 ms
64 bytes from 192.168.3.1: icmp_seq=14 ttl=64 time=372 ms
^C
--- 192.168.3.1 ping statistics ---
14 packets transmitted, 14 received, 0% packet loss, time 13329ms
rtt min/avg/max/mdev = 0.417/281.846/1093.241/323.479 ms, pipe 2
```

I did not setup any Macvlans, is there something I should else I should do to disable anything there?

Ysurac, 2020-08-06:
If there is any macvlan set in Network->Macvlan, remove it (some are set by default if there is not any ethernet port).
Check that there is no duplicate IP, check cables,...
You should be able to ping it without problem, there is no changes on this part.

alexciarlillo, 2020-08-06:
So I tried removing the switch from the equation, plugging in the ROOter device into eth1, setting the interface in OpenMPTCProuter to eth1.3... issue still persisted. Then I disabled the VLAN interface on the ROOter device, disabled DHCP on the default LAN interface (192.168.1.1), plugged this into eth1 of OpenMPTCP router, and set the interface accordingly and the problem went away.

So it has something to do with the VLAN configuration and my hunch is that it is a problem on the ROOter device, although they both run OpenWRT under the hood 🤷 

I may end up trying my own OpenWRT build with the necessary modem packages added manually so I can setup a true bridge mode back to the router.

Thanks for all your help! It seems like this is not an OpenMPTCP router issue though.

Ysurac, 2020-08-07:
You should be able to use OpenMPTCProuter directly, there is already many modem packages installed.

alexciarlillo, 2020-08-07:
Yeah I am going to try that too. The reason I was using the raspberry pi / ROOter devices was so that I could power them via PoE in a more remote location, closer to the 4g antennas. But if I have to run more ethernet cable or even power to that location to get it working right, I will.

closed 2020-08-17 by alexciarlillo