Ysurac/openmptcprouter #1124: OMR Bypass does not work with 'ShadowSocks Obfuscation' enabled

robertoglxda, 2020-08-09:
## Expected Behavior
Enabling OMR Bypass for Netflix should bypass the VPN.

## Current Behavior
While OMR Bypass is working fine with the default settings, it does not work when the 'Enable ShadowSocks Obfuscating' is enabled (system/openmptcprouter/advanced settings). I've tried with Netflix and Bittorrent: in both cases, no traffic is possible (the Netflix page does not load, and torrents do not download or upload anything).

## Possible Solution
At the moment, only disabling ShadowSocks Obfuscating is working on my setup.

## Steps to Reproduce the Problem

  1. Set Netlix as Service in OMR Bypass
  2. Enable ShadowSocks Obfuscating
  3. Try load netflix.com

## Context (Environment)

## Specifications
- OpenMPTCProuter version: 0.54.7
- OpenMPTCProuter VPS version: 0.1015 4.19.104-mptcp
- OpenMPTCProuter platform: RPI4 with ext4 image 'openmptcprouter-v0.54.7-r012509-04a21c26a0-bcm27xx-bcm2711-rpi-4-ext4-factory.img.gz'

robertoglxda, 2020-08-09:
Ok it actually seems unrelated to the obfuscation (my error testing this). My OMR bypass is not working anymore, and I can't understand why. I'll get back with more info.

Ysurac, 2020-08-09:
Make sure DNS IP is still OpenMPTCProuter IP.

robertoglxda, 2020-08-09:
> Make sure DNS IP is still OpenMPTCProuter IP.

Where should I set this? It doesn't change anything if I set this on my pc.

Ysurac, 2020-08-09:
On you client PC.
You can also check that DoH is disabled, should be if in Network->DHCP & DNS there is the entry "/use-application-dns.net/" in DNS forwardings (if it's not here, add it).

robertoglxda, 2020-08-09:
> On you client PC.
> You can also check that DoH is disabled, should be if in Network->DHCP & DNS there is the entry "/use-application-dns.net/" in DNS forwardings (if it's not here, add it).

I can confirm that the entry is present, but it always has and I cannot access Netflix with OMR Bypass enabled for Netflix. I'm still trying to understand why, I'll give you more info when I get a more clear picture of the situation. It was working until a couple of weeks ago so I'm not sure why it doesn't work anymore.

Ysurac, 2020-08-09:
Netflix page doesn't load at all even if you are not authenticated ?
You tried to reboot the router ?

robertoglxda, 2020-08-09:
> Netflix page doesn't load at all even if you are not authenticated ?
> You tried to reboot the router ?

Correct, I cannot load tha Netflix page from any device (phone, pc) on the network. As soon I disable the OMR bypass, I can access Netflix but I'm recognized in a different country (that is correct, and this is the reason I was using the OMR bypass feature). I get a 'ERR_CONNECTION_REFUSED' on Chrome and a generic error on Edge.

Yes, I've rebooted the router multiple times.

Ysurac, 2020-08-09:
Maybe a new domain need to be bypassed for Netflix... You can check with your browser developer tools what is blocked.
OMR-ByPass is working if you bypass a domain like "ifconfig.co" ?

robertoglxda, 2020-08-09:
> ifconfig.co

It seems a more generic issue than Netflix: I get the ERR_CONNECTION_REFUSED when I enable OMR Bypass also for this domain.
As I said, I thing that also with torrents download I've this issue: when I enable the bypass for the Bittorrent protocol, I cannot download anymore the torrents.

robertoglxda, 2020-08-24:
I fixed the issue.
Under 'Network->interfaces->WAN1->Edit->Firewall Settings' the 'assigned firewall-zone' was 'unpecified'.
Setting it to 'wan' fixes the issue.

We can close this, thanks for your help!

closed 2020-08-24 by robertoglxda