Ysurac/openmptcprouter #1127: v2ray plugin not downloading or installing 404 not found

Rwestrem, 2020-08-10:
## Current Behavior
Install v2ray plugin
--2020-08-10 15:52:34--  https://www.openmptcprouter.com/server-test/bin/v2ray-plugin-linux-amd64-v1.2.0-8-g59b8f4f.tar.gz
Resolving www.openmptcprouter.com (www.openmptcprouter.com)... 68.70.205.3, 68.70.205.1, 68.70.205.4, ...
Connecting to www.openmptcprouter.com (www.openmptcprouter.com)|68.70.205.3|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2020-08-10 15:52:35 ERROR 404: Not Found.

## Steps to Reproduce the Problem

  1. try to install on any platform and still get 404 not found

Darakko, 2020-08-10:
I confirm the same problem.
The file https://www.openmptcprouter.com/server-test/bin/v2ray-plugin-linux-amd64-v1.2.0-8-g59b8f4f.tar.gz is needed by all 3 platforms (Debian 9 + Debian 10 + Ubuntu 18.04).

Worked fine 24h ago.

Rwestrem, 2020-08-10:
> I confirm the same problem.
> The file https://www.openmptcprouter.com/server-test/bin/v2ray-plugin-linux-amd64-v1.2.0-8-g59b8f4f.tar.gz is needed by all 3 platforms (Debian 9 + Debian 10 + Ubuntu 18.04).
> 
> Worked fine 24h ago.

I just tried the link on my android phone and it downloaded without issue.  Its only on pc.

Ysurac, 2020-08-10:
Fixed.
I changed the server for the new release and forget this file.

closed 2020-08-10 by Ysurac

Rwestrem, 2020-08-10:
> I confirm the same problem.
> The file https://www.openmptcprouter.com/server-test/bin/v2ray-plugin-linux-amd64-v1.2.0-8-g59b8f4f.tar.gz is needed by all 3 platforms (Debian 9 + Debian 10 + Ubuntu 18.04).
> 
> Worked fine 24h ago.

Is there any suggestions on how to move forward with my vps install with this issue?  I do have the file.

Darakko, 2020-08-10:
No download from wget/Debian, Firefox/Linux, Firefox/Windows, iexplore/Windows, Edge/Windows, Chrome/Windows.

As a temporary workaround, just modify the script on line 507
Remplace https://www.openmptcprouter.com/${VPSPATH}/bin/v2ray-plugin-linux-amd64-${V2RAY_PLUGIN_VERSION}.tar.gz with the URL of your file (or replace wget by a copy if the file is locale).

If you could upload the file somewhere, this will be cool for other people :-)